Koenkk/zigbee2mqtt #17140: Getting error "Failed to execute LQI for 'Coordinator'" and not reachable devices - multiple add-on restarts fixed it

mdzhigarov, 2023-03-24:
### What happened?

I'm using HomeAssistant with z2m add-on and have created several automations to control the heating of my home. I noticed that temperature has dropped and realized that my automations have been stuck for more than 10h because all of my zigbee devices are unreachable.

Scanning the zigbee network ("Map" button) is reporting error "Failed to execute LQI for 'Coordinator'".
I checked the Coordinator USB - it was plugged all in (because sometimes it tends to disconnect although it's plugged in a usb2 hub). Then restarted the Add-on - the issue still persisted. 
Then restarted the add on a second and a third time - then it worked!

Now the network is fine as usual but I'm worried it might happen again. I'm not comfortable with having to check the zigbee network manually every day to see if the network is stable.

Is there a health check endpoint that I can use to ensure reachability to all of my devices?

### What did you expect to happen?

Stable network and coordinator

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ZBDongle-E

### Debug log

[log (2).txt](https://github.com/Koenkk/zigbee2mqtt/files/11065810/log.2.txt)

FredyReady, 2023-03-25:
Hello,

got same issue. Looking into Z2M Log I cna observe that it will not start:

`[10:38:46] INFO: Preparing to start...
[10:38:47] INFO: Socat not enabled
[10:38:48] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2023-03-25 10:38:53: Logging to console and directory: '/config/zigbee2mqtt/log/2023-03-25.10-38-53' filename: log.txt
Zigbee2MQTT:info  2023-03-25 10:38:53: Starting Zigbee2MQTT version 1.30.2 (commit #unknown)
Zigbee2MQTT:info  2023-03-25 10:38:53: Starting zigbee-herdsman (0.14.96)
Zigbee2MQTT:error 2023-03-25 10:39:14: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-03-25 10:39:14: Failed to start zigbee
Zigbee2MQTT:error 2023-03-25 10:39:14: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-03-25 10:39:14: Exiting...
Zigbee2MQTT:error 2023-03-25 10:39:14: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:540:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:338:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)`

Seems to be related with #15959 

Unfortunately for me, multiple restarts of Z2M does not work, neither complete reboot of whole HA system.

Current version: 1.30.2-1
Adapter ZBDongle-E with Coordinator FW

Whole Zigbee network is now down since last update.

Hopefully this will get fixed soon :)

unrly, 2023-03-29:
Same issue in Home Assistant v1.30.2-1. This morning I suddenly had a bunch of devices drop out and report Offline. Then I realized that none of my devices are working, regardless if they are Online or Offline Availability. It is pinging devices as it's updating the Last Seen, but they aren't able to be controlled either. I removed a device and tried to readd it but it's not even recognized when Permit Join is active.

I'm not getting the herdsman error.

I updated to 1.30.2 a month ago.

Rebooted the system several times. Turned it off, unplugged the dongle for a bit, and turned it back on. Replaced the usb extension cable. I'll try reflashing next I guess.

Version: 1.30.2-1
Adapter: ZBDongle-P

closed 2023-05-06 by github-actions[bot]

KevinGaudry, 2024-05-18:
I have the same issue but with a SLZB-06, connected via LAN. Is there any way to fix it ?

bricololo44, 2024-07-19:
I have a slzb-06p7, when i connected the sonoff router, i obtained a default lsi on coordinatrice.
Need to restor a last image without sonoff router