Koenkk/zigbee2mqtt #17141: Sonoff ITEAD dongle "P" 2 host 2 different behaviorus

JacekKac, 2023-03-25:
### What happened?

HOST A: running raspberry pi 4 / raspian 11: sonoff dongle identified corectly running OK 
HOST B : running raspberry pi 3b / raspian 11: sonoff dongle is running very unstable and badly identified. 

i am pretty sure that both dongles are the same - sonoffs P - not "E". HOST B is working, but devices are dropings, sometimes the adapter is disconnecting not sure if it could be OS problem or sth else. 

Full logs:

**HOST A:** 
```
usb 1-1.3: USB disconnect, device number 6
[5783521.486337] usb 1-1.1: new full-speed USB device number 7 using xhci_hcd
[5783521.633928] usb 1-1.1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[5783521.633940] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[5783521.633947] usb 1-1.1: Product: Sonoff Zigbee 3.0 USB Dongle Plus
[5783521.633954] usb 1-1.1: Manufacturer: ITead
[5783521.633960] usb 1-1.1: SerialNumber: 464f59958012ec11a5ec23c7bd930c07
[5783521.636797] cp210x 1-1.1:1.0: cp210x converter detected
[5783521.649160] usb 1-1.1: cp210x converter now attached to ttyUSB0

```
```
ls usb -v

Bus 001 Device 007: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x10c4 Silicon Labs
  idProduct          0xea60 CP210x UART Bridge
  bcdDevice            1.00
  iManufacturer           1 ITead
  iProduct                2 Sonoff Zigbee 3.0 USB Dongle Plus
  iSerial                 3 464f59958012ec11a5ec23c7bd930c07
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

`usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_464f59958012ec11a5ec23c7bd930c07-if00-port0 -> ../../ttyUSB0
`
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

**Host B** 

```
$ ls -la /dev/serial/by-id/
total 0
drwxr-xr-x 2 root root 60 Mar 24 21:10 .
drwxr-xr-x 4 root root 80 Mar 24 21:10 ..
lrwxrwxrwx 1 root root 13 Mar 24 21:10 usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0

```
```
[6590275.656324] ch341 1-1.4:1.0: device disconnected
[6590275.950820] usb 1-1.4: new full-speed USB device number 77 using dwc_otg
[6590276.083872] usb 1-1.4: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.34
[6590276.083895] usb 1-1.4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[6590276.083906] usb 1-1.4: Product: USB Serial
[6590276.084613] ch341 1-1.4:1.0: ch341-uart converter detected
[6590276.088411] usb 1-1.4: ch341-uart converter now attached to ttyUSB0
```

lsusb -v

```
Bus 001 Device 077: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x7523 CH340 serial converter
  bcdDevice           81.34
  iManufacturer           0
  iProduct                2 USB Serial
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              104mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1
      bInterfaceProtocol      2
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
`
usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0`

```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### What did you expect to happen?

both identified the same

### Zigbee2MQTT version

1.31

### Adapter firmware version

not importatn

### Adapter

both sonoff P dongle

frfstu, 2023-03-26:
> ### What happened?
>
> HOST A: running raspberry pi 4 / raspian 11: sonoff dongle identified corectly running OK HOST B : running raspberry pi 3b / raspian 11: sonoff dongle is running very unstable [...]

I have the same story but already in openHAB! There are 2 absolute coordinators (silicon labs, texas instruments) and both of them work incorrectly in openhab! At the same time, the Coordinator JetStick ZB6UX (CC2652) with firmware zStack3x0 CC1352P2_CC2652P_launchpad_coordinator_20220219 was fully functional and worked correctly in Home Assistant until August 2022 (until the release of the Home Assistant control server on the Internet and spontaneous update of the Supervisor and other components to the current version of HAOS, after which the system ceased to function completely and was no longer loaded (in fact, this was the reason for my transition to openHAB)! What is happening now, you can read here: https://github.com/xsp1989/zigbeeFirmware/issues/37 and here: https://github.com/Koenkk/zigbee2mqtt/issues/16965#issuecomment-1461567616. At the same time, please pay attention to the comment of the coordinator firmware developer: https://github.com/xsp1989/zigbeeFirmware/issues/37#issuecomment-1479491975.

JacekKac, 2023-03-26:
i need to close the issue i double check the hardware on a host B location, it's not sonoff its Egony EBYTEv4

closed 2023-03-26 by JacekKac