Koenkk/zigbee2mqtt #17142: Ikea Symfonisk sound remote gen2 E2123 dotted buttons not working anymore after OTA update

DannyDeGaspari, 2023-03-25:
### What happened?

Following PR #5583 I updated zigbee2mqtt with the Docker development branch. The device showed up with firmware version 1.0.012, all working well also the dotted buttons with short and long press. Then I did an OTA update to version 1.0.32, it took 2 hours and went well. The device is still working but the dotted buttons do not report anything any more in the logs. I tried to reconfigure the device but that results in an error shown in the log below.
A second attempt of reconfiguring does not do anything, I need to restart zigbee2mqtt before I can do another attempt of reconfiguring which then also fails.

### What did you expect to happen?

The dotted buttons to report button presses in the logs.

### How to reproduce it (minimal and precise)

OTA update firmware of E2123 device to 1.0.32

### Zigbee2MQTT version

1.30.2-dev commit: 3e252e4

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff zigbee 3.0 USB dongle plus E

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11069311/log.txt)

MattWestb, 2023-03-26:
Deleting  the device and restarting Z2M and adding it new.
The ShortCut buttons have being moved to new endpoints in the updated firmware and the system is not knowing that so need being added new.

DannyDeGaspari, 2023-03-26:
That does it !
Thanks.

closed 2023-03-26 by DannyDeGaspari