Koenkk/zigbee2mqtt #17147: TuYa PJ-ZGD01 Garage door opener does not work

qhajkasem, 2023-03-26:
### What happened?

The device is recognized and works ok but only if using it with the contact sensor. If the contact sensor's state is false, the trigger does not activate when toggling it to true. You have to toggle it two times before it activates. WIth the contact sensor state to true, it reacts in the same way but now while toggling the trigger to false. 
Trying to activate it with the following command does not do anything at all if the contact sensor is set to false:
mosquitto_pub -h %ipaddress% -t "zigbee2mqtt/%deviceid%/set" -m '{ "trigger": "true [or false]" '}
If the contact sensor is set to true the above commands works for both true and false every time I run it.

Note: this issue has been logged before (#12823) but that one was closed without a solution.

### What did you expect to happen?

The relay should be triggered at all times. 

### How to reproduce it (minimal and precise)

Join the device, do not hold a magnet close to the sensor and toggle the switch in the Z2M frontend. 
Same applies for the mosquito command but that one does nothing at all.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20221226

### Adapter

Slaesh's cc2652rb_stick

### Debug log

With the contact sensor's state as false and toggling the switch in the frontend to true and then to false (the false setting activates the relay):

debug 2023-03-26 05:46:58: Received MQTT message on 'zigbee2mqtt/Device-XXX/set' with data '{"trigger":true}'
debug 2023-03-26 05:46:58: Publishing 'set' 'trigger' to 'Device-XXX'
info  2023-03-26 05:46:58: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"garage_door_contact":false,"linkquality":120,"trigger":true}'
debug 2023-03-26 05:46:58: Received Zigbee message from 'Device-XXX', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4608}' from endpoint 1 with groupID 0
info  2023-03-26 05:46:58: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"action":"trigger","garage_door_contact":false,"linkquality":134,"trigger":true}'
info  2023-03-26 05:46:58: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"action":"","garage_door_contact":false,"linkquality":134,"trigger":true}'
info  2023-03-26 05:46:58: MQTT publish: topic 'zigbee2mqtt/Device-XXX/action', payload 'trigger'
debug 2023-03-26 05:47:01: Received MQTT message on 'zigbee2mqtt/Device-XXX/set' with data '{"trigger":false}'
debug 2023-03-26 05:47:01: Publishing 'set' 'trigger' to 'Device-XXX'
info  2023-03-26 05:47:01: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"garage_door_contact":false,"linkquality":134,"trigger":false}'
debug 2023-03-26 05:47:01: Received Zigbee message from 'Device-XXX', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4864}' from endpoint 1 with groupID 0
info  2023-03-26 05:47:01: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"action":"trigger","garage_door_contact":false,"linkquality":134,"trigger":false}'
info  2023-03-26 05:47:01: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"action":"","garage_door_contact":false,"linkquality":134,"trigger":false}'
info  2023-03-26 05:47:01: MQTT publish: topic 'zigbee2mqtt/Device-XXX/action', payload 'trigger'
debug 2023-03-26 05:47:04: Received Zigbee message from 'Device-XXX', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":3}],"seq":5376}' from endpoint 1 with groupID 0
info  2023-03-26 05:47:04: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"garage_door_contact":false,"linkquality":134,"trigger":false}'

And the output with the mosquito_pub command (and the contact sensor's state as false):

debug 2023-03-26 05:48:18: Received MQTT message on 'zigbee2mqtt/Device-XXX/set' with data '{ "trigger": "true" }'
debug 2023-03-26 05:48:18: Publishing 'set' 'trigger' to 'Device-XXX'
info  2023-03-26 05:48:18: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"garage_door_contact":false,"linkquality":131,"trigger":"true"}'
debug 2023-03-26 05:48:18: Received Zigbee message from 'Device-XXX', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":7168}' from endpoint 1 with groupID 0
info  2023-03-26 05:48:18: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"action":"trigger","garage_door_contact":false,"linkquality":134,"trigger":"true"}'
info  2023-03-26 05:48:18: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"action":"","garage_door_contact":false,"linkquality":134,"trigger":"true"}'
info  2023-03-26 05:48:18: MQTT publish: topic 'zigbee2mqtt/Device-XXX/action', payload 'trigger'
debug 2023-03-26 05:48:20: Received MQTT message on 'zigbee2mqtt/Device-XXX/set' with data '{ "trigger": "false" }'
debug 2023-03-26 05:48:20: Publishing 'set' 'trigger' to 'Device-XXX'
info  2023-03-26 05:48:20: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"garage_door_contact":false,"linkquality":134,"trigger":"false"}'
debug 2023-03-26 05:48:20: Received Zigbee message from 'Device-XXX', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":7424}' from endpoint 1 with groupID 0
info  2023-03-26 05:48:20: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"action":"trigger","garage_door_contact":false,"linkquality":134,"trigger":"false"}'
info  2023-03-26 05:48:20: MQTT publish: topic 'zigbee2mqtt/Device-XXX', payload '{"action":"","garage_door_contact":false,"linkquality":134,"trigger":"false"}'
info  2023-03-26 05:48:20: MQTT publish: topic 'zigbee2mqtt/Device-XXX/action', payload 'trigger'

qhajkasem, 2023-04-26:
Still an issue for me and haven't found a workaround.

minimmi, 2023-05-30:
I currently have the same problem. I want to trigger the relay regardless of the state of the contact sensor. Is there a solution for this in the meantime?

qhajkasem, 2023-05-31:
I never found a solution or workaround so I replaced it with another device (CM10ZW from Orvibo) and discarded the Tuya opener.

minimmi, 2023-05-31:
Thanks for your feedback. I have currently solved it in my automations as follows: 

```
if:
  - type: is_open
    condition: device
    device_id: XX
    entity_id: binary_sensor.gargentor_garage_door_contact
    domain: binary_sensor
then:
  - type: turn_off
    device_id: XX
    entity_id: switch.gargentor_trigger
    domain: switch
else:
  - type: turn_on
    device_id: XX
    entity_id: switch.gargentor_trigger
    domain: switch

qhajkasem, 2023-05-31:
I tried something similar but it wasn't as reliable as I hoped (could be a mistake on my end of course but it sometimes just wouldn't open or close).

closed 2023-07-08 by github-actions[bot]

TRIROG, 2023-07-17:
Still an issue

MardariG, 2023-07-27:
I got same issue. From what I understood this is dependent on the board door is using. My board for example is expecting this trigger to happen under a second. So for it to work the relay must really fast become ON and OF. Currently when I trigger it, it becomes ON then waits a few seconds, then becomes OFF, and my board do not react to this at all. Not sure if this can be fixed on zigbee2mqtt side or at firmware level though.

MardariG, 2023-07-27:
#12823

tatarinjn, 2024-01-07:
Такая же проблема! Реле работает только в случае если датчик положения "Закрыто" После открытия ворот их уже не закрыть.

noahcub, 2024-07-13:
> Thanks for your feedback. I have currently solved it in my automations as follows:
>
> ```
> if:
>   - type: is_open
>     condition: device
>     device_id: XX
>     entity_id: [...]

I have the same problem. Can you post the complete automation in home assistant?

PaoloVittori, 2025-02-18:
Try this automation

  - alias: "Sync garage door opener trigger"
    id: "sync_garage_door_opener_trigger"
    mode: single
    trigger: []
    condition: []
    action:
      - if:
          - type: is_open
            condition: device
            device_id: xxx
            entity_id: binary_sensor.gargentor_garage_door_contact
            domain: binary_sensor
        then:
          - type: turn_off
            device_id: xxx
            entity_id: switch.gargentor_trigger
            domain: switch
        else:
          - type: turn_on
            device_id: xxx
            entity_id: switch.gargentor_trigger
            domain: switch