Koenkk/zigbee2mqtt #17148: Xiaomi Aqara C1 Pet Feeder (ZNCWWSQ01LM) Unhandled attribute

eikowagenknecht, 2023-03-26:
### What happened?

During interview, errors are shown for unknown attributes and the following is logged:

```
Debug 2023-03-25 18:44:14Received Zigbee message from '0x54ef44100070e06b', type 'attributeReport', cluster 'aqaraOpple', data '{"65521":{"data":[0,5,9,8,0,7,209,1,100],"type":"Buffer"}}' from endpoint 1 with groupID 0
Warning 2023-03-25 18:44:14zigbee-herdsman-converters:aqara_feeder: Unhandled attribute ��� = d
Debug 2023-03-25 18:44:14Received Zigbee message from '0x54ef44100070e06b', type 'attributeReport', cluster 'aqaraOpple', data '{"65521":{"data":[0,5,10,13,9,0,85,1,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Warning 2023-03-25 18:44:14zigbee-herdsman-converters:aqara_feeder: Unhandled attribute U = 
```

### What did you expect to happen?

No errors in the interview process

### How to reproduce it (minimal and precise)

Reset device, then pair it

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20221226

### Adapter

zStack3x0

eikowagenknecht, 2023-03-29:
One more from another device:

```
Debug 2023-03-29 12:37:06Received Zigbee message from '0x54ef4410006a1576', type 'readResponse', cluster 'aqaraOpple', data '{"65521":{"data":[49,50,51],"type":"Buffer"}}' from endpoint 1 with groupID 0
Error 2023-03-29 12:37:06Exception while calling fromZigbee converter: Attempt to access memory outside buffer bounds}
Debug 2023-03-29 12:37:06RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:387:5) at boundsError (node:internal/buffer:84:11) at Buffer.readUInt8 (node:internal/buffer:252:5) at /app/node_modules/zigbee-herdsman-converters/devices/xiaomi.js:167:51 at Array.forEach (<anonymous>) at Object.convert (/app/node_modules/zigbee-herdsman-converters/devices/xiaomi.js:163:38) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
```

eikowagenknecht, 2023-03-29:
Another problem: Firmware version is not shown in UI:

State:

```
    "update": {
        "installed_version": 9505,
        "latest_version": 9761,
        "state": "available"
    },
```

OTA page: No version shown

[image]

Updating works though as far as I can see.

eikowagenknecht, 2023-03-29:
After editing the schedule partially (e.g. minutes field only), the button to save is grayed out:

[image]

It only becomes active after changing all 4 fields (day, hour, minute, portions) to something else each and then back again.

eikowagenknecht, 2023-03-29:
Turning the "LED indicator" setting on and off has no effect on the device whatsoever:

[image]

eikowagenknecht, 2023-03-29:
The properties are named in a confusing way. It would be better to name them as ZHA does:

[image]

Should be

- last_feeding_source
- last_feeding_size
- portions_today
- weight_today

eikowagenknecht, 2023-03-29:
Values are not checked at all.

Minutes can be set to >=60 or <0
Hours can also be set to >=24 or <0
Portions can be set to <=0 or >10 (device max)

[image]

eikowagenknecht, 2023-03-29:
It is unclear in which time zone the schedule operates. From observation it seems to be my local time zone (CEST), but how will it handle summer / wintertime changes for example? This should be at least documented somewhere.

eikowagenknecht, 2023-03-29:
And another one: If the feeding size is set to 10 then the "last feeding" ist shown as Null:

[image]

austrisv, 2023-04-14:
> It is unclear in which time zone the schedule operates. From observation it seems to be my local time zone (CEST), but how will it handle summer / wintertime changes for example? This should be at least documented somewhere.

it appears to me it was GMT time (I am GMT+3 currently). Will wait for day to end and verify by checking when "feeder portions per day" does it's reset.

austrisv, 2023-04-14:
for a reference, this is my setup. Also observed similar "Unhandled attribute" errors upon binding. But it succeeded eventually, after few attempts. Then was able to do OTA fw upgrade. 
When setting a schedule over MQTT - it generated some errors when day count got over 4days. But looks like saved ok and is working now.
 
Zigbee2MQTT version: 1.29.2
Coordinator type: zStack3x0
Coordinator revision: 20220219
Frontend version: 0.6.121

eikowagenknecht, 2023-05-15:
Still current

Rustymage, 2023-05-30:
I have the greyed out schedule and LED indicator non functioning issues you have mentioned. Seems prevalent if you view some of the stale/closed non resolved issues in the repo.

Hopefully someone with experience can assist.

thbrd, 2023-07-25:
Bought a C1 yesterday. Same error's on my side:

1: Led indicator enable/disable not working
2: Same "Unknown attribute" messages
3: All other above error's

thbrd, 2023-10-03:
Bumb Bumb

KirovAir, 2023-10-24:
I just received mine and I'll have a look at it when I have some spare time. Seems like a nice puzzle to solve.

fribse, 2023-11-15:
I see the errors as well, I've tried pairing and unpairing a few times, and also had it enrolled in ZHA which didn't show any of the errors. ZHA doesn't turn off the LED either.

gjdass, 2023-11-24:
New issue here.
Updating the schedule does not work at all.

1. take on schedule record and change the hour for example
2. change a second entry
3. the first entry I edited is back to its previous state ...

Also, saving the schedule will "apparently" work the first 5min, then the default schedule is back and my schedule is completly removed.
I don't see any error, just the device seems to do whatever it wants (and reseting also).

Other new issue : scheduling mode is no longer taken into account. Changing it from schedule to manual has no effect, I can see in the payload it stays at schedule ...

This makes the device unusable with HA and Zigbee2Mqtt right now.

kirovilya, 2023-12-05:
Unfortunately I sold my feeder...

I tested the schedule on a small number of positions (3 positions), so I don’t know how the device behaves with a larger number of positions.

gjdass, 2023-12-05:
> @gjdass, If it comes to worst, you should still be able to set the schedule over a MQTT message.
> 
> 
> 
> publish to `zigbee2mqtt/FRIENDLY_NAME/set `
> 
> with payload `{"schedule": [{"days": VALUE, "hour": VALUE, "minute": VALUE, "size": VALUE}]}`
> 
> like 
> 
>    `{"schedule": [
> 
>         {
> 
>             "days": "workdays",
> 
>             "hour": 5,
> 
>             "minute": 30,
> 
>             "size": 2
> 
>         },
> 
>         {
> 
>             "days": "everyday",
> 
>             "hour": 9,
> 
>             "minute": 30,
> 
>             "size": 2
> 
>         }}`
> 
> 
> 
> > This makes the device almost unusable with HA and Zigbee2Mqtt right now. Let's say it's very "random" each time I try to edit things, and I never know if I'm gonna need to reset or not etc.
> 
> >
> 
> 
> 
> 

Hey !

Yep it works (tried it) but after 5-6 minutes or so, it's back to the default schedule. "By itself". 
So I'm kinda blocked with that currently. I really have no clue why it behaves like that.
Tried several resets etc.

jeells102, 2023-12-10:
I've also had these issues, and I've had it where my Z2M has crashed. I have 2 pet feeders and when Z2M crashes and it restarts, only one of them reconnects. I posted my issue on the HA community forum here: https://community.home-assistant.io/t/help-me-understand-why-z2m-addon-crashes/652653/13

I have just updated my script (I call the script as part of my automation) so it's the same as thbrd in the hopes that it helps prevent issues.

jeells102, 2024-01-09:
I'm seeing unhandled error messages with any change I make to the pet feeder in Z2M. 

```
debug 2024-01-09 16:00:37: Received Zigbee message from '0x54ef44100070ea46', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[5,33,158,0,13,35,33,38,0,0,9,33,0,16],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 2024-01-09 16:00:37: zigbee-herdsman-converters:aqara_feeder: Unhandled key 247 = !� 
```

```
debug 2024-01-09 16:05:39: Received Zigbee message from '0x54ef44100070ea46', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
info  2024-01-09 16:05:39: MQTT publish: topic 'zigbee2mqtt/0x54ef44100070ea46', payload '{"child_lock":null,"error":null,"feed":null,"feeding_size":null,"feeding_source":null,"last_seen":"2024-01-09T16:05:39+00:00","led_indicator":null,"linkquality":80,"mode":null,"portion_weight":null,"portions_per_day":null,"schedule":[{"days":"everyday","hour":6,"minute":0,"size":2},{"days":"everyday","hour":9,"minute":0,"size":3},{"days":"everyday","hour":12,"minute":0,"size":4},{"days":"everyday","hour":15,"minute":0,"size":2},{"days":"everyday","hour":20,"minute":0,"size":4}],"serving_size":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"weight_per_day":null}'
debug 2024-01-09 16:05:39: Successfully pinged '0x54ef44100070ea46' (attempt 1/2)
debug 2024-01-09 16:05:39: Received Zigbee message from '0x54ef44100070ea46', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[5,33,158,0,13,35,33,38,0,0,9,33,0,16],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 2024-01-09 16:05:39: zigbee-herdsman-converters:aqara_feeder: Unhandled key 247 = !� 
```

```
info  2024-01-09 16:13:02: MQTT publish: topic 'zigbee2mqtt/0x54ef44100070ea46', payload '{"child_lock":null,"error":null,"feed":null,"feeding_size":null,"feeding_source":null,"last_seen":"2024-01-09T16:13:02+00:00","led_indicator":null,"linkquality":76,"mode":"schedule","portion_weight":null,"portions_per_day":null,"schedule":[{"days":"everyday","hour":6,"minute":0,"size":2},{"days":"everyday","hour":9,"minute":0,"size":3},{"days":"everyday","hour":12,"minute":0,"size":4},{"days":"everyday","hour":15,"minute":0,"size":2},{"days":"everyday","hour":20,"minute":0,"size":4}],"serving_size":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"weight_per_day":null}'
debug 2024-01-09 16:13:02: Received Zigbee message from '0x54ef44100070ea46', type 'attributeReport', cluster 'aqaraOpple', data '{"65521":{"data":[128,2,2,48],"type":"Buffer"}}' from endpoint 1 with groupID 0
error 2024-01-09 16:13:02: Exception while calling fromZigbee converter: Attempt to access memory outside buffer bounds}
debug 2024-01-09 16:13:03: RangeError: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:405:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt8 (node:internal/buffer:254:5)
    at /app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:212:51
    at Array.forEach (<anonymous>)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:206:38)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:51)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:101:22)
```

thbrd, 2024-01-16:
@Koenkk Can you help us or maybe do you know someone else who can help fix these problems? With every zigbee2mqtt update, I am hoping to see fixes for the C1 pet feeder. If I can help you in any way, please let me know :)

Koenkk, 2024-01-17:
@jeells102 coud you provide the herdsman debug log of the following error?

```
RangeError: Attempt to access memory outside buffer bounds
```

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

jeells102, 2024-01-18:
@Koenkk Hopefully below provides you with what you need. 

```
2024-01-18T12:30:20.766Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":54163,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":1303103,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[28,95,17,32,10,241,255,65,4,128,2,2,48]}}
2024-01-18T12:30:20.766Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":32,"manufacturerCode":4447,"commandIdentifier":10},"Payload":[{"attrId":65521,"dataType":65,"attrData":{"type":"Buffer","data":[128,2,2,48]}}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":54163,"endpoint":1,"linkquality":94,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-18T12:30:20.767Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2024-01-18 12:30:20: Received Zigbee message from '0x54ef44100070ea46', type 'attributeReport', cluster 'aqaraOpple', data '{"65521":{"data":[128,2,2,48],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2024-01-18 12:30:20: Exception while calling fromZigbee converter: Attempt to access memory outside buffer bounds}
Zigbee2MQTT:debug 2024-01-18 12:30:20: RangeError: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:405:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt8 (node:internal/buffer:254:5)
    at /app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:212:51
    at Array.forEach (<anonymous>)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:206:38)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:51)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:101:22)
Zigbee2MQTT:info  2024-01-18 12:30:20: MQTT publish: topic 'zigbee2mqtt/0x54ef44100070ea46', payload '{"child_lock":null,"error":null,"feed":null,"feeding_size":null,"feeding_source":null,"last_seen":"2024-01-18T12:30:20+00:00","led_indicator":null,"linkquality":94,"mode":null,"portion_weight":null,"portions_per_day":null,"schedule":[{"days":"everyday","hour":6,"minute":0,"size":2},{"days":"everyday","hour":9,"minute":0,"size":2},{"days":"everyday","hour":12,"minute":0,"size":2},{"days":"everyday","hour":15,"minute":0,"size":2},{"days":"everyday","hour":20,"minute":0,"size":2}],"serving_size":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"weight_per_day":null}'
2024-01-18T12:30:20.914Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,65,61,1,43,79,157]
2024-01-18T12:30:20.914Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,65,61,1,43,79,157]
2024-01-18T12:30:20.914Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [65,61,1,43,79] - 157
2024-01-18T12:30:20.914Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":15681,"relaycount":1,"relaylist":[20267]}
2024-01-18T12:30:20.914Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-18T12:30:20.971Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,65,61,1,13,0,112,0,145,20,20,0,0,12,1,6,6,0,11,16,232,3,3,0,0,0,43,79,28,235]
2024-01-18T12:30:20.971Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,65,61,1,13,0,112,0,145,20,20,0,0,12,1,6,6,0,11,16,232,3,3,0,0,0,43,79,28,235]
2024-01-18T12:30:20.972Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,65,61,1,13,0,112,0,145,20,20,0,0,12,1,6,6,0,11,16,232,3,3,0,0,0,43,79,28] - 235
2024-01-18T12:30:20.972Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":15681,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":112,"securityuse":0,"timestamp":1315985,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,6,6,0,11,16,232,3,3,0,0,0]}}
2024-01-18T12:30:20.972Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":6},"Payload":{"status":0,"manufacturerCode":4107,"imageType":1000,"fileVersion":3},"Command":{"ID":6,"response":7,"parameters":[{"name":"status","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"upgradeEndRequest"}},"address":15681,"endpoint":1,"linkquality":112,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
2024-01-18T12:30:20.972Z zigbee-herdsman:controller:log Skipping command 'upgradeEndRequest' because it is missing from the lookup
Zigbee2MQTT:info  2024-01-18 12:30:20: MQTT publish: topic 'zigbee2mqtt/Dining Room Light Switch', payload '{"brightness":254,"last_seen":"2024-01-18T12:30:20+00:00","linkquality":112,"power_on_behavior":null,"state":"OFF"}'
2024-01-18T12:30:20.974Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...] ufacturerCode":4107,"imageType":1000,"fileVersion":3},"Command":{"ID":6,"response":7,"parameters":[{"name":"status","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"upgradeEndRequest"}},"address":29521,"endpoint":1,"linkquality":102,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
2024-01-18T12:33:56.918Z zigbee-herdsman:controller:log Skipping command 'upgradeEndRequest' because it is missing from the lookup
```

jeells102, 2024-01-19:
@Koenkk That's amazing news! Thanks for working on this. Again, once this is released into production I'll test and let you know.

thbrd, 2024-01-25:
@Koenkk Thank you for fixing this issue! Will test this when it is in production.

fribse, 2024-02-01:
I've just updated to [1.35.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.2)
So far the schedule looks right, and it seems to keep it as it should.
The LED on/off still doesn't work.
And the 'feed button' still looks odd:
[image]
I've set it to feed at 6.30 and 18.00 tomorrow, so I'll see if the schedule works.

jeells102, 2024-02-03:
@Koenkk I have updated and unfortuntely, my Zigbee network is still crashing with the pet feeders added. 

`2024-02-03T10:47:34.880Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,221,3,0,92]
2024-02-03T10:47:34.880Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,221,3,0,92]
2024-02-03T10:47:34.880Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [221,3,0] - 92
2024-02-03T10:47:34.880Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":989,"relaycount":0,"relaylist":[]}
2024-02-03T10:47:34.880Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-02-03T10:47:34.881Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,221,3,0,92]
2024-02-03T10:47:34.881Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,221,3,0,92]
2024-02-03T10:47:34.881Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [221,3,0] - 92
2024-02-03T10:47:34.881Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":989,"relaycount":0,"relaylist":[]}
2024-02-03T10:47:34.881Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-02-03T10:47:34.959Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,221,3,0,92]
2024-02-03T10:47:34.959Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,221,3,0,92]
2024-02-03T10:47:34.960Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [221,3,0] - 92
2024-02-03T10:47:34.960Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":989,"relaycount":0,"relaylist":[]}
2024-02-03T10:47:34.960Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-02-03T10:47:34.962Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,221,3,1,242,0,51,0,124,38,66,0,0,7,8,0,10,0,0,16,0,221,3,29,8]
2024-02-03T10:47:34.962Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,221,3,1,242,0,51,0,124,38,66,0,0,7,8,0,10,0,0,16,0,221,3,29,8]
2024-02-03T10:47:34.962Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,221,3,1,242,0,51,0,124,38,66,0,0,7,8,0,10,0,0,16,0,221,3,29] - 8
2024-02-03T10:47:34.963Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":989,"srcendpoint":1,"dstendpoint":242,"wasbroadcast":0,"linkquality":51,"securityuse":0,"timestamp":4335228,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,0,10,0,0,16,0]}}
2024-02-03T10:47:34.964Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":0,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":989,"endpoint":1,"linkquality":51,"groupID":0,"wasBroadcast":false,"destinationEndpoint":242}'
2024-02-03T10:47:34.965Z zigbee-herdsman:controller:endpoint DefaultResponse 0x94deb8fffe483239/1 6(10, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2024-02-03T10:47:34.965Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x94deb8fffe483239:989/1 (0,0,7)
2024-02-03T10:47:34.965Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":989,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":31,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,0,11,10,0]}}
2024-02-03T10:47:34.965Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,221,3,1,1,6,0,31,0,30,5,24,0,11,10,0,239]
2024-02-03T10:47:34.966Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2024-02-03 10:47:34: Received Zigbee message from 'Office Light Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-02-03 10:47:34: MQTT publish: topic 'zigbee2mqtt/Office Light Switch', payload '{"brightness":220,"last_seen":"2024-02-03T10:47:34+00:00","linkquality":51,"power_on_behavior":null,"state":"OFF"}'
2024-02-03T10:47:34.968Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,221,3,1,1,0,51,0,32,40,66,0,0,7,8,0,10,0,0,16,0,221,3,29,169]
2024-02-03T10:47:34.968Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,221,3,1,1,0,51,0,32,40,66,0,0,7,8,0,10,0,0,16,0,221,3,29,169]
2024-02-03T10:47:34.968Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,221,3,1,1,0,51,0,32,40,66,0,0,7,8,0,10,0,0,16,0,221,3,29] - 169
2024-02-03T10:47:34.968Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":989,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":51,"securityuse":0,"timestamp":4335648,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,0,10,0,0,16,0]}}
2024-02-03T10:47:34.968Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":0,"manuf [... cut ...]
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:390:28)
    at sendAttr (/app/node_modules/zigbee-herdsman-converters/src/lib/lumi.ts:1640:17)`

Koenkk, 2024-02-03:
@jeells102 found the issue and pushed a fix for it.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

jeells102, 2024-02-03:
@Koenkk That was incredibly quick! Thanks so much. Will need to wait till it's out in production but will let you know. Thank you

jeells102, 2024-02-17:
@Koenkk I've been using the new update and I've not had any issues! Thanks so much for fixing this :)

thbrd, 2024-03-02:
@Koenkk Thank you so much for making these fixes! There is only 1 thing I noticed: when I set the Led indicator to false it is ON. When I set it to ON it is off 🤣 no big deal. But an easy fix, or am I the only one having this issue.

Koenkk, 2024-03-04:
@thbrd fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

lorenzomartini1, 2024-03-13:
@Koenkk in 1.36.0, when I try to set more the five schedules, starting from the 6th I got this error:

Publish 'set' 'schedule' to 'Smart Pet Feeder' failed: 'Error: Write 0x54ef44100077fc1a/1 manuSpecificLumi({"65521":{"value":{"type":"Buffer","data":[0,2,3,8,0,8,200,66,55,102,49,54,49,101,48,49,48,48,44,55,102,48,53,49,101,48,49,48,48,44,55,102,48,55,49,101,48,49,48,48,44,55,102,48,98,48,48,48,49,48,48,44,55,102,49,50,49,101,48,49,48,48,44,55,102,48,101,49,101,48,50,48,48,0]},"type":65}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'

5 schedules are working well

can you help me?

Koenkk, 2024-03-14:
@lorenzomartini1 I don't know how many schedules this device supports.

foxygnu, 2024-04-02:
I don't know how many schedules this device supports but it is at least 6. You can look at this test of the product: https://www.lesnumeriques.com/objet-connecte/aqara-smart-pet-feeder-c1-p71951/test.html . You can see [here](https://cdn.lesnumeriques.com/optim/test/20/206879/fe0a2368-distributeur-de-croquettes-aqara-smart-pet-feeder-c1-un-ecosysteme-connecte-pour-chiens-et-chats__w1280_wtmk.jpg) a test with 6 schedules.

sippeeey, 2024-07-03:
thanks for great work so far.

is it possible to change the "Feed" entity from a Dropdown into a Button, cause the problem is, it feeds also when dropdown is change to "nothing". it makes no sense.

[image]
[image]

antz29, 2024-08-27:
Ok, main issue I'm seeing is that it doesn't show any error status at all, so no report if the food is empty. Any way to fix this? I'm happy to take a look but not sure where to start 😄

foxygnu, 2024-08-28:
The dispenser does not have any sensor signaling that there are no more food in the tank.

antz29, 2024-08-28:
Oh right, ok, that's a bit of a pita but explains things! :)

foxygnu, 2024-08-28:
Apart from that it is a very good automatic food dispenser, it never jams and the portions are very consistent (in terms of weight).

antz29, 2024-08-29:
Yeh, seems that way so far, had a petkit one that was giving me loads of problems. Nice excuse for me to play around with building an esphome weight sensor 👍

closed 2025-01-13 by github-actions[bot]