Koenkk/zigbee2mqtt #17152: Availability not working - powered devices not being pinged

jonatrey, 2023-03-26:
### What happened?

The device availability feature is not working as intended for devices that are powered off. There is no ping happening to the devices. 

- Last seen metadata is correct,
- After rebooting zigbee2mqtt, the correct availability is published,
- When powering on a device, the new correct availability is published (from unavailable to online)
- After powering off the device, the availability is not published - the device is not pinged
- This is happening with all devices

Hardware
- My network has about 60 devices, some permanently powered, some on battery and some on traditional switches. I disabled the availability feature for the permanently powered devices.
- Adapter Conbee II with the recommended firmware version (0x26580700)

Log entries:
- Log level is in Debug, no single entries (no ping happening of failing) from https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/availability.ts file
- Can see periodic OTA requests in the debug logs

```
Config:
advanced:
  log_level: debug
  ..
  adapter_delay: 100
  cache_state: true
  last_seen: ISO_8601
  adapter_concurrent: 2
  legacy_availability_payload: false
availability:
  active:
    timeout: 5
  passive:
    timeout: 1500
```

### What did you expect to happen?

I expect to have my powered on devices pinged at 5 min intervals.

### How to reproduce it (minimal and precise)

Power cycle a bulb, 

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Debug log

[zigbee_debug_log.log](https://github.com/Koenkk/zigbee2mqtt/files/11072749/zigbee_debug_log.log)

jonatrey, 2023-03-27:
More precisions:
- The behavior is the same if I just use the "simple availability" setting from the UI
- The devices where availability is configured and powered on have for instance "last seen: 2 hours ago, availability online" - if I update a setting on the light, then the last seen is updated.
- If I physically power off these devices, everything remains the same (luminosity, color temp, and last seen is just increasing)
- Neither the Mqttlogs nor the zigbee-herdsman logs show any ping happening or messages received from these devices.

jonatrey, 2023-03-28:
Tried to execute manual pings to see the behavior.

`
Warning 2023-03-28 19:45:12Failed to ping 'Ledstrip' (attempt 1/1, Read 0x001788010491ffbd/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received))
(I saw that popping up in the logs)
`

For the most standard attributes I don't get any response, it's probably timing out after 10000 secs? For other less standard attributes I get UNSUPPORTED_ATTRIBUT
`Publish 'set' 'read' to 'Bande bureau' failed: 'Error: Read 0x001788010491ffbd/11 genBasic(["deviceEnabled"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUT`

I was able to obeserve the same behavior on all devices hinting at an adapter or config issue. The regular log entry from zigbee-herdsman:deconz:driver is:

`2023-03-28T18:33:35.881Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":241,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4107,"imageType":276,"fileVersion":16784922},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":53703,"endpoint":11,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
`

jonatrey, 2023-03-30:
@petre2dor Can you provide more details on the Adapteur you use, adapter version and Z2M version to see if there is a pattern? Do you also have logs of the failed ping?

petre2dor, 2023-03-30:
Right, of course. Sorry that I did not provide all details in the first comment.

Behavior seems to be identical with what @jonatrey described. 

Hardware: 

- Mosquitto broker reports  37 devices

- Adapter Conbee II 0x26780700

Z2M version: 1.30.2

I don't know how to manually ping the devices, but I tried the following:

1. I unplugged Office: Round Light while it was on. 
2. Then clicked on refresh state:

>  - debug 2023-03-30 20:46:06Received MQTT message on 'zigbee2mqtt/Office: Round Light/get' with data '{"state":""}'
>  - debug 2023-03-30 20:46:06Publishing get 'get' 'state' to 'Office: Round Light'

3. Clicked on OFF (it was stuck on ON):
> - debug 2023-03-30 20:55:06Received MQTT message on 'zigbee2mqtt/Office: Round Light/set' with data '{"state":"OFF"}'
> - debug 2023-03-30 20:55:06Publishing 'set' 'state' to 'Office: Round Light'

dobdib, 2023-04-03:
I've been experiencing the same issue since version 1.30.2, and the problem persists with the current version 1.30.3. I'm using a Conbee II 0x26720700 adapter in conjunction with the Mosquitto broker.

Kubset, 2023-04-04:
Same issue, Zigbee2MQTT version: 1.30.3-1, Conbee II adapter with Mosquitto broker.

petre2dor, 2023-04-06:
@jonatrey Just in case you need help debugging, I am available for a session. If you need the hardware, maybe we can arrange that.

dobdib, 2023-04-23:
Does anyone have any news ?

jonatrey, 2023-04-23:
No update, I tried to repair a few lamps with a sonoff dongle, the availability worked - I assume this is an issue with the Conbee II stick and the adapter needs to be updated / patched.

marioedani, 2023-05-02:
Hello, I have exactly the same issue. I'm using a ConBeeII adapter too (rev. 0x26720700). Zigbee2MQTT is version 1.30.4

liberty29, 2023-06-01:
Hi, same problem here, also with a Conbee II. I am still on Z2M 1.30.1. I do not think this is a problem with de adapter. The problem started with 1.30.2 or 1.30.3 (can't remember which) and _disappeared when I restored 1.30.1_.

Something must have broken in 1.30.2 or 1.30.3 as far as I can see. @Koenkk could you please take a look at the problem reported by the OP?

dobdib, 2023-06-02:
Hi, availability works again with release 1.31.0-1 !

liberty29, 2023-06-03:
Yes, it does! I cannot see it in the release notes, but it is working again. Thanks for the heads-up!

Koenkk, 2023-06-03:
I believe this was fixed by https://github.com/Koenkk/zigbee-herdsman/pull/664

jonatrey, 2023-06-03:
This issue is indeed fixed.

closed 2023-06-03 by jonatrey