Koenkk/zigbee2mqtt #17153: PC321-Z-TY 3-phase Clamp Power Meter #14790 (New issues)

Der-Nax, 2023-03-26:
### What happened?

Hi,

Please look on closed issue PC321-Z-TY 3-phase Clamp Power Meter #14790
There are several messages according to the device.
The issue is that we have trouble with power information on phase 2, 3 and common power.
Please look on the screenshots
[image]
[image]
[image]

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20221226

### Adapter

zStack3x0

harrolf, 2023-04-30:
I have a similar issue as above.

Something goes wrong with the following property readings (/ calculations?):
- `power`, varies between 0 and 429.496.000-ish
- `power_b`, mostly around 39300-ish
- `power_c`, mostly around 39300-ish
- `power_factor_b`, mostly 100
- `power_factor_c`, mostly 100

`power_a`, and `power_factor_a` seems to be good.

This is the MQTT sample:
```
{
    "ac_frequency": 50,
    "current": 1.76,
    "current_a": 0.726,
    "current_b": 0.954,
    "current_c": 0.076,
    "device_status": 0,
    "energy": 1.99,
    "energy_a": 0.784,
    "energy_b": 0.795,
    "energy_c": 0.413,
    "linkquality": 65,
    "power": 15,
    "power_a": 82,
    "power_b": 39275,
    "power_c": 39304,
    "power_factor_a": 48,
    "power_factor_b": 100,
    "power_factor_c": 100,
    "temperature": 37.6,
    "voltage_a": 226.2,
    "voltage_b": 226.4,
    "voltage_c": 226.8
}
```

imcdona, 2023-05-16:
This issue is unrelated to zigee2mqtt and can be closed. You can confirm this by pairing the clamp to a Tuya Zigbee gateway and noting that the power usage is out of whack in the official Tuya app as well.

**If you want to narrow down the issue try this:**
If the _power_ on phase A is out of whack but _power_ on phase B is reporting properly, try swapping the clamps around. You'll note whatever phase is reporting whacky will also report whacky when moved to the other clamp. This rules out the clamp as the source of the issue.

I can only speculate as to what causes the device to report bad power values but it could be internal wiring.🤷‍♂️ To rule this out I plan to watch the power value as I turn off breakers. If after turning off a particular breaker the power value normalizes then at least I'll know where to start looking for a possible wiring problem.

If you're a fan of Tuya devices I recommend picking up one of their Zigbee gateways from AliExpress. They can be purchased for under $10 USD and come in handy for troubleshooting purposes or as an aid when adding new device support to z2m.

Dogmeat88, 2023-05-19:
@Der-Nax @harrolf 
My [original post on the issue](https://github.com/Koenkk/zigbee2mqtt/issues/14790#issuecomment-1399168836) was about exporting power to the grid.
If you do not have a gid-tied solar set up and are still experiencing this issue, then is is possible you just have the CT clamps on backwards. Try reversing the direction the troublesome ones.

closed 2023-06-26 by github-actions[bot]

trozman, 2023-11-25:
> @Der-Nax @harrolf My [original post on the issue](https://github.com/Koenkk/zigbee2mqtt/issues/14790#issuecomment-1399168836) was about exporting power to the grid. If you do not have a gid-tied [...]

Just bumped in this thread because I have the same issue when exporting to grid. I saw there are several stale threads about the same issue. Have you found the solution?
https://github.com/Koenkk/zigbee2mqtt/issues/14790#issuecomment-1826298264

Dogmeat88, 2023-11-25:
Hi @trozman, sorry no solution that I know of. 
I did try @imcdona suggestion and got a Tuya hub and the values are the same (very high when negative (exporting))
So Z2M is passing through exactly what the device is sending.
Baring a firmware update, the only solution I can think of is to run a conversion once it reaches HA, but I have not tried this.

trozman, 2023-11-26:
I'm thinking about the same ugly patch when it reaches HA. Maybe I'll create a template sensor with these corrections: 

if power_b > 10 000 then power_b=power_b-38867
if power > 30 000 then power=power-429425500

I'm trying to determine the exact offsets.
Maybe this kind of conversion - patch could be done even in zigbee2mqtt.

harrolf, 2023-11-28:
Hi @trozman, @Dogmeat88  I'm pretty sure my clamps are in the right direction. 
The only pattern I can see is that if the Power exceeds (let's say 100W) then it (probably) shows the right value for clamp b & c.

The "lowest" high I found was 38004W, so for me it looks like the offset should be around that number. Not sure if that will vary per device.

[image]

trozman, 2023-11-29:
> ooks like the offset s

I'm only experiencing these glitches when exporting and with some trial and error, I found the offset is **~38330**.

My 'patched' sensor in HA looks like this (see below). I don't like it, but at least I get power measures that I can use further (export calculation).

```
#tuya patch, shows negative if exporting
  - sensor:
      - name: "Tuya power b (patch)"
        unique_id: "tuya_power_b_2"
        state: >
          {% if states('sensor.tuyaelectricitymeter_power_b') | float(0) > 5000 %}
                  {{(states('sensor.tuyaelectricitymeter_power_b') | float(0)-39330) | round (2) }}
          {% else %}
                  {{(states ('sensor.tuyaelectricitymeter_power_b') | float(0)) |round(2) }}
          {% endif %}
        availability: >
          {{ states('sensor.tuyaelectricitymeter_power_b') | is_number }}
        unit_of_measurement: "W"
        device_class: power
```

trevhunter, 2024-06-02:
FWIW, I had the exact same issue: The instantaneous power was showing >39200 for 'power_a' and 100% for 'power_factor_a', but Volts, Current, and the Cumulative Energy for A were normal. 

I swapped the direction of the clamp, so that it actually pointed opposite of the direction that the arrow on it indicated that it should be and it seems to fix it. `power_a` is now being reported the same as `voltage_a * current_a * power_factor_a`

I was so sure that the clamp was on the right way that I didn't even try putting it on the other way. But it was the last thing I could think of.... now the clamp is on "backwards", but hey... it works.

GSV3MiaC, 2025-03-01:
Late to the party but mine does it too on the phase feeding my greenhouse. I guarantee there is never any reverse current (well no PV up there anyway), but sometimes when my PW2 is charging the value of power on that line (I am using it for 3 single phase lines) hits 'max' (basically 120A * 240V, which is what the clamps are) BUT:

Current on that phase/line is a reasonable 0.5 A, voltage is stable around 240V, accumulated energy use does NOT leap up in line with the imaginary 26kW power draw (and no, the wires to the greenhouse would not even support 26kW /  120A). Furthermore the total reported for the whole 3 phase meter does not show the same 26kW peak (which hangs around for several minutes sometimes).
Maybe there IS a (small) reverse current instantaneously as the PW2 draws 10kW, and whatever capacitors are in the greenhouse kit feed some back in disgust?

Obviously the firmware is broken - yes I know V*A is not the same as W, but if V*A is 140, there is no way W is 26,000.

atiweb, 2026-01-18:
For those landing here looking for **WiFi** Tuya energy meters (not Zigbee), I wanted to share a working solution for the **PC321-W-TY** WiFi energy meter.

While this issue is specifically about Zigbee devices, the PC321-W-TY is sometimes confused with similar model numbers. If you have the **WiFi version** (Tuya-based), I've documented a complete working integration using LocalTuya:

🔗 **https://github.com/atiweb/PC321-W-TY**

This includes complete DPS mappings, scaling factors, and has been running stable for months.

Just wanted to clarify this in case someone ends up here searching for PC321 energy meter solutions! 😊