Koenkk/zigbee2mqtt #17156: SONOFF Zigbee 3.0 USB Dongle Plus V2 (EFR32MG21) - failed to connect SRSP - SYS - ping after 6000ms - ref.: #13523

xenblog, 2023-03-27:
### What happened?

When adding "adapter: ezsp" to configuration.yaml i get the below error when saving the config:

Failed to save add-on configuration, does not match regular expression ^zstack|deconz|zigate$. Got {'data_path': '/config/zigbee2mqtt', 'external_converters': [], 'devices': 'devices.yaml', 'groups': 'groups.yaml', 'homeassistant': True, 'permit_join': False, 'mqtt': {'base_topic': 'zigbee2mqtt'}, 'serial': {'port': '/dev/ttyACM0', 'adapter': 'ezsp'}, 'advanced': {'log_level': 'warn', 'pan_id': 6754, 'channel': 11, 'network_key': [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13], 'availability_blocklist': [], 'availability_passlist': []}, 'device_options': {}, 'blocklist': [], 'passli...

### How to reproduce it (minimal and precise)

SONOFF Zigbee 3.0 USB Dongle Plus V2 (ZDongle-E)

Fresh install of Zigbee2Mqtt on HA

### Zigbee2MQTT version

1.18.1-1

### Adapter firmware version

20230309.1

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2 (ZDongle-E)

kirovilya, 2023-03-27:
please use latest version 1.30.2

Koenkk, 2023-03-28:
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0