Koenkk/zigbee2mqtt #17161: Identify successful scene recall

Flavien, 2023-03-27:
### What happened?

When using `zigbee2mqtt/<group>/set` with a `{"scene_recall": x}` command, if the scene is already active (and if there is only one light in the group), no message is sent to the the `zigbee2mqtt/<group>` topic.

This happens presumably because the state remains the same, but by not receiving a message to that topic, it makes it impossible (afaik) to know if the `scene_recall` command has succeeded.

### What did you expect to happen?

There should be a robust way to know whether or not a command has succeeded, including for scene recalls.

### How to reproduce it (minimal and precise)

- Recall a scene.
- The `zigbee2mqtt/<group>` topic is updated
- Recall the same scene
- The `zigbee2mqtt/<group>` topic is not updated (even with the same data)

### Zigbee2MQTT version

1.29.2 commit: bb3e8f6

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

junosuarez, 2023-03-28:
This is an interesting topic. In the current design, "scenes" are non-transactional, so there is no easy robust way to verify. What you're describing as watching for the changes in the group topic is essentially watching for side-effects (state changes) of recalling the scene. In the case where there are no state changes (because the scene is already active), there are no new messages published to the group topic. How do we know if it succeeded, then?

Three options occur to me:
1. there is a success message logged in the `zigbee2mqtt/bridge/logging` topic - this is what is shown in the webUI confirmation message. Example:
```
Topic: zigbee2mqtt/bridge/loggingQoS: 0
{"level":"info","message":"Successfully recalled group scene"}
```
Unfortunately, this doesn't have a transaction identifier, so there's no way to be sure that this corresponds to the specific scene recall action you're trying to confirm. In practice, it might be "good enough" if you're not recalling multiple scenes at the same time. In a large house, thought, with multiple scene triggers and many groups, this might not be enough.

2. Trust, but verify - in this option, first send the recall message, then wait some amount of time, then check the group state with a `zigbee2mqtt/<group>/get` message. 

3. Make setting transactional (requires a code change). Basically, you'd add a correlation id to the `/set` message, and listen for a response message. It would probably be cleanest, integration-wise, for zigbee2mqtt to publish this on another topic, for example `<group>/ack` or `<group>/message_status` or something. This would be the most robust option, and would allow the requestor to get either an explicit acknowledgement of success, an explicit error, or no status (in which case the requestor could assume failure and try again), by listening for these messages and matching them up with the id in the original request. It might be possible to implement this in an extension.

junosuarez, 2023-03-28:
Note, I don't think this is a "problem" in the sense the labels are used in this repo - this is a "feature request".

Flavien, 2023-03-30:
So I've looked into the code, and the problem has nothing to do with scenes being "transactional" or not. Like with most commands, individual devices report status changes, and this is how a coordinator can identify that a device has received a command (scene or otherwise).

I've identified the problem, which exists specifically for scenes.
1. Normally, when a member of a group changes state, Zigbee2MQTT publishes the change on the state topic of the group, accordingly.
2. However, to avoid spamming updates, Zigbee2MQTT has a cache for group states. So updates would only be reported if there is a change.
3. However, the cache is invalidated when a `set `command is sent to the group directly ([code here](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/groups.ts#L140)). So if you publish `{ "brightness": 50 }` to the topic, it will publish an update, even if all devices in the group already have brightness set to 50.
4. Unfortunately, this doesn't work for `scene_recall` (hence this issue report). This is because there is a [specific list of attributes](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/groups.ts#L16) that result in this cache invalidation, and `scene_recall` is not part of it.

To work around this, I have written an extension that will simulate a group status update whenever something gets published to the `/set` topic of a group. This way the cache is tricked into getting invalidated. Here is the extension for those interested:

```
const Group = require('../model/group')

const topicRegex = new RegExp(/^(\w+)\/([\w\/]+)\/set$/);

class GroupStateRefresh {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.zigbee = zigbee
        this.mqttBaseTopic = settings.get().mqtt.base_topic;
        this.eventBus = eventBus;
        this.mqtt = mqtt;
        this.logger = logger;
    }

    async start() {
        this.logger.info('[GroupStateRefresh] Extension loaded.');
        this.eventBus.onMQTTMessage(this, this.onMqttMessage.bind(this));
    }

    async onMqttMessage(data) {
        const { topic, message } = data;
        
        const match = topic.match(topicRegex);
        if (!match || match[1] !== this.mqttBaseTopic) {
            return;
        }
        
        const entity = this.zigbee.resolveEntity(match[2]);
        if (entity == null) {
            return;
        }
        
        if (!(entity instanceof Group.default)) {
            return;
        }
        
        this.logger.info(`[GroupStateRefresh] Forcing state refresh for group '${match[2]}'.`);
        
        this.eventBus.emitStateChange({entity, from: {}, to: {}, reason: "forced-group-refresh", update: {"state": null}});
    }

    async stop() {
        this.eventBus.removeListeners(this);
    }
}

module.exports = GroupStateRefresh;
```

closed 2023-05-08 by github-actions[bot]