Koenkk/zigbee2mqtt #17167: Support/Contributors wanted: Matter.js Library to add Matter support

Apollon77, 2023-03-28:
### Is your feature request related to a problem? Please describe

Hello,

I'm one of the developers that currently work on a JavaScript/Typescript implementation of the Matter protocol with the goal to have this as basis to offer to expose devices and also controller functionality. The project is also located under the roof of the official Matter organization: https://github.com/project-chip/matter.js
As I hear this is highly interesting for this project to to expose the zigbee devices maybe as a bridge too or such :-)

We are in the process to finalize anything and are also searching for contributors. There are several things, as easiest implementing the "Clusters" (Matter calls "device features" like onoff/color(level-control that way) or even other topics and functionality.

So if you like to support there to also support zigbee2mqtt I would be happy to see you over there :-)

Ingo

### Describe the solution you'd like

see above

### Describe alternatives you've considered

see above

### Additional context

see above

sjorge, 2023-03-28:
I think at some point Koen at least briefly looked at this https://github.com/Koenkk/zigbee2mqtt/tree/matter

Apollon77, 2023-03-28:
:-) Yes and we want to "prepare the future " to do it better and more easy.

Koenkk, 2023-03-29:
I've pinned this issue for more exposure. It would be great if Matter.js allowed Zigbee2MQTT to acts as a bridge between Matters <-> Zigbee! 😄

epower53, 2023-04-17:
What I'd love is a Matter <-> MQTT bridge. I don't have the coding chops to help with something like this, but it sure would integrate nicely with many setups. Choose your automation engine from there (HA, Node-Red, etc)... all would still use MQTT for device communication, allowing mixed z-wave/zigbee/Matter environments with little effort from the perspective of the casual user (like me).

Apollon77, 2023-04-17:
The challenge there might be that Matter works very much with "commands" and not with directly changing properties. there are easy command like on/off/toggle where you can directly tmap the outcome to properties, but also commands like "dim the ligth within X seconds and Effect Y to level Z" ... so yes you could simply send this command to MQTT ... also fine ...  Other commands might need to return data ... then it could become more interesting.

So yes, can all be possible ... the devil is in the details (like always) ;-)

epower53, 2023-04-17:
> Other commands might need to return data ... then it could become more interesting.
Interesting - I was unaware of this difference. So, devices are not required to send an update with their final state after they complete command execution? This would be surprising...

Apollon77, 2023-04-18:
These are separated things. You can subscribe to attribute changes and there are also events that could be sent out by the device. So yes ideally a command end in 1-n "attribute changes" and potentially events

Apollon77, 2023-05-19:
no stale

sjorge, 2023-05-23:
The newly created https://github.com/Koenkk/zigbee2mqtt-user-extensions is probably a good home to house a WIP matter extension.

Luligu, 2024-06-18:
Hi,
I already developed a matterbridge internal extension fully working and tested, as I have been requested.
It has not been merged cause z2m requires 100 coverage test in a dev!
So I let it go...
If there is new interest we can speak about it... the matterbridge internal code is marked as deprecated but still present.