Koenkk/zigbee2mqtt #17168: AwoX 33957

arteck, 2023-03-28:
### What happened?

https://www.zigbee2mqtt.io/devices/33957.html#awox-33957 

send in LOg 

Publish 'set' 'state' to 'Kellertreppe_Unten' failed: 'Error: Command 0xa4c138655d3d660b/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":30}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":13301,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20221226

### Adapter

sonoff 3.0

Koenkk, 2023-03-29:
Does it also happen with 20221102? https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

arteck, 2023-03-31:
I will check it

Koenkk, 2023-04-01:
It depends on the situation, some get it some don't.

Koenkk, 2023-04-01:
Let's continue in https://github.com/Koenkk/Z-Stack-firmware/issues/439

closed 2023-04-01 by Koenkk