Koenkk/zigbee2mqtt #17181: EM357 driver crashes

Tobster77, 2023-03-30:
### What happened?

I just installed Z2M in Home Assistant some days ago, latest software versions. After a couple of hours and normal operation (switching the plug perfectly works), the Z2M add-on always crashes. Note, there is only the Elelabs EZBUSBA (EM357 chipset) adapter and an Osram Smart Plug+.

Here is a log - can someone help? Thanks a lot!

```
Zigbee2MQTT:info  2023-03-29 22:43:55: Device '0x7cb03eaa00b1fd59' joined
Zigbee2MQTT:info  2023-03-29 22:43:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x7cb03eaa00b1fd59","ieee_address":"0x7cb03eaa00b1fd59"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-03-29 22:43:55: Starting interview of '0x7cb03eaa00b1fd59'
Zigbee2MQTT:info  2023-03-29 22:43:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x7cb03eaa00b1fd59","ieee_address":"0x7cb03eaa00b1fd59","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-03-29 22:44:06: Successfully interviewed '0x7cb03eaa00b1fd59', device has successfully been paired
Zigbee2MQTT:info  2023-03-29 22:44:06: Device '0x7cb03eaa00b1fd59' is supported, identified as: OSRAM Smart+ plug (AB3257001NJ)
Zigbee2MQTT:info  2023-03-29 22:44:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart+ plug","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"AB3257001NJ","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"OSRAM"},"friendly_name":"0x7cb03eaa00b1fd59","ieee_address":"0x7cb03eaa00b1fd59","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2023-03-29 22:44:06: Configuring '0x7cb03eaa00b1fd59'
Zigbee2MQTT:info  2023-03-29 22:44:06: MQTT publish: topic 'homeassistant/switch/0x7cb03eaa00b1fd59/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x7cb03eaa00b1fd59/set","device":{"identifiers":["zigbee2mqtt_0x7cb03eaa00b1fd59"],"manufacturer":"OSRAM","model":"Smart+ plug (AB3257001NJ)","name":"0x7cb03eaa00b1fd59","sw_version":"V1.05.09"},"name":"0x7cb03eaa00b1fd59","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x7cb03eaa00b1fd59","unique_id":"0x7cb03eaa00b1fd59_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Zigbee2MQTT:info  2023-03-29 22:44:06: MQTT publish: topic 'homeassistant/sensor/0x7cb03eaa00b1fd59/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x7cb03eaa00b1fd59"],"manufacturer":"OSRAM","model":"Smart+ plug (AB3257001NJ)","name":"0x7cb03eaa00b1fd59","sw_version":"V1.05.09"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x7cb03eaa00b1fd59 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x7cb03eaa00b1fd59","unique_id":"0x7cb03eaa00b1fd59_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2023-03-29 22:44:06: MQTT publish: topic 'homeassistant/sensor/0x7cb03eaa00b1fd59/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x7cb03eaa00b1fd59"],"manufacturer":"OSRAM","model":"Smart+ plug (AB3257001NJ)","name":"0x7cb03eaa00b1fd59","sw_version":"V1.05.09"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"0x7cb03eaa00b1fd59 update state","state_topic":"zigbee2mqtt/0x7cb03eaa00b1fd59","unique_id":"0x7cb03eaa00b1fd59_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Zigbee2MQTT:info  2023-03-29 22:44:06: MQTT publish: topic 'homeassistant/binary_sensor/0x7cb03eaa00b1fd59/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x7cb03eaa00b1fd59"],"manufacturer":"OSRAM","model":"Smart+ plug (AB3257001NJ)","name":"0x7cb03eaa00b1fd59","sw_version":"V1.05.09"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","name":"0x7cb03eaa00b1fd59 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x7cb03eaa00b1fd59","unique_id":"0x7cb03eaa00b1fd59_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
Zigbee2MQTT:info  2023-03-29 22:44:06: MQTT publish: topic 'homeassistant/update/0x7cb03eaa00b1fd59/update/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/bridge/request/device/ota_update/update","device":{"identifiers":["zigbee2mqtt_0x7cb03eaa00b1fd59"],"manufacturer":"OSRAM","model":"Smart+ plug (AB3257001NJ)","name":"0x7cb03eaa00b1fd59","sw_version":"V1.05.09"},"device_class":"firmware","entity_picture":"https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png","latest_version_template":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/0x7cb03eaa00b1fd59","name":"0x7cb03eaa00b1fd59","payload_install":"{\"id\": \"0x7cb03eaa00b1fd59\"}","state_topic":"zigbee2mqtt/0x7cb03eaa00b1fd59","unique_id":"0x7cb03eaa00b1fd59_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'
Zigbee2MQTT:info  2023-03-29 22:44:06: Successfully configured '0x7cb03eaa00b1fd59'
Zigbee2MQTT:info  2023-03-29 22:44:11: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00b1fd59', payload '{"linkquality":252,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Zigbee2MQTT:info  2023-03-29 22:44:12: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00b1fd59', payload '{"linkquality":253,"state":"OFF","update":{"installed_version":16909577,"latest_version":16909577,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-03-29 22:44:26: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00b1fd59', payload ''
Zigbee2MQTT:info  2023-03-29 22:44:26: MQTT publish: topic 'homeassistant/switch/0x7cb03eaa00b1fd59/switch/config', payload 'null'
Zigbee2MQTT:info  2023-03-29 22:44:26: MQTT publish: topic 'homeassistant/sensor/0x7cb03eaa00b1fd59/linkquality/config', payload 'null'
Zigbee2MQTT:info  2023-03-29 22:44:26: MQTT publish: topic 'homeassistant/sensor/0x7cb03eaa00b1fd59/update_state/config', payload 'null'
Zigbee2MQTT:info  2023-03-29 22:44:26: MQTT publish: topic 'homeassistant/binary_sensor/0x7cb03eaa00b1fd59/update_available/config', payload 'null'
Zigbee2MQTT:info  2023-03-29 22:44:26: MQTT publish: topic 'homeassistant/update/0x7cb03eaa00b1fd59/update/config', payload 'null'
Zigbee2MQTT:info  2023-03-29 22:44:26: MQTT publish: topic 'zigbee2mqtt/Stehlampe', payload '{"linkquality":253,"state":"OFF","update":{"installed_version":16909577,"latest_version":16909577,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-03-29 22:44:27: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/rename', payload '{"data":{"from":"0x7cb03eaa00b1fd59","homeassistant_rename":true,"to":"Stehlampe"},"status":"ok","transaction":"kisve-13"}'
Zigbee2MQTT:info  2023-03-29 22:44:28: MQTT publish: topic 'homeassistant/switch/0x7cb03eaa00b1fd59/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Stehlampe/set","device":{"identifiers":["zigbee2mqtt_0x7cb03eaa00b1fd59"],"manufacturer":"OSRAM","model":"Smart+ plug (AB3257001NJ)","name":"Stehlampe","sw_version":"V1.05.09"},"name":"Stehlampe","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/Stehlampe","unique_id":"0x7cb03eaa00b1fd59_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[... cut ...]
    at listOnTimeout (node:internal/timers:528:9)
    at processTimers (node:internal/timers:502:7)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

### What did you expect to happen?

Continuous operation of the Zigbee network

### How to reproduce it (minimal and precise)

Use the hardware above in Z2M and wait for some hours

### Zigbee2MQTT version

 1.30.2-1

### Adapter firmware version

6.0.0.0 build 212

### Adapter

Elelabs EZBUSBA (EM357 chipset)

MattWestb, 2023-03-30:
And i think Z2M is not having so good support for the old protocol versions like you device is using (likely V4) and is not well tested and badly supported and V8 or newer is recommended (protocol is not the EZSP version its the version of the command set the firmware is using).
ZHA is having better support for old protocols but still you need updating the firmware then its very old and buggy.

MattWestb, 2023-03-30:
I know they was many hacking of the flash programs for getting more devices working and have braking most of them.
You can trying https://github.com/NabuCasa/universal-silabs-flasher that looks working with most devises and is one the way as addon in HA but is not merged.

Tobster77, 2023-04-01:
@MattWestb: Thanks. Sorry for the delays, but I needed two days to get the tool running... Now it works, but EM357 chipset has *.ebl firmware files while the universal-silabs-flasher only accepts *.gbl firmware files.

Any hint?

Tobster77, 2023-04-01:
I tried also the silabs-flasher and the Silicon Labs Simplicity Studio, but all programs whether crash or do not find the Elelabs EZBUSBA. Feels like this is not an EM357 at all, but Zigbee2MQTT finds and uses the stick.

Weird.

MattWestb, 2023-04-01:
Yes i knowing our Puddle have working on the EBL support but i dont know if its in place and i have not testing EBL files then all my EZSP i doing GBL (the new format for EFR32MGX).
You can try getting it in botloader mode and if you is there you can using one terminal program with X-mode support and upload the file after pressing the key for uploading.
Do your device have one bootlooader button inside so you can booting it in bootloader mode with one button pressing ?

I think you largest problem is that you firmware is very old then most EM35X is normally shipped with 6.5.X firmware and you is having 6.0.0.0.
And "type":"EZSP v6"} i think working but not all commands i OK in Z2M and need being V8 for have all command supported.

Your linked firmware is the best we have getting and its Gary that have cooking most of the community firmware that is working best.

Tobster77, 2023-04-02:
I tried my best. The `ncp.py` script reports that the stick is in bootloader mode before it fails during uploading of the firmware:
``` 
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Btl v6.3.12.1 b0

Successfully restarted into bootloader mode! Starting upload of NCP image... 
ERROR:xmodem.XMODEM:send error: expected ACK; got '\x18' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got '\x18' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got '\x18' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got '\r' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got '\n' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'S' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'e' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'r' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'i' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'a' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'l' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got ' ' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'u' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'p' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'l' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'o' for block 2
ERROR:xmodem.XMODEM:send error: expected ACK; got 'a' for block 2
ERROR:xmodem.XMODEM:send error: NAK received 17 times, aborting.
NCP upload failed. Please reload a correct NCP image to recover.
Rebooting NCP...
```
I tried to edit the  `ncp.py` script, so it exits after switching to bootloader moder and I opened `minicom` then, but I receive no bootloader menu on `minicom`.

MattWestb, 2023-04-02:
Then you have the bootloader menu only pressing ` 1 ` one time so its start sending `C` then start the x-modem upload.
The bootlooader com shall being without hard or software flow control.
The firmware shall being one of the god ones.

Tobster77, 2023-04-02:
I did exactly this.

Tobster77, 2023-04-02:
Okay, I did some research where I found comments on the web that while Elelabs EZBUSBA is advertised with EM357 chipset, indeed it has an EM358x chipset when opening the device. I tried this firmware, and now I was able to upload it!!!

Contrarily, Z2M is not able anymore to start as it fails during loading the herdsman driver. Furthermore, I cannot boot to bootloader mode anymore :-(

MattWestb, 2023-04-02:
I hope you  was flashing the EM358x firmware and not the EM357 one and also you must looking with speed its suing the many old firmware can having different baud rate.

Tobster77, 2023-04-02:
I used successfully used NCP_USW_EM3585_6710-LR.

Now I have an EM3585, but the archive only has LR (Long Range) firmwares. Is this okay?

MattWestb, 2023-04-02:
Be sure what chip is in the dongle and the normal is EM3581 but it can being other versions.
The LR is having one extra AP (Amplifier) that is between the chip and the antenna so the model of the IC and then AP for the extra AP its one long range firmware.

If its one EM3581 chip i think this shall working https://github.com/walthowd/husbzb-firmware/blob/master/ncp-uart-sw-6.7.8-115k.ebl or https://github.com/walthowd/husbzb-firmware/blob/master/ncp-uart-sw-6.7.8.ebl is both made of Gary that you is have the other from.

Tobster77, 2023-04-02:
It is definitely EM3585.

I tried to switch to 57k6, but I cannot get to bootloader anymore.

MattWestb, 2023-05-03:
The boot loader shall being intact but the port speed or the app is not OK so the flasher cant sending command to it for rebooting in bootloader mode.
Some devices is having test points for forcing booting the bootloader like in this guide:
https://github.com/walthowd/husbzb-firmware#husbzb-1-firmware-recovery

Also possible testing https://github.com/NabuCasa/universal-silabs-flasher that you can forcing the speed to the firmware is using if not being standard.

Tobster77, 2023-05-12:
@MattWestb: Thank you. Unfortunately, the universal-silabs-flasher does only support *.gbl firmware files. The EM358 only has *.ebl firmware files. I am afraid that this flasher will never support this legacy chip.

closed 2023-06-21 by github-actions[bot]