Koenkk/zigbee2mqtt #17183: Switch Suddently Started Broadcasting Commands to whole network

vaderag, 2023-03-30:
### What happened?

I have an [Osram Switch 4x EU-LIGHTIFY](https://www.zigbee2mqtt.io/devices/4058075816459.html#osram-4058075816459) and it was working just fine, and all of a sudden (unsure of the trigger) it has started broadcasting an off command to my entire zigbee network

Seems there is a related issue https://github.com/Koenkk/zigbee2mqtt/issues/962 (for a different switch)

### What did you expect to happen?

Not to broadcast to the entire network

Assumption this is a binding issue but honestly I'm not experienced enough to know what these options mean. Am posting the screenshot of the bind pages in case someone spots something obviously wrong (to be clear, I haven't touched anything in this page!)

[image]

### How to reproduce it (minimal and precise)

Press a button. Broadcast occurs
Broadcast does not seem to be picked up by home assistant as device state changes are not updated

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 

### Debug log

No relevant logs

vaderag, 2023-04-30:
Not stale

vaderag, 2023-06-01:
Still a thing

vaderag, 2023-07-03:
Still a thing

kkossev, 2023-07-06:
> Still a thing

Have you tried to pair the switch again?

This happens if the switch is reset and then not configured properly during the Zigbee pairing.

TEF2one, 2023-07-10:
Well I have the same issue, I have 4 of the [OSRAM Switch Mini ](https://www.zigbee2mqtt.io/devices/AC0251100NJ_AC0251600NJ_AC0251700NJ.html) and half are now broadcasting to all with the other are fine and I cannot find any difference in their configurations.

TEF2one, 2023-07-10:
Ok, I got one of the remote to work without broadcast by re-pairing to a specific router that is not the coordinator.

closed 2023-08-17 by github-actions[bot]