Koenkk/zigbee2mqtt #17187: Danfoss Ally valve no converter available for pi_heating _demand

Riky-AH, 2023-03-30:
### What happened?

When i tried to set the pi_heating _demand on a Danfoss Ally 014G2461 Zigbee2mqtt reply with "Zigbee2MQTT:error 2023-03-30 15:47:10: No converter available for 'pi_heating_demand' (50)" 

### How to reproduce it (minimal and precise)

-Connect the Danfoss Ally valve to Zigbee2mqtt
-Publish to the valve zigbee2mqtt/[valve friendly name]/set {"pi_heating_demand": 50}

### Zigbee2MQTT version

1.29.1

### Adapter firmware version

{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}

### Adapter

Sonoff Dongle Plus 3.0 ZBDongle-E

sjorge, 2023-03-30:
IIRC from my brief time with this device, this is expected as it does not support setting pi_heating_demand.

Riky-AH, 2023-03-31:
But the Zigbee2MQTT website (https://www.zigbee2mqtt.io/devices/014G2461.html Climate section) say that the pi_heating_demand is supported with both /get and /set, It's possible that it will work with a custom converter or this parameter was put on the website by error and there is no way to use it?

sjorge, 2023-03-31:
@Koenkk could that code generating it be wrong? https://www.zigbee2mqtt.io/devices/H1.html also lists it under the climate section as settable and it 100% for sure isn't on that one.

@Riky-AH you could try via de dev console
[image]

But I think it might be an error in the docs that get generated.

Riky-AH, 2023-03-31:
I have tried with the dev console, i can read it but if i try to write i get an error who say that this parameter is read-only

[image]

And also in the Values section i can see the pi_heating_demand value but there is nothing to change it.

sjorge, 2023-03-31:
So the READ_ONLY is the device saying you can only read the attribute not write to it. So It's as I remember where the devices does not support writing to it. I'm pretty sure the code that generates the device page has a bug somewhere then, as the H1 also is listed wrong on there.

closed 2023-05-10 by github-actions[bot]

rodolpheAV, 2025-01-31:
Hello

I know this is a very old fix but it seems to me that the fix was not correct. instead of changing the doc to say that the property are read only, it now says that the property is write only which is wrong : 

[image]

So this devices doc is wrong and I think many others too.