Koenkk/zigbee2mqtt #17190: [New device support]: Moes smoke detector TS0601 _TZE200_ytibqbra

Yakie996, 2023-03-30:
### Link

https://www.aliexpress.com/item/1005004279342985.html?spm=a2g0o.order_list.order_list_main.27.303f1802QKcU78

### Database entry

{"id":40,"type":"EndDevice","ieeeAddr":"0xa4c1387a919fb276","nwkAddr":57088,"manufId":4417,"manufName":"_TZE200_ytibqbra","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1680193394293,"defaultSendRequestWhen":"immediate"}

### Comments

I would like to learn how to create this new device for Z2M, I am new to home assistant and have very little experience with coding. If someine can help me creating this I would be very greatful.

Kind regards
Yarni

Yakie996, 2023-03-30:
Hi Koenkk,

Thanks that worked! Later I will try to add some exposes so I can link them togheter as a alarm goes off and test the alarm.
Great thanks for helping me!

Koenkk, 2023-03-30:
Added, you can remove the external converter after updating to the next release.

closed 2023-03-30 by Koenkk

Yakie996, 2023-03-31:
This is what I can see on Tuya IOT

[image]
[image]

Koenkk, 2023-04-01:
I need to have the datapoints for that: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

Yakie996, 2023-04-01:
Great!

Alarm switch = 17
Silence = 16
Detection State = 1
Fault alarm = 11
Battery level = 15

Yakie996, 2023-04-01:
I got the alarm_switch working 
silence does not seem to work, when the alarm goes off this changes nothing when enabling.
battery level and tamper doesnt work yet  

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ntcy3xu1', '_TZE200_ytibqbra']),
    model: 'TS0601_smoke_1',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.smoke(), e.tamper(), e.battery_low(), tuya.exposes.silence(), tuya.exposes.faultAlarm(),exposes.binary('alarm_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable the alarm'),],
    meta: {
        tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.trueFalse0],
            [4, 'tamper', tuya.valueConverter.raw],
            [14, 'battery_low', tuya.valueConverter.trueFalse0],
            [16, 'silence', tuya.valueConverter.raw],
            [11, 'fault_alarm', tuya.valueConverter.trueFalse1],
            [17, 'alarm_switch', tuya.valueConverter.onOff]
        ],
    },
};

module.exports = definition;
[image]
`

Koenkk, 2023-04-02:
For battery level, can you check with `[15, 'battery_level', tuya.valueConverter.raw]` and see if it appears under the "State" (from screenshot above)

Yakie996, 2023-04-05:
Does not work z2m crashes

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ntcy3xu1', '_TZE200_ytibqbra']),
    model: 'TS0601_smoke_1',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.smoke(), e.tamper(), tuya.battery_level(), tuya.exposes.silence(), tuya.exposes.faultAlarm(),exposes.binary('alarm_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable the alarm'),],
    meta: {
        tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.trueFalse0],
            [4, 'tamper', tuya.valueConverter.raw],
            [15, 'battery_level', tuya.valueConverter.raw]
            [16, 'silence', tuya.valueConverter.raw],
            [11, 'fault_alarm', tuya.valueConverter.trueFalse1],
            [17, 'alarm_switch', tuya.valueConverter.onOff]
        ],
    },
};

module.exports = definition;`

Koenkk, 2023-04-05:
Can you provide the debug log of the crash?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Yakie996, 2023-04-14:
```
[11:53:50] INFO: Preparing to start...
[11:53:51] INFO: Socat not enabled
[11:53:52] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:21
exposes: [e.smoke(), e.tamper(), tuya.battery_level(), tuya.exposes.silence(), tuya.exposes.faultAlarm(),exposes.binary('alarm_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable the alarm'),],
                                      ^
TypeError: tuya.battery_level is not a function
    at /app/data/extension/externally-loaded.js:21:39
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

this is the log

Koenkk, 2023-04-14:
- `[15, 'battery_level', tuya.valueConverter.raw]` should be `[15, 'battery', tuya.valueConverter.raw]`
- `tuya.battery_level()` should be `e.battery()`

Yakie996, 2023-04-15:
that worked, thank you. This is the complete code that works for me.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ntcy3xu1', '_TZE200_ytibqbra']),
    model: 'TS0601_smoke_1',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.smoke(), e.tamper(), e.battery(), tuya.exposes.faultAlarm(),  tuya.exposes.silence(),exposes.binary('alarm_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable the alarm')],
    meta: {
        tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.trueFalse0],
            [4, 'tamper', tuya.valueConverter.raw],
            [15, 'battery', tuya.valueConverter.raw],
            [16, 'silence', tuya.valueConverter.raw],
            [17, 'alarm_switch', tuya.valueConverter.onOff],
            [11, 'fault_alarm', tuya.valueConverter.trueFalse1],
        ],
    },
};

module.exports = definition;

```
[image]

Koenkk, 2023-04-16:
Thanks, what does `alarm_switch`/`silence` do? I'm confused what happens if you e.g. turn both ON

huythach83, 2023-04-23:
> Thanks, what does `alarm_switch`/`silence` do? I'm confused what happens if you e.g. turn both ON

Alarm_switch: just like the "Test" button on the device. You turn it on and after a few seconds, the device will act as if it detects smoke.

Silence: when the alarm is active, turn this to on and it will silence the alarm

Koenkk, 2023-04-23:
@huythach83 does silence and alarm switch automatically return to false after some time?

huythach83, 2023-04-24:
> @huythach83 does silence and alarm switch automatically return to false after some time?

Yes it does

Koenkk, 2023-04-24:
Great, thanks!