Koenkk/zigbee2mqtt #17194: [Feature request]: add hysteresis settings for SMaBit AV2010/32 Wall-Thermostate

aw1604, 2023-03-31:
### Is your feature request related to a problem? Please describe

I am using a SMaBit AV2010/32 Wall-mounted Thermostate which worked right out of the box with z2m. However, this thermostate has a rather wide hysteresis setting by default (+/- 0.25 C). I would like to be able to set (and read/verify) the hysteresis_high and hysteresis_low values via z2m to avoid big temperature changes.

### Describe the solution you'd like

device settings (specific) for hysteresis_high and hysteresis_low in the z2m frontend for the device, like temperature_calibration (which is already supported)

### Describe alternatives you've considered

It seems like the HA / HACS ZHA-Toolkit is able to manage these settings, but for this I would need to convert from z2m to HA-ZHA (Which I had used in the past before I found z2m and I'm quite happy with it !)

### Additional context

On this page https://community.home-assistant.io/t/need-help-advice-for-hardware-wall-thermostat-heater-thermostats/149771/7 are zha-toolkit "set" and "retrieve" scripts which may contain useful attribute / attribute type for this device.

Device : https://www.zigbee2mqtt.io/devices/AV2010_32.html

sjorge, 2023-03-31:
At a quick glance at the linke ha post, it seems this is available as a attribute on the hvacThermostst cluster

I'll have a closes look over the weekend.

sjorge, 2023-03-31:
Interesting, 4-nocks seems to be the manufacturer of the Bitron Thermostats
https://www.4-noks.com/shop/smart-automation/termostato-touch-ttr2/

sjorge, 2023-03-31:
Will implement something easier to use once the herdsman PR gets merge, but if you can't wait you can manually change it using the write convertor for now.

```
 mosquitto_pub -t zigbee2mqtt/thermostat/home/set -m '{"write":{"cluster":"hvacThermostat","options":{"manufacturerCode": 4209},"payload":{"257": {"value": 10, "type": 33}, "258": {"value": 10, "type": 33}}}}'
```

And verify the new value

```
mosquitto_pub -t zigbee2mqtt/thermostat/home/set -m '{"read":{"attributes":[257, 258],"cluster":"hvacThermostat","options":{"manufacturerCode": 4209}}}'
```

aw1604, 2023-04-01:
Thank you so much ! Setting (and verify) new hysteresis values using the mosquitto_pub commands worked !

closed 2023-04-01 by aw1604

sjorge, 2023-04-01:
I have a PR open to make it a bit easier.

```
root@amethyst:~# mosquitto_pub -t zigbee2mqtt/thermostat/home/get/hysteresis_low -m ''
root@amethyst:~# mosquitto_pub -t zigbee2mqtt/thermostat/home/get/hysteresis_high -m ''
root@amethyst:~# mosquitto_pub -t zigbee2mqtt/thermostat/home/set/hysteresis_low -m '25'
root@amethyst:~# mosquitto_pub -t zigbee2mqtt/thermostat/home/set/hysteresis_high -m '25'
```

I choice not to add this to the exposes payload as this doesn't seem like something the average user wants to mess with or will change it often. It's more of a set once and forget type deal.

mattiabrunello5, 2024-07-25:
hi i have an astrel (ZED-TTR2-RI)  (https://products.astrelgroup.com/en/product/rialto-smart-thermostat/), with seem to be the same as yours, I try to integrate in z2m but in the dash board show unsupported ( I try to use an external component ) but doesn't work can you help me ?