Koenkk/zigbee2mqtt #17195: Holding volume buttons on IKEA Symfonisk E2123 remote sometimes results in delayed/outdated events

mcuelenaere, 2023-03-31:
### What happened?

When pressing (or holding) the volume up/down buttons on the remote, you can observe that previous (outdated) events are sometimes mixed-in with current ones.

The easiest to reproduce this is by holding eg volume up for some time and then switching to pressing the volume down button (a few times).

At first I thought this was an issue in HA, but when checking the raw MQTT feed it is visible there as well + you can see it in the logs too.

Unsure if this issue is caused by the actual remote or some kind of buffering in zigbee2mqtt/zigbee coordinator stack.

### What did you expect to happen?

The order of events should be correct and no outdated events should be delivered when they no longer make sense.

### How to reproduce it (minimal and precise)

Pair IKEA Symfonisk Gen2 remote (E2123) with zigbee2mqtt, hold the volume up button for some time and then switch to pressing the volume down button (a few times).

You'll see that instead of the expected `[volume_up_hold, volume_up_hold, volume_up_hold, volume_down, volume_down]` stream. you might get something like `[volume_up_hold, volume_up_hold, volume_up_hold, volume_down, volume_up_hold, volume_up_hold, volume_down]`.

### Zigbee2MQTT version

1.30.2-dev commit: 1368a78

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

Attempt 1, holding volume up followed by volume down presses:
```
zigbee2mqtt  | Zigbee2MQTT:debug 2023-03-31 12:25:48: Received Zigbee message from 'douche-sonos-remote', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":255}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: Configuring 'douche-sonos-remote'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"volume_up_hold","battery":200,"linkquality":58,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"","battery":200,"linkquality":58,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote/action', payload 'volume_up_hold'
zigbee2mqtt  | Zigbee2MQTT:debug 2023-03-31 12:25:48: Received Zigbee message from 'douche-sonos-remote', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":255}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"volume_up_hold","battery":200,"linkquality":65,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"","battery":200,"linkquality":65,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote/action', payload 'volume_up_hold'
zigbee2mqtt  | Zigbee2MQTT:debug 2023-03-31 12:25:48: Received Zigbee message from 'douche-sonos-remote', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":255}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"volume_up_hold","battery":200,"linkquality":65,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"","battery":200,"linkquality":65,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:48: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote/action', payload 'volume_up_hold'
zigbee2mqtt  | Zigbee2MQTT:debug 2023-03-31 12:25:49: Received Zigbee message from 'douche-sonos-remote', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":255}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:49: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"volume_up_hold","battery":200,"linkquality":65,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:49: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"","battery":200,"linkquality":65,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:49: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote/action', payload 'volume_up_hold'
zigbee2mqtt  | Zigbee2MQTT:debug 2023-03-31 12:25:49: Received Zigbee message from 'douche-sonos-remote', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":255}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:49: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"volume_up_hold","battery":200,"linkquality":61,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:49: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"","battery":200,"linkquality":61,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:49: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote/action', payload 'volume_up_hold'
zigbee2mqtt  | Zigbee2MQTT:debug 2023-03-31 12:25:50: Received Zigbee message from 'douche-sonos-remote', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":0,"rate":255}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:50: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"volume_up_hold","battery":200,"linkquality":76,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:50: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"","battery":200,"linkquality":76,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:50: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote/action', payload 'volume_up_hold'
zigbee2mqtt  | Zigbee2MQTT:debug 2023-03-31 12:25:50: Received Zigbee message from 'douche-sonos-remote', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":1,"rate":255}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:50: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"volume_down","battery":200,"linkquality":94,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:50: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"","battery":200,"linkquality":94,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:50: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote/action', payload 'volume_down'
zigbee2mqtt  | Zigbee2MQTT:debug 2023-03-31 12:25:51: Received Zigbee message from 'douche-sonos-remote', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":1,"rate":255}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:25:52: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"volume_down","battery":200,"linkquality":109,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
[... cut ...]
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:26:38: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"volume_up","battery":200,"linkquality":98,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:26:38: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote', payload '{"action":"","battery":200,"linkquality":98,"update":{"installed_version":16777266,"latest_version":16777266,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-03-31 12:26:38: MQTT publish: topic 'zigbee2mqtt/douche-sonos-remote/action', payload 'volume_up'
```

mcuelenaere, 2023-03-31:
CC'ing @ilueckel to see if they suffer from this as well or not (given that they contributed https://github.com/Koenkk/zigbee-herdsman-converters/pull/5583)

dof250, 2023-04-06:
For me it's not only on holding. Whenever I press volume_up a few times the commands are issued instantly, but then after 1-2 seconds it executes it anoter 1-2 times.

ilueckel, 2023-04-07:
Sadly I can not recreate the behavior with my remote.

mcuelenaere, 2023-04-07:
> For me it's not only on holding. Whenever I press volume_up a few times the commands are issued instantly, but then after 1-2 seconds it executes it anoter 1-2 times.

I have that as well, but the holding down is easier to reproduce.

junosuarez, 2023-04-11:
In the [zigbee message payload we get back from the device](https://github.com/Koenkk/zigbee2mqtt/issues/16808#issuecomment-1470973890), it includes a counter so we know the order of the button press events. In theory, it would be possible to fix this issue in the zigbee-herdsman-converter by ensuring that we only process a button action when the counter is higher than the last one seen (modulo 1 byte).

closed 2023-05-19 by github-actions[bot]

kumme74, 2023-06-24:
I can reproduce the behavior. Having here a E2123 with stock 1.0.012 and one I updated to 1.0.35.
The remote with 1.0.35 sends out the delayed messages for the periodic volume_up_hold and volume_down_hold messages.

I think that the counters @junosuarez mentioned are only implemented for the dot buttons or did I miss something.
The current behavior limits the volume up and down buttons as only the single click events can be evaluated.

Is it possible that I configured something wrong or does someone know if it is possible to downgrade the device to stock firmware?

ilueckel, 2023-12-10:
One of my remotes is now also affected by this error. Here is my herdsman debug log

```
2023-12-10T12:18:13.918Z zigbee-herdsman:adapter:ezsp:uart <-- [4310b1a97d312a15b658]
2023-12-10T12:18:13.919Z zigbee-herdsman:adapter:ezsp:uart <-- [9c4a24ab1593499c03d817e1919874f8d7528cfd8154b97e]
2023-12-10T12:18:13.919Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,3,0): 4310b1a9112a15b6589c4a24ab1593499c03d817e1919874f8d7528cfd8154b97e
2023-12-10T12:18:13.919Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2023-12-10T12:18:13.919Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2023-12-10T12:18:13.920Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 529001450000040108000101400100004dffbc0c5fffff0511310501ff
2023-12-10T12:18:13.920Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":77},"lastHopLqi":255,"lastHopRssi":-68,"sender":24332,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[17,49,5,1,255]}}
2023-12-10T12:18:13.921Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":77},"lqi":255,"rssi":-68,"sender":24332,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[17,49,5,1,255]}}
2023-12-10T12:18:13.924Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":49,"manufacturerCode":null,"commandIdentifier":5},"Payload":{"movemode":1,"rate":255},"Command":{"ID":5,"parameters":[{"name":"movemode","type":32},{"name":"rate","type":32}],"name":"moveWithOnOff"}},"address":24332,"endpoint":1,"linkquality":255,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2023-12-10 13:18:13: Received Zigbee message from 'Sonos Fernbedienung Wohnzimmer', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":1,"rate":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-12-10 13:18:13: MQTT publish: topic 'zigbee2mqtt/Sonos Fernbedienung Wohnzimmer', payload '{"action":"volume_down","battery":100,"last_seen":"2023-12-10T12:18:13.924Z","linkquality":255,"update":{"installed_version":16777269,"latest_version":16777269,"state":"idle"},"update_available":null,"voltage":2800}'
Zigbee2MQTT:info  2023-12-10 13:18:13: MQTT publish: topic 'zigbee2mqtt/Sonos Fernbedienung Wohnzimmer', payload '{"action":"","battery":100,"last_seen":"2023-12-10T12:18:13.924Z","linkquality":255,"update":{"installed_version":16777269,"latest_version":16777269,"state":"idle"},"update_available":null,"voltage":2800}'
Zigbee2MQTT:info  2023-12-10 13:18:13: MQTT publish: topic 'zigbee2mqtt/Sonos Fernbedienung Wohnzimmer/action', payload 'volume_down'
2023-12-10T12:18:16.487Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2023-12-10T12:18:16.487Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2023-12-10T12:18:16.488Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2023-12-10T12:18:16.488Z zigbee-herdsman:adapter:ezsp:uart --> DATA (3,5,0): 5300010500
2023-12-10T12:18:16.488Z zigbee-herdsman:adapter:ezsp:uart --> [357d3121a9512a16c77e]
2023-12-10T12:18:16.489Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (4)
2023-12-10T12:18:16.490Z zigbee-herdsman:adapter:ezsp:uart <-- [547d31a1a9512a]
2023-12-10T12:18:16.491Z zigbee-herdsman:adapter:ezsp:uart <-- [d1477e]
2023-12-10T12:18:16.491Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,4,0): 5411a1a9512ad1477e
2023-12-10T12:18:16.492Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (6)
2023-12-10T12:18:16.492Z zigbee-herdsman:adapter:ezsp:uart --> [8610be7e]
2023-12-10T12:18:16.494Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 5380010500
2023-12-10T12:18:16.495Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
2023-12-10T12:18:16.496Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (4) success
2023-12-10T12:18:20.720Z zigbee-herdsman:adapter:ezsp:uart <-- [647d31b1a9]
2023-12-10T12:18:20.721Z zigbee-herdsman:adapter:ezsp:uart <-- [7d312a15b6589c4a24ab1593499c03]
2023-12-10T12:18:20.721Z zigbee-herdsman:adapter:ezsp:uart <-- [d86fe1919874f8d7528cfd81b1df7e]
2023-12-10T12:18:20.722Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,4,0): 6411b1a9112a15b6589c4a24ab1593499c03d86fe1919874f8d7528cfd81b1df7e
2023-12-10T12:18:20.722Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2023-12-10T12:18:20.722Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2023-12-10T12:18:20.723Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 539001450000040108000101400100004dffc40c5fffff0511310501ff
2023-12-10T12:18:20.723Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":77},"lastHopLqi":255,"lastHopRssi":-68,"sender":24332,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[17,49,5,1,255]}}
2023-12-10T12:18:20.724Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"seq [... cut ...]
Zigbee2MQTT:info  2023-12-10 13:18:27: MQTT publish: topic 'zigbee2mqtt/Sonos Fernbedienung Wohnzimmer/action', payload 'volume_down'
```

This log shows three actions zigbee2mqtt publishes, although the button is only pressed one time. As seen from the logs, the coordinator receives the message three times - so I guess this an bug from the remote? In contrast to the description before, my error also triggers on simple press.

My finding so far is that the `transactionSequenceNumber` doesn't change, therefore implementing the `hasAlreadyProcessedMessage` in the converter should be enough.