Koenkk/zigbee2mqtt #17196: Iluminize 511.012 does not report back state

brushmate, 2023-03-31:
### What happened?

The Iluminize 511.012 ZigBee controller supports an external switch to control the state. Unfortunately these state changes do not show up in Zigbee2MQTT.

### What did you expect to happen?

I would have expected to see the changes in Zigbee2MQTT / Home Assistant.

### How to reproduce it (minimal and precise)

Setup the LED controller and attach an external switch. Press the switch and see that nothing is reported back to Zigbee2MQTT.

I found  #2502 where someone reported a similar issue. @Koenkk [asked to add an external converter](https://github.com/Koenkk/zigbee2mqtt/issues/2502#issuecomment-949834121) for further debugging. I tried to adapt that, but then Zigbee2MQTT fails to start with the following stacktrace:

```
/app/node_modules/zigbee-herdsman-converters/index.js:65
            assert.fail(`'${definition.model}' has configure in extend and device, this is not allowed`);
                   ^
AssertionError [ERR_ASSERTION]: '511.012' has configure in extend and device, this is not allowed
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:65:20)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
```

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20210708

### Adapter

cod.m ZigBee CC2652P2 Raspberry Pi Module

brushmate, 2023-03-31:
I just got feedback from Iluminize. According to them this is known behavior and smart home hubs should actively poll the state of the controller. Is there such a functionality in Zigbee2MQTT and can it be enabled for this controller?

brushmate, 2023-04-01:
I found the corresponding setting in the Zigbee2MQTT frontend. You can find it under _Devices_ > _Your device_ > _Reporting_. I have configured two reportings for this controller:

```
Endpoint         : 1
Cluster          : LevelCtrl
Attribute        : currentLevel
Min rep interval : 1
max rep interval : 5
Min rep change   : 0
```
```
Endpoint         : 1
Cluster          : OnOff
Attribute        : onOff
Min rep interval : 1
max rep interval : 5
Min rep change   : 0
```

I think I also found the related documentation of this feature: https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-device-configure-reporting

Since this is not a Zigbee2MQTT issue, I am closing it. I will try to convince the vendor to implement 'native reporting'.

closed 2023-04-01 by brushmate

gregoriusus, 2024-11-25:
@brushmate Did they give you any feedback? I have sam issue and thinking to buy another controller.

brushmate, 2024-11-26:
No, I don't think so. But to be fair, I don't remember if I contacted them again. At least I can't find anything in my email history.

gregoriusus, 2025-07-23:
If I understand correctly, there is still a hope that state is reported back?