Koenkk/zigbee2mqtt #17198: [New device support]: Vimar 30805 Dimmer Switch

Sim0cYz, 2023-03-31:
### Link

https://www.vimar.com/it/it/catalog/product/index/liv/L1100?page=7

### Database entry

0xf4ce3654af33eb93 (0x88D6)

### Comments

I tried to follow the guide to support new devices, but I can't control the device (can't get status and can't switch on/off, or dim the light)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['DimmerSwitch_v1.0'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '30805', // Vendor model number, look on the device for a model number
    vendor: 'Vimar', // Vendor of the device (only used for documentation and startup logging)
    description: 'Connected Dimmer 220-240V', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: extend.light_onoff_brightness(),
};

module.exports = definition;
```

Sim0cYz, 2023-05-01:
Still unable to make it work

closed 2023-06-09 by github-actions[bot]

dryuk94, 2023-12-11:
This issue was resolved here: https://github.com/Koenkk/zigbee2mqtt/issues/18323

use this configuration:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['DimmerSwitch_v1.0'],
    model: '30805',
    vendor: 'Vimar',
    description: 'IoT connected dimmer mechanism 220-240V',
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    endpoint: (device) => {return {default: 11}},
    exposes: [e.light_brightness()],
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(11);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'genIdentify']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;

Koenkk, 2023-12-12:
`DimmerSwitch_v1.0` is already supported in z2m 1.34.0