Koenkk/zigbee2mqtt #17199: Vimar 14594 (roller shutter) not updating status

Sim0cYz, 2023-03-31:
### What happened?

Hello!
I paired my 7 devices and I can control them, but they aren't reporting current status if shutters are moved with physical buttons. If I go in "Exposes" page of each device and refresh, current status is successfully updated. 

https://www.zigbee2mqtt.io/devices/14594.html#vimar-14594

### What did you expect to happen?

Status of the roller shutter updated if opened or closed with physical buttons on the device.

### How to reproduce it (minimal and precise)

If roller shutter is closed, open with the physical button. Check back in Home Assistant, it's still reported closed (even hours later).

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle E

Sim0cYz, 2023-05-01:
Still not working as expected

closed 2023-06-09 by github-actions[bot]

datsrt02, 2023-08-09:
U should add clushter for reporting in setting of devices, choose Closure/currentPositionLiftPercentage and TiltPercentage. Then type the min interval and max is time delay u want to it will send report

danielescali73, 2023-08-15:
Vimar 14594 Still not working as expected.

datsrt02, 2023-08-22:
> > U should add cluster for reporting in setting of devices, choose Closure/currentPositionLiftPercentage and TiltPercentage. Then type the min interval and max is time delay u want to it will send report
> 
> @datsrt02 Can you explain better?

[image]

Please take a look this image, in setting of device, select tab "Reporting" then add two clusters as that image then apply, restart z2m and try test your vimar device

hexstudy, 2023-08-22:
I have tried with at one and work ok from device and from z2m
Thanks

danielescali73, 2023-08-22:
it works for me too. good job.
What will happen now? will someone release a version with the fix?