Koenkk/zigbee2mqtt #17205: Nimly PRO /  Onesti / easyCodeTouch_v1 - Not fully supported - How to improve?

TheQue42, 2023-04-01:
### What happened?

Hi, 

I've got a nimly pro lock, which according to https://github.com/Koenkk/zigbee-herdsman-converters/pull/4892 is supposed to be supported by z2m.

But I can only view/perform lock/unlock, control keypad volume and autolock.  But according to the nimly support, the zigbee connection should provide the information about which user unlocked the lock as well, so that I can know if the lock was unlocked by keypad/fingerprint, and which userid.
So this information should be possible to handle in z2m as well I assume, providing we can decode the output from the lock?

So what do I need to provide? I have here the output debug output from herdsman, where I unlock first with fingerprint, close the door and let it lock, and then unlock again, with an rfid tag.

### What did you expect to happen?

Information in the   "action": null, "action_source_name": null, or  "action_user": null, indicating WHO/HOW the lock was unlocked.

### How to reproduce it (minimal and precise)

Just unlock the lock with fingerprint/pin/rfid.

### Zigbee2MQTT version

1.30.3 commit: 24c6b2e

### Adapter firmware version

    20221226

### Adapter

sonoff, zStack3x0

### Debug log

Zigbee2MQTT:info  2023-04-01 10:09:24: MQTT publish: topic 'zigbee2mqtt/Nimly Doorlock', payload '{"action":null,"action_source_name":null,"action_user":null,"auto_relock":true,"auto_relock_time":1,"battery":100,"last_seen":"2023-04-01T08:09:24.877Z","linkquality":94,"lock_state":"locked","pin_code":null,"sound_volume":"silent_mode","state":"LOCK"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +13s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,11,205] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,11,205] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,11] - 205 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":11} +30ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,1,1,26,169,11,1,0,98,0,217,148,108,0,0,7,8,81,10,0,0,48,1,143,132,28,86] +81ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,1,1,26,169,11,1,0,98,0,217,148,108,0,0,7,8,81,10,0,0,48,1,143,132,28,86] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,1,1,26,169,11,1,0,98,0,217,148,108,0,0,7,8,81,10,0,0,48,1,143,132,28] - 86 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":257,"srcaddr":43290,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":98,"securityuse":0,"timestamp":7115993,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,81,10,0,0,48,1]}} +82ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":81,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":48,"attrData":1}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":43290,"endpoint":11,"linkquality":98,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}' +109ms
  zigbee-herdsman:controller:endpoint DefaultResponse 0xf4ce3683a2ad1ec0/11 257(10, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +108ms
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xf4ce3683a2ad1ec0:43290/11 (0,0,1) +108ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":43290,"destendpoint":11,"srcendpoint":1,"clusterid":257,"transid":12,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,81,11,10,0]}} +109ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,26,169,11,1,1,1,12,0,30,5,24,81,11,10,0,204] +109ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +14ms
Zigbee2MQTT:info  2023-04-01 10:09:24: MQTT publish: topic 'zigbee2mqtt/Nimly Doorlock', payload '{"action":null,"action_source_name":null,"action_user":null,"auto_relock":true,"auto_relock_time":1,"battery":100,"last_seen":"2023-04-01T08:09:24.985Z","linkquality":98,"lock_state":"locked","pin_code":null,"sound_volume":"silent_mode","state":"LOCK"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +108ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,12,202] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,12,202] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,12] - 202 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":12} +26ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,30,68,129,0,0,1,1,26,169,11,1,0,94,0,190,199,108,0,0,10,8,82,10,0,1,27,0,0,1,10,143,132,28,117] +181ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,30,68,129,0,0,1,1,26,169,11,1,0,94,0,190,199,108,0,0,10,8,82,10,0,1,27,0,0,1,10,143,132,28,117] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 30 - 2 - 4 - 129 - [0,0,1,1,26,169,11,1,0,94,0,190,199,108,0,0,10,8,82,10,0,1,27,0,0,1,10,143,132,28] - 117 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":257,"srcaddr":43290,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":7129022,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[8,82,10,0,1,27,0,0,1,10]}} +184ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":82,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":256,"dataType":27,"attrData":167837696}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":43290,"endpoint":11,"linkquality":94,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}' +210ms
[... 178 lines cut ...]
  zigbee-herdsman:controller:endpoint DefaultResponse 0xa4c1384403103ff1/1 0(10, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +6s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xa4c1384403103ff1:39180/1 (0,0,1) +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":39180,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":20,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,176,11,10,0]}} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,12,153,1,1,0,0,20,0,30,5,24,176,11,10,0,25] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +12ms
Zigbee2MQTT:info  2023-04-01 10:09:55: MQTT publish: topic 'zigbee2mqtt/ElementFläkt Sovrum', payload '{"child_lock":"UNLOCK","current":0,"energy":4.11,"indicator_mode":"off/on","last_seen":"2023-04-01T08:09:55.869Z","linkquality":83,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":237}'
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,242,116,1,1,0,94,0,79,104,139,0,0,15,8,90,10,1,0,32,192,226,255,32,54,228,255,32,0,242,116,29] - 143 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":29938,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":9136207,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,90,10,1,0,32,192,226,255,32,54,228,255,32,0]}} +5ms

TheQue42, 2023-04-01:
I've shortend the log a bit, since it was too large. Can of course provide a larger long if needed.

TheQue42, 2023-05-02:
Yes, its still an issue.

jkaberg, 2023-05-29:
Shouldn't this be filed over at https://github.com/Koenkk/zigbee-herdsman-converter? Anyhow voicing my interest as I have two of these.

jkaberg, 2023-06-01:
So just looking briefly into this, comparing [Yale](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/yale.js) with [Onesti](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/onesti.js) - there seems to be missing some parts? I'm assuming something like [this](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/yale.js#L30) is needed (operation_event)?

Perhaps @Koenkk has time to shed some details on the inner working of how this is supposed to work?

jkaberg, 2023-06-01:
Looking over the code again I notice that on https://github.com/Koenkk/zigbee-herdsman-converters/blob/53455e815702e2465c8040022c3e267965d3d8b7/devices/onesti.js#L15 we have 
`fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery, fz.lock_programming_event, fz.easycodetouch_action],`
The same `fromZigbee` for [Yale](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/yale.js#L9) contains
`fromZigbee: [fz.lock, fz.battery, fz.lock_operation_event, fz.lock_programming_event, fz.lock_pin_code_response, fz.lock_user_status_response],`

Perhaps all we need to add is `fz.lock_user_status_response`, an oversight from previous implementation?

Also on https://github.com/Koenkk/zigbee-herdsman-converters/blob/53455e815702e2465c8040022c3e267965d3d8b7/devices/onesti.js#L16 we might be missing `tz.lock_userstatus` (again ref Yale code) - but perhaps not needed?

jkaberg, 2023-06-03:
I tried my suggestion, but no dice so then I'm out of ideas?

abualy, 2023-06-06:
same issue here, i managed for example to set a new pin code using the docs https://www.zigbee2mqtt.io/devices/easyCodeTouch_v1.html#pin-code-composite and communication directly through my mqtt gateway, but i cannot see the pin_code when following the docs
also seeing and setting the pin_code for example is not working on home_assistant , with the `homeassistant: true` config on my zigbee2mqtt pod

heinekmadsen, 2023-06-07:
This also have my interest
On the z2m devices page for Heimgard lock is need a "expose_pin" setting. https://www.zigbee2mqtt.io/devices/HT-SLM-2.html

jkaberg, 2023-06-09:
Seems like @eirikd did the groundwork in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2791 - do you have any thoughts as to the above?

jkaberg, 2023-06-09:
Zigbee protocol documentation: https://github.com/Koenkk/zigbee2mqtt/files/6015013/E-life.Zigbee.Modul.User.Manual.v2.0.pdf - last page (page 16) documents the missing functionality

heinekmadsen, 2023-06-09:
@jkaberg nice find... hope some one with knowledge has time to get the last bits and pieces together so we have the full support.
I have been in contact with the Danish distributor and they confirm that their gateway uses zigbee for communication with the lock, and they get all the info using zigbee

heinekmadsen, 2023-06-12:
@Koenkk Can you give us a hint in any direction on how we can continue top improve this?

MarkMikkelsen83, 2023-06-30:
Im still very interested in this - Hopefully to see the ID and what unlocking methode is used...

mongojarle, 2023-06-30:
Someone told me, that they had read, that in order to get the ID and method of unlocking reported over zigbee, the user had to be set up using zigbee. I have not tested this myself.

MarkMikkelsen83, 2023-06-30:
> Someone told me, that they had read, that in order to get the ID and method of unlocking reported over zigbee, the user had to be set up using zigbee. I have not tested this myself.

And is that possible from Zigbee2mqtt and home assistant already?

mongojarle, 2023-06-30:
Does not work from the Z2M interface at least, so I guess programming the lock is not working.

`2023-06-30 18:04:38 Publish 'set' 'pin_code' to '0xf4ce362a9c8f621f' failed: 'Error: Command 0xf4ce362a9c8f621f/11 closuresDoorLock.setPinCode({"userid":10,"userstatus":1,"usertype":0,"pincodevalue":"3591"}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43537 - 11 - 60 - 257 - 5 after 10000ms)'`

abualy, 2023-06-30:
yes it is possible, by publishing the message to z2m as q service call:
```yaml
service: mqtt.publish
data:
  qos: 0
  retain: false
  payload: >-
    {"user": "004", "user_type": "unrestricted", "user_enabled": "true",    
    "pin_code": "6666"}
  topic: zigbee2mqtt/0xfxxxxxxxx/set/pin_code
```

jkaberg, 2023-06-30:
> Someone told me, that they had read, that in order to get the ID and method of unlocking reported over zigbee, the user had to be set up using zigbee. I have not tested this myself.

All my (doorlock) users are setup via z2m interface, and still reports nothing

mongojarle, 2023-06-30:
Ok, disregard my last post, it is working to program it when publishing as @abualy described. But it is not reporting user or method when used on the lock.

MarkMikkelsen83, 2023-07-01:
Seems strange that we can't pull that data ?

heinekmadsen, 2023-07-01:
The data is there for sure (hope so)... Had a chat with EasyAccess/Nimly support who stated that the Nimly gateway get that information from Zigbee.

I think the issue is that the information is not getting passed to any or the correct action_source_name, action_user, and action.

I everyone upvote this issue by clicking on the small Smiley icon in the first post in this issue, and select a thumbs up it might get higher priority (Some github repos use this to prioritize, not sure if that's how zigbee2mqtt does prioritizing, but sure wouldn't hurt with a thumbs up)

MarkMikkelsen83, 2023-07-01:
Could the support not provide some more info about this ?

izzor, 2023-07-03:
I have sniffed my Zigbee network with a CC2531 sniffer module, Wireshark, Sonoff Z8Dongle-E and a Nimly lock. The lock seems to use a new or undocumented way of reporting user, action and method.

When I open my door the lock sends a Report Attribute (0x0A) with attribute 0x0100 and a 32-bit bitmap.
I cannot find the attribute 0x0100 in the Zigbee Cluster Library and Wireshark does not recognize the attribute.
The 32-bit bitmap match the Operation Event Notification Command described in section 7.3.2.16.27.
Octet 1 – Event source
Octet 2 – Event code
Octet 3,4 – User id

On my lock I get the following
0x0202001f – Keypad 0x02, Unlock door 0x02, User 31
0x0302001f – Fingerprint 0x03, Unlock door 0x02, User 31
0x0402001f – RFID 0x04, Unlock door 0x02, User 31
Opening with the key does not report anything

I tried to sniff the lock when connected to the Nimly Gateway but without the Trust Center Link Key I can't get any further.

jkaberg, 2023-07-03:
> I have sniffed my Zigbee network with a CC2531 sniffer module, Wireshark, Sonoff Z8Dongle-E and a Nimly lock. The lock seems to use a new or undocumented way of reporting user, action and method. [...]

This seems to correspond with the docs? https://github.com/Koenkk/zigbee2mqtt/files/6015013/E-life.Zigbee.Modul.User.Manual.v2.0.pdf

izzor, 2023-07-03:
It’s almost the same but the E-life module must be based on an older revision of ZCL. The chapter 7.3.2.17.27, Operation Notification Response, does not exist in revision 8 of the ZCL specification.

It also looks as if the event source enumerations have changed. I rev 8 Keypad has code 0x00 Nimly reports 0x02, RFID has code 0x03 and Nimly reports 0x04

The attribute 0x0100 is not mentioned in revision 8 so I guess there must be a newer one, somewhere.

heinekmadsen, 2023-07-05:
Nice work sniffing the traffic...

I'm not very much into zigbee communication, but seems impressive :) 

So what's the next step? Do we need changes to herdsman in order to get this working, or?

izzor, 2023-07-05:
Yes,
Both herdsman and herdsman-converters need some changes.
I looked into the code of herdsman and I guess adding the attribute would be easy. I think a line like this:
` lockOperation: {ID: 256, type: DataType.bitmap32},`
in the closuresDoorLock section of cluster.ts could work.

But most of the work has to be made in herdsman-converters. If we get a rainy summer I might have time to sort it out :)

heinekmadsen, 2023-07-05:
> Yes, Both herdsman and herdsman-converters need some changes. I looked into the code of herdsman and I guess adding the attribute would be easy. I think a line like this: ` lockOperation: {ID: 256, [...]

Well not that I wish anyone a rainy summer, but I might wish you'll get a little rain :)
But then again, good things take time, and the best things takes even longer... I'll be ready for testing when ever its ready

MarkMikkelsen83, 2023-07-19:
> Yes, Both herdsman and herdsman-converters need some changes. I looked into the code of herdsman and I guess adding the attribute would be easy. I think a line like this: ` lockOperation: {ID: 256, [...]

>Fingers crossed, this would be awesome

slackspace-io, 2023-07-23:
Hi, Just thought i'd add to earlier comments on issues registering pins. I've spent a bit, and the only way I've found to disable a pin is sending a null for the pin code. The user enabled true/false does not affect. The pin remained active until I sent a null for pin_code. 

Incase others find this thread, here is an example to create/activate a user/pin.

Enable/Create user 10, with pin 1234

```
service: mqtt.publish
data:
  qos: 0
  retain: false
  payload: >-
    {"user": "010", "user_type": "unrestricted", "user_enabled":
    "true","pin_code": "1234"}
  topic: zigbee2mqtt/your_lock/set/pin_code

```

Disable user 10

```
service: mqtt.publish
data:
  qos: 0
  retain: false
   payload: >-
    {"user": "010", "user_type": "", "user_enabled":
    "false","pin_code": null }
  topic: zigbee2mqtt/your_lock/set/pin_code
```

slackspace-io, 2023-07-27:
Hi @TheQue42 @jkaberg  @abualy @heinekmadsen  

I've submitted a couple PRs for this. The first one is in the dev branch, but unfortunately has a couple issues due to me :)

But feel free to check, the 'state'  should give you an indication of what is reported:

```
{
    "auto_relock": true,
    "battery": 100,
    "last_lock_source": "self",
    "last_lock_user": 0,
    "last_seen": "2023-07-27T19:55:38+00:00",
    "last_unlock_source": "keypad",
    "last_unlock_user": 3,
    "last_used_pin_code": null,
    "linkquality": 116,
    "lock_state": "locked",
    "pin_code": null,
    "sound_volume": "high_volume",
    "state": "LOCK",
    "auto_relock_time": 1,
    "last_used_pincode": "1234"
}
```

I split last unlock and lock into unique fields for user and source. 
Source can be zigbee/self, rfid, keypad, fingerprint
user is the numerical value

last_used_pin_code is the last pin entered on the device, or rather the numbers entered. I have not tried to polish this as longer numbers do become hex values. If your pins is 1234 but you entered 912345 -- that is what will be shown. 

The two known bugs in this release:

- last_used_pin_code/last_used_pincode -- see in the state reported both are listed bu only last_used_pincode has a value.  This is a typo on me, it will become last_used_pin_code if/when the PR is merged. 
- The 'Exposes' page is broken, specifically last unlock/lock source rows. (As well as pin_code due to above). 

 -- On setting a pin via zigbee, honestly I find that section more harm than good. It seems to send messages as you populate each field. I wiped my user while messing with it, and could easily see someone accidentally create users. I have not spent any time on this pin setting section.  For setting pins I publish MQTT above.

MarkMikkelsen83, 2023-07-27:
Does this mean that we can now pull these data's ? When will they be available?

oskretc, 2023-07-28:
Looking forward to the PR merge, I'll try to test dev branch if time later today.

slackspace-io, 2023-07-28:
> Does this mean that we can now pull these data's ? When will they be available?

@MarkMikkelsen83  The first set of changes are available in the 'dev' branch of zigbee2mqtt or 'Edge' as it's called in home assistant addon I believe - https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#installation if you are not comfortable running that version, then will need to wait for the next official release. I do not know how when that would be.

Hopefully my bug fixes will be merged in the next dev release assuming I didn't make any new mistakes after they review the code :)

heinekmadsen, 2023-07-31:
@slackspace-io Great work!
Seems to be working just fine.

I'll keep testing

MarkMikkelsen83, 2023-07-31:
Is it available on the normal Zigbee2mqtt ?

slackspace-io, 2023-08-01:
@MarkMikkelsen83  the improvements are in today's normal release (version 1.32.2)

Hope they work for others!

TheQue42, 2023-08-02:
> @MarkMikkelsen83 the improvements are in today's normal release (version 1.32.2)
> 
> Hope they work for others!

So many thanks!!! This is wonderful! Have I understood it correctly, that the pin_code is actually used only if I want to set pin-codes?

slackspace-io, 2023-08-02:
You are referring to the section 'pin_code' within the  'Exposes' portion of zigbee2mqtt? 

If so, yes that seemingly is the intent-- But, I have found it to be very difficult to use. It seems to send commands after entry to each field. I've managed to disable users, and enable users while messing with it. -- I have not done any changes to this section with my improvements. 

Personally, I find this section a bit more 'dangerous' than helpful. Since you could set pins on use accounts by accident, or wipe by accident. I publish a message to MQTT directly when I want to manager users (The cleaner's PIN only becomes active the day/hours they are expected, etc).

What are others experience with this section? if others find it difficult/dangerous, I can try to find some time to see if any options to improve it..

dislabled, 2023-08-04:
Thank you for the awesome work @slackspace-io.

I am having random opening of the door after latest stable update. I am not sure it is related, but I'm finding it hard to troubleshoot as it happens completely random, and the door seems to report "self" at source for unlock, even though i am triggering it manually from zigbee. Anyone else not seeing zigbee as source?
It does report keypad when i am using code.

See logs from manually triggering open:

`Debug 2023-08-04 13:35:56 Received MQTT message on 'zigbee2mqtt/main_door_lock/set' with data 'UNLOCK'
Debug 2023-08-04 13:35:56 Publishing 'set' 'state' to 'main_door_lock'
Debug 2023-08-04 13:35:59 Received Zigbee message from 'main_door_lock', type 'commandUnlockDoorRsp', cluster 'closuresDoorLock', data '{"status":0}' from endpoint 11 with groupID 0
Debug 2023-08-04 13:35:59 Received Zigbee message from 'main_door_lock', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 11 with groupID 0
Info 2023-08-04 13:35:59 MQTT publish: topic 'zigbee2mqtt/main_door_lock', payload '{"auto_relock":false,"auto_relock_time":0,"battery":100,"last_lock_source":"self","last_lock_user":0,"last_unlock_source":"self","last_unlock_user":0,"last_used_pin_code":null,"linkquality":184,"lock_state":"unlocked","pin_code":null,"sound_volume":"low_volume","state":"UNLOCK"}'
Debug 2023-08-04 13:35:59 Received Zigbee message from 'main_door_lock', type 'attributeReport', cluster 'closuresDoorLock', data '{"256":167903232}' from endpoint 11 with groupID 0
Info 2023-08-04 13:35:59 MQTT publish: topic 'zigbee2mqtt/main_door_lock', payload '{"auto_relock":false,"auto_relock_time":0,"battery":100,"last_lock_source":"self","last_lock_user":0,"last_unlock_source":"self","last_unlock_user":0,"last_used_pin_code":null,"linkquality":188,"lock_state":"unlocked","pin_code":null,"sound_volume":"low_volume","state":"UNLOCK"}'`

slackspace-io, 2023-08-05:
@dislabled I can't think of what could cause 'random unlocks' -- can you explain a bit more? Are you 'randomly' hearing the lock unlocking without user interaction?  Or are the logs claiming the door has been opened? 

If you have auto lock true -- the door does lock 'itself' a few seconds even if you selected 'unlock' from within zigbee2mqtt (or via HA). This relock will appear as 'self' in the state. 

I need to check myself tonight if perhaps I have a bug and not reporting 'zigbee' as a source correctly. - I have it expected 00 as the first octet to be zigbee, while 0a is 'self. It is also possible there is not a different message sent for zigbee vs self and I just hallucinated it when making these improvements:) I'll try to test tonight or tomorrow and clarify.

I'd expect one of the two scenarios, assuming you have 'auto relock' true.

1.) Send 'unlock' via zigbee2mqtt/mqtt. 
2.) Latest 'unlock' source becomes zigbee(tho I agree it seems to be reporting 'self' now this is etiher due to a bug or that I was early incorrect and infact self/zigbee is the 'same' 0x0a beint sent).
3.) Few seconds later, door relocks automatically
4.) latest 'lock' source becomes self. (This would be correct).

dislabled, 2023-08-05:
I am not blaming this random unlocks on the update, im just trying to troubleshoot what is causing the unlocks.
And i don't want to hijack this thread with this issue. It's just related when I am trying to find out if the unlock command is infact coming from zigbee, or the lock itself is bugging out.

My automations in HA turns off auto_relock at daytime, and turns it back on at nighttime. There is also some automations that unlocks the door when house occupancy goes from 0 to 1.
Will include a log from HA, and that log will explicity tell you the reason for unlocks. So I am convinced that its not some automation running wild and opening the door:

[image]

DewGew, 2023-08-17:
I have a Nimly Touch. Does z2m support this one also? Its same lock but without fingerprint reader.

DewGew, 2023-08-20:
Is it possible to add 2FA, and anti burgler functions?

Dorenix, 2023-09-21:
Thank you for the great work @slackspace-io! Sorry for my lack of knowledge, but how do you set the time window for time year/week day dependent access? I can not find it in the Zigbee2MQTT device description.

larsfjaera, 2023-10-20:
> Thank you for the great work @slackspace-io! Sorry for my lack of knowledge, but how do you set the time window for time year/week day dependent access? I can not find it in the Zigbee2MQTT device description.

I was wondering the same thing. However, I figured I'd just use scheduled automations to enable/disable a user essentially achieveing the same functionality.

jkaberg, 2024-01-02:
Anyone experiencing that these locks stops to respond and becomes unavailable? Currently I beleve this happends after a while with no (physical) usage 

Doesn't seem like removing and inserting batteries fixes it either, will check more tonight

izzor, 2024-01-02:
My module behaved the same for a couple of months, unavailable and available after removing/inserting the batteries.
Finally the module stopped connecting, now flashing leds, and worst of all it began to drain my batteries.

I've had some previous problems with my modules and I've got excellent support from Nimly before.
So, I contacted Nimly support and they confirmed my observations. There is a problem with the module. I was told to hold on to my broken module and wait for a replacement module.
They could not give me an estimated date for a new module but I have an open case in their CRM.

jkaberg, 2024-01-02:
@izzor whats the "Firmware build date" on your's? Mine's 20220614

I did reach out to Nimly recently about this, their respons was
```
What kind of Zigbee controller are you using? Do you have our Connect Gateway as a home central or another supported gateway?
 
If you have a colored mark on your module, it can be an indication of which generation you have. Please send a picture of both your modules.
```
I'll follow up on this, however I did notice the "Do you have our Connect Gateway as a home central or another supported gateway?" - this might become an issue in the future with these locks.

izzor, 2024-01-02:
@jkaberg My first module had 202211-something. After I made a new installation of HA and Z2M, it refused to connect to any new zigbee network, not even Nimlys own gateway worked. It turned out that there was a bug in network discovery.
My second module was 20230210 and it worked a few weeks before it got silent.
And my latest module is also 20230210, it behaved as described.

I  have a few month left of the consumer warranty so hope the get it right soon.

jkaberg, 2024-01-15:
So I received 2 modules with build date 20230211 FW ver 4.5.24. It'll be interesting to see if there's any difference to your experience @izzor

newlund, 2024-01-21:
I also received a replacement module from Nimly support with build date 20230211 FW ver 4.5.24.
Strange since they said this would be a newly developed firmware?
So far it seems to work slightly better but I already lost connection with it once. Had to remove/insert the batteries in order to get it to connect again. :(

DewGew, 2024-03-20:
> I also received a replacement module from Nimly support with build date 20230211 FW ver 4.5.24. Strange since they said this would be a newly developed firmware? So far it seems to work slightly better but I already lost connection with it once. Had to remove/insert the batteries in order to get it to connect again. :(
> 
> Also, noticed that battery percentage never goes above 50% I submitted a PR that should fix this: [Koenkk/zigbee-herdsman-converters#6940](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6940)

Is it also possible to add 2FA, and anti burgler functions?

newlund, 2024-03-20:
> 
> Is it also possible to add 2FA, and anti burgler functions?

Where? What? 🙂

DewGew, 2024-03-20:
> 
> 
> > 
> 
> > Is it also possible to add 2FA, and anti burgler functions?
> 
> 
> 
> Where? What? 🙂

Expose those functions. Page 13 in manual https://nimly.se/wp-content/uploads/2024/02/se-Touch-Installation-Manual-16052023.pdf

newlund, 2024-03-21:
To me, those seem not available to set or read via the zigbee module according to this spec:
https://github.com/Koenkk/zigbee2mqtt/files/6015013/E-life.Zigbee.Modul.User.Manual.v2.0.pdf

But someone more experienced have to answer that.

Dauzzen, 2024-03-21:
> I also received a replacement module from Nimly support with build date 20230211 FW ver 4.5.24. Strange since they said this would be a newly developed firmware? So far it seems to work slightly [...]

I've been running version 4.5.26 (Fw build 20240201) for a couple of weeks now and it seems to work without any issues at all. Any difference in battery consumption is difficult to say yet.

[image]

newlund, 2024-03-21:
> I've been running version 4.5.26 (Fw build 20240201) for a couple of weeks now and it seems to work without any issues at all. Any difference in battery consumption is difficult to say yet. [...]

Nice! Thanks for info! :)
Nimly support have sent a new module to me again. I hope I get the same version.
This is the third module they have sent me. The first two had the same build date and version (20230211 FW ver 4.5.24)
I still have problems with disconnects from time to time.

izzor, 2024-03-21:
I've had my upgraded module for two months now, and I must say it works absolutely perfect.
No disconnects and no battery drain. My battery is now down from 88%, when I installed the module, to 77%.

One thing I've done in my Zigbee network was to replace all IKEA Trådfri switches to Nous smartplug and installing wall mounted Sunricher switches.
Before, my lock was connected to the IKEA switch I had in the outlet next to the door. Now it connects to the Sunricher switch behind my wall switch.

I suspect that the IKEA switch may have some problem when other Zigbee devices connects to them. 

I finally have the lock I wanted. Connected to Zigbee2Mqtt, HomeAssistant and unlock events reported to my phone.

DewGew, 2024-03-21:
I got mine replaced to FW 20240201 / 4.5.26  last week also. I hope its working.

Does last_actions_source report correct? keypad and rfid shows properly but if I press button or use Z2M it show "self".

[21 more replies]

closed 2025-06-02 by github-actions[bot]