Koenkk/zigbee2mqtt #17206: [Feature request]: Reset limit support for ZNCLDJ14LM (Xiaomi Aqara Curtain Motor C2)

gofort, 2023-04-01:
### Is your feature request related to a problem? Please describe

[Xiaomi ZNCLDJ14LM](https://www.zigbee2mqtt.io/devices/ZNCLDJ14LM.html) is a new version of [Xiaomi ZNCLDJ11LM](https://www.zigbee2mqtt.io/devices/ZNCLDJ11LM.html).

At the same time, the old version has `reset_limits` option, but I can't find it in the new version.

[image]

[image]

Please add the option to `reset_limits` for ZNCLDJ14LM.

### Describe the solution you'd like

I'd like to see the ability to reset curtain limits in UI.

### Describe alternatives you've considered

I tried to reset the limits using the button on the device itself, but it reset only pairing, not the limits of curtains.
I have checked the whole internet trying to understand if the button on the device can reset limits, but it looks like it can't and it is available only using the Aqara hub.

### Additional context

Added screenshot to the feature description.

closed 2023-05-10 by github-actions[bot]

mwargocki, 2023-05-21:
This motor starts to be popular so it would be great if there would be full support of it.

deiger, 2024-02-12:
Are `hand_open` and `reset_limits` supported by the motor?