Koenkk/zigbee2mqtt #17209: Innr firmware downgrading

Ulrar, 2023-04-01:
### What happened?

With the latest version I now see my Innr spots (RS 229 T) in the OTA tab, and they all have an update available.
I updated one as a test and it succeeded, however the new version seems to be lower than the old one. That said the build date did go up.
See two of the same as an example, the first one was updated and went from 2.3 to 2.00.08 : 

[image]

### What did you expect to happen?

I expect the firmware version to go up after an update, but maybe those versions are just poorly numbered and should be ignored ?

### Zigbee2MQTT version

1.30.3 commit: 24c6b2e

### Adapter firmware version

20221226

### Adapter

TubeSZB PoE

satter, 2023-04-01:
I think this firmware version in the OTA tab is purely informative

Better check versions reported by devices state, one of my Tuya plugs for example:
```
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
```
And version on the OTA tab is missing

[image]

What versions does your devices report in state tab?

Ulrar, 2023-04-01:
The updated one : 

```
"update": {
        "installed_version": 537422384,
        "latest_version": 537422384,
        "state": "idle"
    }
```

The old ones : 

```
"update": {
        "installed_version": 269837568,
        "latest_version": 537422384,
        "state": "available"
    }
```

Looks like you might be right, maybe the reported version is invalid somehow ?

Koenkk, 2023-04-02:
The swBuildID (2.00.08) does not determine a newer/older file version, that is done by the file version (e.g. `269837568`). However this can indeed be confusing, I've notified from Innr.

Koenkk, 2023-04-02:
Thanks for the clarification @GerryInnr !

closed 2023-04-02 by Koenkk