Koenkk/zigbee2mqtt #17211: [New device support]: OWON PCT504-E

pcrednet, 2023-04-01:
### Link

https://www.owon-smart.com/zigbee-fancoil-thermostat-with-remote-control-via-app-pct504-z-product/

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0x3c6a2cfffed152ec","nwkAddr":31579,"manufId":4412,"manufName":"OWON Technology Inc.","powerSource":"DC Source","modelId":"PCT504-E","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,513,514,516,1029],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{}},"hvacThermostat":{"attributes":{"runningState":4,"localTemp":1750,"occupiedCoolingSetpoint":2500,"occupiedHeatingSetpoint":1750,"ctrlSeqeOfOper":4,"systemMode":4,"runningMode":4,"tempSetpointHold":1,"tempSetpointHoldDuration":65535,"programingOperMode":0}},"hvacUserInterfaceCfg":{"attributes":{"tempDisplayMode":0,"keypadLockout":0}},"hvacFanCtrl":{"attributes":{"fanMode":1}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":4,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":516,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":514,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1}],"configuredReportings":[{"cluster":514,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10}],"meta":{}},"2":{"profId":260,"epId":2,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":4,"dateCode":"20200513","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1680382735261,"defaultSendRequestWhen":"immediate"}

### Comments

Same as ( i think... )PCT504

created new file devices.js with another zigbeemodel and works perfectly except humidity and ocupancy.

pcrednet, 2023-04-01:
[image]

pcrednet, 2023-04-01:
[image]
[image]

pcrednet, 2023-04-01:
[image]

Koenkk, 2023-04-02:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

pcrednet, 2023-04-04:
I have switched to dev

it recognizes it well, it gives an error at the beginning of configuration

Humidity is still not displayed, null value

db:
`{"id":6,"type":"Router","ieeeAddr":"0x3c6a2cfffed152ec","nwkAddr":35509,"manufId":4412,"manufName":"OWON Technology Inc.","powerSource":"DC Source","modelId":"PCT504-E","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,513,514,516,1029],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"zclVersion":3}},"hvacThermostat":{"attributes":{"localTemp":2300,"occupiedCoolingSetpoint":2500,"occupiedHeatingSetpoint":2100,"ctrlSeqeOfOper":4,"systemMode":4,"runningMode":4,"tempSetpointHold":1,"tempSetpointHoldDuration":65535,"programingOperMode":1,"runningState":4}},"hvacUserInterfaceCfg":{"attributes":{"tempDisplayMode":0,"keypadLockout":0}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":4,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":516,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":514,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1},{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e90e5","endpointID":1}],"configuredReportings":[{"cluster":514,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10}],"meta":{}},"2":{"profId":260,"epId":2,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":4,"dateCode":"20200513","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1680595307719,"defaultSendRequestWhen":"immediate"}`

Log:
`error 2023-04-04 10:01:47: Failed to configure '52ec_owon', attempt 3 (Error: ConfigureReporting 0x3c6a2cfffed152ec/1 hvacThermostat([{"attribute":"unoccupiedHeatingSetpoint","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":10}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Object.thermostatUnoccupiedHeatingSetpoint (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:160:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/owon.js:316:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13))`

[image]

[image]

Koenkk, 2023-04-04:
First the configure error has to be fixed.

Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PCT504', 'PCT504-E'],
    model: 'PCT504',
    vendor: 'OWON',
    description: 'HVAC fan coil',
    fromZigbee: [fz.fan, fz.thermostat, fz.humidity, fz.occupancy, fz.legacy.hvac_user_interface],
    toZigbee: [tz.fan_mode,
        tz.thermostat_occupied_heating_setpoint, tz.thermostat_unoccupied_heating_setpoint,
        tz.thermostat_occupied_cooling_setpoint, tz.thermostat_unoccupied_cooling_setpoint,
        tz.thermostat_min_heat_setpoint_limit, tz.thermostat_max_heat_setpoint_limit,
        tz.thermostat_min_cool_setpoint_limit, tz.thermostat_max_cool_setpoint_limit,
        tz.thermostat_local_temperature,
        tz.thermostat_keypad_lockout,
        tz.thermostat_system_mode, tz.thermostat_running_mode, tz.thermostat_running_state, tz.thermostat_programming_operation_mode],
    exposes: [e.humidity(), e.occupancy(),
        exposes.climate().withSystemMode(['off', 'heat', 'cool', 'fan_only', 'sleep']).withLocalTemperature()
            .withRunningMode(['off', 'heat', 'cool'])
            .withRunningState(['idle', 'heat', 'cool', 'fan_only'])
            .withSetpoint('occupied_heating_setpoint', 5, 30, 0.5).withSetpoint('unoccupied_heating_setpoint', 5, 30, 0.5)
            .withSetpoint('occupied_cooling_setpoint', 7, 35, 0.5).withSetpoint('unoccupied_cooling_setpoint', 7, 35, 0.5),
        e.fan().withModes(['low', 'medium', 'high', 'on', 'auto']),
        e.programming_operation_mode(['setpoint', 'eco']), e.keypad_lockout(),
        e.max_heat_setpoint_limit(5, 30, 0.5), e.min_heat_setpoint_limit(5, 30, 0.5),
        e.max_cool_setpoint_limit(7, 35, 0.5), e.min_cool_setpoint_limit(7, 35, 0.5)],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['genBasic', 'genIdentify', 'genGroups', 'hvacThermostat', 'hvacUserInterfaceCfg', 'hvacFanCtrl',
            'msTemperatureMeasurement', 'msOccupancySensing'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.fanMode(endpoint);
        await reporting.bind(endpoint, coordinatorEndpoint, ['hvacThermostat']);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        // await reporting.thermostatUnoccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatOccupiedCoolingSetpoint(endpoint);
        await reporting.thermostatUnoccupiedCoolingSetpoint(endpoint);
        await reporting.thermostatTemperature(endpoint, {min: 60, max: 600, change: 0.1});
        await reporting.thermostatSystemMode(endpoint);
        await reporting.thermostatRunningMode(endpoint);
        await reporting.thermostatRunningState(endpoint);
        await reporting.humidity(endpoint, {min: 60, max: 600, change: 1});
        await reporting.thermostatKeypadLockMode(endpoint);

        await endpoint.read('hvacThermostat', ['systemMode', 'runningMode', 'runningState',
            'occupiedHeatingSetpoint', 'unoccupiedHeatingSetpoint',
            'occupiedCoolingSetpoint', 'unoccupiedCoolingSetpoint', 'localTemp']);
        await endpoint.read('msRelativeHumidity', ['measuredValue']);

        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['msOccupancySensing']);
        await reporting.occupancy(endpoint2, {min: 1, max: 600, change: 1});
        await endpoint2.read('msOccupancySensing', ['occupancy']);
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

pcrednet, 2023-04-05:
Hello, good morning.
I have put the code you have passed me.
The humidity works fine. The occupancy sensor still does not update, if I read the value if it reads it, but by itself it does not send update to see the status.

Koenkk, 2023-04-05:
Does the configure succeed now? You can manually trigger it via the z2m frontend -> device -> yellow refresh button.

pcrednet, 2023-05-05:
Hello, sorry for the delay...

It not changes automatically, if i refresh then show status.

Thx

pcrednet, 2023-06-05:
Hello again. 
Do you want me to do some more testing? Proximity server does not update Its the only thing.
I have spoken with the manufacturer and they have indicated that this is the model that is being supplied now, therefore it would be interesting to add the support.

Thank you so much.

closed 2023-07-14 by github-actions[bot]