Koenkk/zigbee2mqtt #17213: Xiaomi SRTS-A01 OTA issues : Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet

mrpinkuk, 2023-04-01:
### What happened?

Im trying to update from the 0.0.0_0825 firmware and clicking check update results in this message Failed to check if update available for 'Living Room Radiator ' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)

### What did you expect to happen?

Check update

### How to reproduce it (minimal and precise)

Click check updates

### Zigbee2MQTT version

1.30.3-dev commit: 64d744e

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Debug log

[Uploading log.txt…]()

DaN66O, 2023-04-02:
I confirm this
The same problem.problem
In more error logs:
```
Warning 2023-04-02 19:33:55zigbee-herdsman-converters:aqara_trv: Unknown key 641 = 4��84�4
Warning 2023-04-02 19:33:55zigbee-herdsman-converters:aqara_trv: Unknown key 642 = `4��84�4
Warning 2023-04-02 19:33:55zigbee-herdsman-converters:aqara_trv: Unknown key 643 = 0
Warning 2023-04-02 19:33:55zigbee-herdsman-converters:aqara_trv: Unknown key 644 = 0
```

MrDodojo, 2023-04-04:
Also trying to update from 0.0.0_0825 firmware. 

Below this you can see the debug log, there is an `Error: Did not receive a current fileVersion` but a couple of messages before it states that a zigbee messagefrom  "Verwarming" (The aqara TRV) got received, which includes the data `"fileVersion":25`
```
Info 2023-04-04 11:59:21Checking if update available for 'Verwarming'
Info 2023-04-04 11:59:21MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Verwarming'","meta":{"device":"Verwarming","status":"checking_if_available"},"type":"ota_update"}'
Debug 2023-04-04 11:59:21Check if update available for '0x54ef4410007021f5' (lumi.airrtc.agl001)
Debug 2023-04-04 11:59:21Using endpoint '1'
Debug 2023-04-04 11:59:21Received Zigbee message from 'Verwarming', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":25,"imageType":5017,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Debug 2023-04-04 11:59:21Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":5017,"fileVersion":25}'
Info 2023-04-04 11:59:21MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Verwarming' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)","meta":{"device":"Verwarming","status":"check_failed"},"type":"ota_update"}'
Info 2023-04-04 11:59:21MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Verwarming"},"error":"Failed to check if update available for 'Verwarming' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)","status":"error","transaction":"vbdt4-4"}'
Error 2023-04-04 11:59:21Failed to check if update available for 'Verwarming' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)
Debug 2023-04-04 11:59:21Error: Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:131:19) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:35) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:45)
```

ShadowRep, 2023-04-04:
I had the same problem.
Removing and reinserting the batteries solved it.

I then was able to upgrade to firmware "0.0.0_01030" dated "Feb 20 2023".

Koenkk, 2023-04-04:
@ShadowRep interesting! I guess this makes the device report the file version, can others verify this?

mrpinkuk, 2023-04-04:
I’m afraid this did not fix the problem on my end

ShadowRep, 2023-04-04:
I am using the SONOFF Zigbee 3.0 USB Dongle Plus (TI CC2652P + CP2102(N)).
Maybe my log helps (I can provide the full log if needed, this seems to be super long already):

Log before removing batteries:

`info  2023-04-03 19:56:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Schlafzimmer Thermostat'","meta":{"device":"Schlafzimmer Thermostat","status":"checking_if_available"},"type":"ota_update"}'
debug 2023-04-03 19:56:06: Check if update available for '0x54ef44100067419f' (lumi.airrtc.agl001)
debug 2023-04-03 19:56:06: Using endpoint '1'
debug 2023-04-03 19:56:08: Received Zigbee message from 'Schlafzimmer Thermostat', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":30,"imageType":5017,"manufacturerCode":4447}' from endpoint 1 with groupID 0
info  2023-04-03 19:56:08: MQTT publish: topic 'zigbee2mqtt/Schlafzimmer Thermostat', payload '{"away_preset_temperature":"6.0","battery":33,"calibrated":true,"child_lock":"LOCK","device_temperature":27,"internal_heating_setpoint":21.5,"last_seen":1680544568222,"linkquality":120,"local_temperature":21.2,"occupied_heating_setpoint":21.5,"power_outage_count":101,"preset":"manual","schedule":"OFF","schedule_settings":"mon,tue,wed,thu,fri|8:00,22.0|18:00,21.0|23:00,21.5|8:00,22.0","sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":2334,"latest_version":2334,"state":"idle"},"update_available":false,"valve_alarm":false,"valve_detection":"OFF","voltage":3200,"window_detection":"OFF","window_open":false}'
debug 2023-04-03 19:56:08: Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":5017,"fileVersion":30}'
info  2023-04-03 19:56:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Schlafzimmer Thermostat' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)","meta":{"device":"Schlafzimmer Thermostat","status":"check_failed"},"type":"ota_update"}'
info  2023-04-03 19:56:08: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Schlafzimmer Thermostat"},"error":"Failed to check if update available for 'Schlafzimmer Thermostat' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)","status":"error","transaction":"5ht26-3"}'
error 2023-04-03 19:56:08: Failed to check if update available for 'Schlafzimmer Thermostat' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)
debug 2023-04-03 19:56:08: Error: Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again
    at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:131:19)
    at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:35)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:45)
`

And after:

`debug 2023-04-03 20:00:56: Device 'Schlafzimmer Thermostat' announced itself
info  2023-04-03 20:00:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Schlafzimmer Thermostat","ieee_address":"0x54ef44100067419f"},"type":"device_announce"}'
info  2023-04-03 20:00:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Schlafzimmer Thermostat"},"type":"device_announced"}'
info  2023-04-03 20:00:56: MQTT publish: topic 'zigbee2mqtt/Schlafzimmer Thermostat', payload '{"away_preset_temperature":"6.0","battery":33,"calibrated":true,"child_lock":"LOCK","device_temperature":27,"internal_heating_setpoint":21.5,"last_seen":1680544856643,"linkquality":120,"local_temperature":21.4,"occupied_heating_setpoint":21.5,"power_outage_count":101,"preset":"manual","schedule":"OFF","schedule_settings":"mon,tue,wed,thu,fri|8:00,22.0|18:00,21.0|23:00,21.5|8:00,22.0","sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":2334,"latest_version":2334,"state":"idle"},"update_available":false,"valve_alarm":false,"valve_detection":"OFF","voltage":3200,"window_detection":"OFF","window_open":false}'
info  2023-04-03 20:00:56: MQTT publish: topic 'zigbee2mqtt/0x54ef4410007766ad', payload '{"battery":100,"detection_interval":10,"device_temperature":26,"illuminance":46,"last_seen":1680544856855,"linkquality":123,"motion_sensitivity":"high","occupancy":false,"power_outage_count":0,"trigger_indicator":true,"update":{"installed_version":6,"latest_version":10,"progress":16.44,"remaining":1081,"state":"updating"},"update_available":false,"voltage":3173}'
info  2023-04-03 20:00:57: MQTT publish: topic 'zigbee2mqtt/0x54ef4410007766ad', payload '{"battery":100,"detection_interval":10,"device_temperature":26,"illuminance":46,"last_seen":1680544857163,"linkquality":127,"motion_sensitivity":"high","occupancy":false,"power_outage_count":0,"trigger_indicator":true,"update":{"installed_version":6,"latest_version":10,"progress":16.44,"remaining":1081,"state":"updating"},"update_available":false,"voltage":3173}'
info  2023-04-03 20:00:57: MQTT publish: topic 'zigbee2mqtt/0x54ef4410007766ad', payload '{"battery":100,"detection_interval":10,"device_temperature":26,"illuminance":46,"last_seen":1680544857473,"linkquality":116,"motion_sensitivity":"high","occupancy":false,"power_outage_count":0,"trigger_indi [... cut ...] installed_version":6,"latest_version":10,"progress":25.88,"remaining":954,"state":"updating"},"update_available":false,"voltage":3173}'
debug 2023-04-03 20:02:56: Received Zigbee message from 'Schlafzimmer Thermostat', type 'attributeReport', cluster 'aqaraOpple', data '{"65522":{"data":[170,113,15,132,12,70,6,65,12,0,21,141,0,1,157,27,152,8,0,7,253],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 2023-04-03 20:02:56: zigbee-herdsman-converters:aqara_trv: Unhandled key 65522 = �q�FA`

ntorring, 2023-04-04:
It appers to work fine on 1.30.2

demarch86, 2023-04-04:
I had the same problem. I downgraded it to 1.30.2 and now it's upgrading the device without problem

Carlettobes2020, 2023-04-04:
I had the same problem. I downgraded it to 1.30.2 and now it's upgrading the device without problem

DaN66O, 2023-04-04:
> can others verify this?

It didn't help me

MrDodojo, 2023-04-05:
So #16345 broke it?

programmdesign, 2023-04-06:
On version 1.30.2 it worked, but only with the issue linked by @MrDodojo

Koenkk, 2023-04-06:
Pushed a fix, can you check if it works again?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

MrDodojo, 2023-04-06:
I'm on 1.30.3-dev commit: 09a5650 now and it still doesn't update, but I'll try again in a few hours as the error messages says to do.

The log:
```
Debug 2023-04-06 21:19:43Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"verwarming","transaction":"u6oym-1"}'
Info 2023-04-06 21:19:43Checking if update available for 'verwarming'
Info 2023-04-06 21:19:43MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'verwarming'","meta":{"device":"verwarming","status":"checking_if_available"},"type":"ota_update"}'
Debug 2023-04-06 21:19:43Check if update available for '0x54ef4410007021f5' (lumi.airrtc.agl001)
Debug 2023-04-06 21:19:43Using endpoint '1'
Debug 2023-04-06 21:19:45Received Zigbee message from 'verwarming', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":25,"imageType":5017,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Debug 2023-04-06 21:19:45Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":5017,"fileVersion":25}'
Info 2023-04-06 21:19:45MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'verwarming' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)","meta":{"device":"verwarming","status":"check_failed"},"type":"ota_update"}'
Info 2023-04-06 21:19:45MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"verwarming"},"error":"Failed to check if update available for 'verwarming' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)","status":"error","transaction":"u6oym-1"}'
Error 2023-04-06 21:19:45Failed to check if update available for 'verwarming' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)
Debug 2023-04-06 21:19:45Error: Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:131:19) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:35) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:45)
```

MrDodojo, 2023-04-07:
Oh @Koenkk, you might've seen this already, sorry to bother you then. The pull request updating zhc #17266 didn't get merged because of conflicts.

Koenkk, 2023-04-08:
@MrDodojo fixed!

MrDodojo, 2023-04-08:
I'm on 1.30.3-dev commit: 09a5650
<https://github.com/Koenkk/zigbee2mqtt/commit/09a5650> now and it still
doesn't update, but I'll try again in a few hours as the error messages
says to do.

The log:
```
Debug 2023-04-06 21:19:43Received MQTT message on
'zigbee2mqtt/bridge/request/device/ota_update/check' with data
'{"id":"verwarming","transaction":"u6oym-1"}'
Info 2023-04-06 21:19:43Checking if update available for 'verwarming'
Info 2023-04-06 21:19:43MQTT publish: topic 'zigbee2mqtt/bridge/log',
payload '{"message":"Checking if update available for
'verwarming'","meta":{"device":"verwarming","status":"checking_if_available"},"type":"ota_update"}'
Debug 2023-04-06 21:19:43Check if update available for '0x54ef4410007021f5'
(lumi.airrtc.agl001)
Debug 2023-04-06 21:19:43Using endpoint '1'
Debug 2023-04-06 21:19:45Received Zigbee message from 'verwarming', type
'commandQueryNextImageRequest', cluster 'genOta', data
'{"fieldControl":0,"fileVersion":25,"imageType":5017,"manufacturerCode":4447}'
from endpoint 1 with groupID 0
Debug 2023-04-06 21:19:45Got OTA request
'{"fieldControl":0,"manufacturerCode":4447,"imageType":5017,"fileVersion":25}'
Info 2023-04-06 21:19:45MQTT publish: topic 'zigbee2mqtt/bridge/log',
payload '{"message":"Failed to check if update available for 'verwarming'
(Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet,
please wait some hours and try
again)","meta":{"device":"verwarming","status":"check_failed"},"type":"ota_update"}'
Info 2023-04-06 21:19:45MQTT publish: topic
'zigbee2mqtt/bridge/response/device/ota_update/check', payload
'{"data":{"id":"verwarming"},"error":"Failed to check if update available
for 'verwarming' (Did not receive a current fileVersion from
'lumi.airrtc.agl001' yet, please wait some hours and try
again)","status":"error","transaction":"u6oym-1"}'
Error 2023-04-06 21:19:45Failed to check if update available for
'verwarming' (Did not receive a current fileVersion from
'lumi.airrtc.agl001' yet, please wait some hours and try again)
Debug 2023-04-06 21:19:45Error: Did not receive a current fileVersion from
'lumi.airrtc.agl001' yet, please wait some hours and try again at
isNewImageAvailable
(/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:131:19)
at Object.isUpdateAvailable
(/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:35) at
OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:45)
```

MrDodojo, 2023-04-08:
@Koenkk worked!

kzajac83, 2023-04-15:
For me is not working.  

In March'23 buy one SRTS-A01 in meantime was issue  #16345 for repetitive FW update.  Next buy last week additional 3 devices and from Wenesday try update FW but cannot. 
I try switch Z2M into dev/Edge instance, try reconnect many times new SRTS-A01's and also remove batteires for longer time and try, alwys wiaitng few hours (usual more than 12h)  but unfortunelty nothing, still cannot update FW. 

My log Z2M from Edge is:

`
Debug 2023-04-15 20:48:43Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Głowica E1 salon","transaction":"08m2a-8"}'
Info 2023-04-15 20:48:43Checking if update available for 'Głowica E1 salon'
Debug 2023-04-15 20:48:43Check if update available for '0x54ef44100060b0ac' (lumi.airrtc.agl001)
Debug 2023-04-15 20:48:43Using endpoint '1'
Debug 2023-04-15 20:48:46Received Zigbee message from 'Głowica E1 salon', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":25,"imageType":5017,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Debug 2023-04-15 20:48:46Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":5017,"fileVersion":25}'
Info 2023-04-15 20:48:46MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Głowica E1 salon"},"error":"Failed to check if update available for 'Głowica E1 salon' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)","status":"error","transaction":"08m2a-8"}'
Error 2023-04-15 20:48:46Failed to check if update available for 'Głowica E1 salon' (Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again)
Debug 2023-04-15 20:48:46Error: Did not receive a current fileVersion from 'lumi.airrtc.agl001' yet, please wait some hours and try again at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:131:19) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:35) at runMicrotasks (<anonymous>) at processTicksAndRejections (node:internal/process/task_queues:96:5) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:45)
`

My view OTA tab below, 1st device is from March with updated FW, rest is from April without update :(

[image]

kzajac83, 2023-04-15:
Sorry for spam, however why there is auto-update switch if not working?  After unistall and install Edge start updating FW:

[image]

DaN66O, 2023-04-16:
I upgraded and got...
The firmware version is the same, but the dates are different
[image]

brebtatv, 2023-04-27:
The same problem

[image]

closed 2023-06-04 by github-actions[bot]

mazixs, 2024-02-17:
its not fixed now