Koenkk/zigbee2mqtt #17225: No battery status for Legrand 067774 double wireless switch

Loic691, 2023-04-03:
### What happened?

No battery status is get from the device.

### What did you expect to happen?

I saw in other post that the isssue have been solved.
I'm not sur this is really done...

### How to reproduce it (minimal and precise)

Reset the device and then pair it again on Z2M

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

7.1.4.0 build 389

### Adapter

Sonoff Dongle E with EFR32 chip

closed 2023-05-11 by github-actions[bot]

cobch, 2023-11-08:
Hi, did you manage to fix your issue. All my Legrand wireless switches have same problem and do not report battery level. Please help (-:

cobch, 2023-11-08:
@Loic691 Did you updated the code of the device?

Loic691, 2023-11-10:
There is  no update for wireless double switch.
And the update doesnt work with 1.30.2 for the wireless home away button.
The update is been detected with ota in z2m but no update start...

mycanaletto, 2023-11-14:
Same problem on 067646

Loic691, 2023-11-24:
Hi no news ?

mycanaletto, 2023-11-24:
The only solution (I've done it) is to buy a Legrand gateway, link the devices to it, and wait for the update to take place (it works, but you can't force it).

cobch, 2023-11-24:
> The only solution (I've done it) is to buy a Legrand gateway, link the devices to it, and wait for the update to take place (it works, but you can't force it).
> 
> 
> 
> Then the devices that were blocked on channel 11 take the other channels.

I have done that as well and I am to version 30. But still when the switch join z2m it is not reporting battery level. Do you have the same issue?

mycanaletto, 2023-11-24:
Yes, and it's a Z2M problem, because since then I've tested it on the deconz gateway and the battery goes back up just fine... (if you use the control unit, because from time to time it loses it, but it comes back when you use it)
[image]

cobch, 2023-11-24:
Exactly, I come from Deconz and it has always been working fine before. Someone can maybe do a PR to fix that. Any idea how to do that?

Loic691, 2024-03-08:
Hi @mycanaletto 
I have several problems with my 067774 double wireless switch and home away.
I try to update z2m to last version but this is same.
My legrand device are on last firmware.
I have the dummy group and the devis are long to send zigbee. It's randomly..
Some times the switch event arrives instantanously in HA maybe after 30 secondes....
Do you have same ?