Koenkk/zigbee2mqtt #17229: [New device support]: Add alarm switch to BSD-2

N3rdix, 2023-04-03:
### Link

https://www.zigbee2mqtt.io/devices/BSD-2.html

### Database entry

{"id":31,"type":"EndDevice","ieeeAddr":"0x04cd15fffef923f4","nwkAddr":59996,"manufId":4617,"manufName":"Bosch","powerSource":"Battery","modelId":"RBSH-SD-ZB-EU","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"RBSH-SD-ZB-EU","manufacturerName":"Bosch","powerSource":3,"zclVersion":8,"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20220222","swBuildId":"1.10"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x84b4dbfffef830f3","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffef830f3","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffef830f3","endpointID":1},{"cluster":64684,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffef830f3","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20220222","swBuildId":"1.10","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1394888596},"lastSeen":1680519698864,"defaultSendRequestWhen":"active"}

### Comments

This smoke detector already works fine with Z2M 1.30.3 but does not expose the alarm as a switch.
Via the Bosch gateway it is possible to e.g. trigger the smoke alarm when a door is opened or similar triggers are fired.

Can this attribute be made available as a switch also in Z2M?

N3rdix, 2023-05-04:
Anything we can do to get this working?

N3rdix, 2023-06-04:
Still needed

closed 2023-07-12 by github-actions[bot]

burpy, 2023-07-26:
this feature would be very helpful! Please let me know how I can help to get it.

vjonas, 2023-11-08:
yes please! I would also like to help the development if possible!

safakaltun, 2023-11-20:
I did some sniffing with multiple devices. Here are the commands that needs to be sent to the Bosch Smoke Alarm II to trigger certain actions. I haven't touched z2m yet. I don't have the time to implement this in the near future. If anyone else has the time to give it a go, please be my guest.

| Action                   | Cluster         | Command           | Data   |       
| --- | --- | --- | --- |   
| Relayed Smoke Alarm On   | IAS Zone 0x0500 | 0x80              | 003c       |   
| Relayed Smoke Alarm Off  | IAS Zone 0x0500 | 0x80              | 0000    |         
| Relayed Burglar Alarm On | IAS Zone 0x0500 | 0x80              | 01b4     |        
| Relayed Burglar Alarm Off| IAS Zone 0x0500 | 0x80              | 0100    |         
| Self test                | IAS Zone 0x0500 | 0x02              | -|

vjonas, 2023-11-20:
I do not think apple home supports having a switch out of the box for a smoke alarm. That is why we should expose it through a different plugin that sends the command to mqtt to the smoke detector zigbee device.
The downside will be: not getting the state of the smoke alarm being in one of the states mentioned above. ( burglar, smoke,...) 
We can send the command, but never be 100% sure it arrived to the device. That'll rely on the network stability and so on.

safakaltun, 2023-11-20:
That sounds like a limitation of Apple HomeKit, and not necessarily a problem with z2m, but maybe I am missing something. I use
Z2m through home assistant. There, you can send the same command every x seconds. This would be similar to how it works with the Bosch Controller. 

You can have a look at how Bosch Twinguard is implemented. If I was doing the implementation, it would look similar to that.

I will have a look at the status reports to see whether the device sends back any status signal or not.

N3rdix, 2023-11-24:
> You can have a look at how Bosch Twinguard is implemented. If I was doing the implementation, it would look similar to that.

I would love to give it a try, I have a fresh BSD2 paired in my test environment. I am still figuring out how I can test your commands in a local converter, maybe someone else has experience with that?

Mikkooooo, 2023-11-24:
I‘ve just added a BSD2 too.
It would be very useful to use the built-in siren for other purposes

vjonas, 2023-11-26:
> Hi Jonas, yes. The device sends IAS Zone Status Change notifications (IAS Zone cluster 0x0500) when the siren starts or stops. I mapped the zone status with their definitions in the table below.
> 
> ZoneStatus	Definition
> 0x0032	smoke alarm active
> 0x00b0	burglar alarm active
> 0x0030	clear
> Below screenshot is an example from one of the events when the intrusion alarm was triggered.
> 
> ![image](https://private-user-images.githubusercontent.com/36919065/285212602-f13f602f-104f-4537-b1db-bfbe1a3761b3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEwMDAxNjQsIm5iZiI6MTcwMDk5OTg2NCwicGF0aCI6Ii8zNjkxOTA2NS8yODUyMTI2MDItZjEzZjYwMmYtMTA0Zi00NTM3LWIxZGItYmZiZTFhMzc2MWIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI2VDExNTc0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY4YjM4MTQ0NTUzNjQ4NzVmNDg1NzQxODZiZjk5OGVkMGNiN2E3Y2JjZWRiNzM5NzQ4ZDZiOGRhY2MyNTM3ZjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uN3hF5LSYuHgmGi9hRYLJcAHjKz9-vQj2zpBLfWkfbA)
> 
> ~On a side note, does anyone know if this device is also triggered in case of a water leak? If yes, it would be nice to get someone with the water sensor to sniff the communication as well.~ Just checked this. The answer is no.

Does the zone status need to returned anywhere? Like in a converter, convertSet, convertGet? What do you suggest.

vjonas, 2023-11-26:
> I would love to give it a try, I have a fresh BSD2 paired in my test environment. I am still figuring out how I can test your commands in a local converter, maybe someone else has experience with [...]

I am in the process of adding this functionality. It already works sending the command through z2m in my local environment. Will come back with updates once I figured out the IAS zone cluster notifications.

Mikkooooo, 2023-11-28:
> I am in the process of adding this functionality. It already works sending the command through z2m in my local environment. Will come back with updates once I figured out the IAS zone cluster notifications.

Hi Jonas,
have you already made progress?

safakaltun, 2023-11-28:
> I am in the process of adding this functionality. It already works sending the command through z2m in my local environment. Will come back with updates once I figured out the IAS zone cluster [...]

I see that you already commented in https://github.com/Koenkk/zigbee2mqtt/issues/13520. I don’t know if you had a look at what @danieledwardgeorgehitchcock did there, but you might be able to follow his work to figure out how to handle the ias zone notifications. If I remember correct, the outdoor siren was also sending ias zone notifications to report tamper status.

Mikkooooo, 2023-11-29:
> I did some sniffing with multiple devices. Here are the commands that needs to be sent to the Bosch Smoke Alarm II to trigger certain actions. I haven't touched z2m yet. I don't have the time to implement this in the near future. If anyone else has the time to give it a go, please be my guest.
> 
> Action Cluster Command Data Comment Relayed Smoke Alarm On IAS Zone 0x0500 0x80 003c Relayed Smoke Alarm Off IAS Zone 0x0500 0x80 0000 Relayed Burglar Alarm On IAS Zone 0x0500 0x80 01b4 Relayed Burglar Alarm Off IAS Zone 0x0500 0x80 0100 Self test IAS Zone 0x0500 0x02 - Already a part of the clusters.ts, so should be easyto implement

Hi, I‘ve tried to send a command to Bosch Smoke Alarm II to trigger Burglar Alarm On.
I‘ve got the log message :
zigbee2mqtt/0x80455eaffed123b2f/set, payload={ "Cluster": "0x0500", "Command": "0x80", "Data": "01b4" }

and then comes the log message
Z2M Server: No converter available for 'Data' ("0100")
11.29.2023 21:12:55
ZigbeePlus
Error
Z2M Server: No converter available for 'Command' ("0x80")
11.29.2023 21:12:55
ZigbeePlus
Error
Z2M Server: No converter available for 'Cluster' ("0x0500")

and the  Burglar Alarm  On is not triggered.

What‘s going wrong ?

vjonas, 2023-12-01:
> > I did some sniffing with multiple devices. Here are the commands that needs to be sent to the Bosch Smoke Alarm II to trigger certain actions. I haven't touched z2m yet. I don't have the time to implement this in the near future. If anyone else has the time to give it a go, please be my guest.
> 
> > 
> 
> > Action Cluster Command Data Comment 
> 
> Relayed Smoke Alarm On IAS Zone 0x0500 0x80 003c 
> 
> Relayed Smoke Alarm Off IAS Zone 0x0500 0x80 0000 
> 
> Relayed Burglar Alarm On IAS Zone 0x0500 0x80 01b4 
> 
> Relayed Burglar Alarm Off IAS Zone 0x0500 0x80 0100 
> 
> Self test IAS Zone 0x0500 0x02 - Already a part of the clusters.ts, so should be easyto implement
> 
> 
> 
> Hi, I‘ve tried to send a command to Bosch Smoke Alarm II to trigger Burglar Alarm On.
> 
> I‘ve got the log message :
> 
> zigbee2mqtt/0x80455eaffed123b2f/set, payload={ "Cluster": "0x0500", "Command": "0x80", "Data": "01b4" }
> 
> 
> 
> and then comes the log message
> 
> Z2M Server: No converter available for 'Data' ("01b4")
> 
> 11.29.2023 21:12:55
> 
> ZigbeePlus
> 
> Error
> 
> Z2M Server: No converter available for 'Command' ("0x80")
> 
> 11.29.2023 21:12:55
> 
> ZigbeePlus
> 
> Error
> 
> Z2M Server: No converter available for 'Cluster' ("0x0500")
> 
> 
> 
> and the  Burglar Alarm  On is not triggered.
> 
> 
> 
> What‘s going wrong ?
> 
> 
> 
> 

It is not supported yet to send these commands to the bsd-2 that is why we must make changes in te repositories to add that functionality. 
I'm a bit swamped currently and will continue on it asap. Keep you guys posted here. 
If you are really eager and are comfortable installing manually: I forked the repo and it's able to send the burglar alarm. 
You can use these repo's already before it's officially released. They should be public. 
You'll need zigbee-herdsman and the zigbee-herdsman-converters.

Mikkooooo, 2023-12-28:
Are there any updates regarding this ?

vjonas, 2023-12-28:
I'm having a bit more time now in xmas holidays and will try to wrap it up and create pull requests.

vjonas, 2024-01-11:
Being processed:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6862

Sneak peak of implementation: 
[image]

vjonas, 2024-01-15:
Merged in master. You can pull the the branch or wait for the docker release.
Can be closed.

firstchair, 2024-01-16:
Thanks @vjonas!!

vjonas, 2024-01-16:
> Thanks @vjonas!!

With pleasure!
Please use the **dev** branch for now.
It will get merged to master soon!