Koenkk/zigbee2mqtt #17231: Philips Hue White and Color Ambiance 9290024688

hellcry37, 2023-04-03:
### What happened?

New Philips Hue White and Color Ambiance 1100
Model : 9290024688

As per my knowledge device works correctly and everything is ok except model shown in the GUI is wrong.

### What did you expect to happen?

Correct model is: 9290024688 (e27 bulb)
Model shown in the GUI is: 9290024689 (b22 bulb)

### Zigbee2MQTT version

1.30.3 commit: 24c6b2e

### Adapter firmware version

20220219

### Adapter

zig-a-zig-ah!

Koenkk, 2023-04-03:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

goddenrich, 2024-03-14:
Hi I have the 9290024689 (b22 bulb) and would like to add it how do I go about doing so? I guess the line 
```
zigbeeModel: ['LCA006'],
```
was not corrrect for the 9290024689 (b22 bulb) model. How do I find that correct model?

goddenrich, 2024-03-14:
Ah I see its that the same device LCA006 covers 2 models one with the bayonette and one with the ES fitting.