Koenkk/zigbee2mqtt #17239: Tuya TS0052 unsupported

kevindegrootgrave, 2023-04-04:
### Link

https://nl.aliexpress.com/item/1005005197915510.html?spm=a2g0o.order_list.order_list_main.4.187579d21vNnWI&gatewayAdapt=glo2nld

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0xa4c138eeb3884a4e","nwkAddr":42491,"manufId":4417,"manufName":"_TZ3000_qyo2yzwk","powerSource":"Mains (single phase)","modelId":"TS0052","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,8,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"switchType":0}},"genLevelCtrl":{"attributes":{"currentLevel":204,"minLevel":2,"maxLevel":254,"onOffTransitionTime":50}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1680604760495,"defaultSendRequestWhen":"immediate"}

### Comments

Hi all,

I just received a Tuya 1 channel Zigbee dimmer module. Z2M doesn't recognize the device. I really really really tried to go through the 'support new devices' page and tried adding an external converter file, but I'm stuck and my knowledge of dpValues and datapoints is sadly insufficient. Can someone please help me? 

According to Z2M the model is a TS0052 and the manufacturer name is TZ3000_qyo2yzwk. I think it's a new version of the TuYa TS1101_dimmer_module_1ch which is supported (they look exactly similar).

Koenkk, 2023-04-04:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: 'TS0052',
    model: 'TS0052',
    vendor: 'TuYa',
    description: 'Zigbee dimmer module 1 channel',
    extend: tuya.extend.light_onoff_brightness({minBrightness: true}),
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

kevindegrootgrave, 2023-04-04:
Hi Koen,

Thanks for your response. Unfortunately it still says the device isn't supported. Don't know if it helps, but this is what the log says:

Info 2023-04-04 20:21:08Zigbee: allowing new devices to join.
Info 2023-04-04 20:21:08MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"o5e5b-1"}'
Info 2023-04-04 20:21:13Device '0xa4c138eeb3884a4e' joined
Info 2023-04-04 20:21:13MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138eeb3884a4e","ieee_address":"0xa4c138eeb3884a4e"},"type":"device_joined"}'
Info 2023-04-04 20:21:13MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138eeb3884a4e"},"type":"device_connected"}'
Info 2023-04-04 20:21:13MQTT publish: topic 'zigbee2mqtt/0xa4c138eeb3884a4e/availability', payload 'online'
Info 2023-04-04 20:21:13Starting interview of '0xa4c138eeb3884a4e'
Info 2023-04-04 20:21:13MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138eeb3884a4e","ieee_address":"0xa4c138eeb3884a4e","status":"started"},"type":"device_interview"}'
Info 2023-04-04 20:21:13MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138eeb3884a4e"},"type":"pairing"}'
Info 2023-04-04 20:22:14Successfully interviewed '0xa4c138eeb3884a4e', device has successfully been paired
Warning 2023-04-04 20:22:14Device '0xa4c138eeb3884a4e' with Zigbee model 'TS0052' and manufacturer name '_TZ3000_qyo2yzwk' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-04-04 20:22:14MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138eeb3884a4e","ieee_address":"0xa4c138eeb3884a4e","status":"successful","supported":false},"type":"device_interview"}'
Info 2023-04-04 20:22:14MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xa4c138eeb3884a4e","supported":false},"type":"pairing"}'

Koenkk, 2023-04-05:
Did you add it to your configuration.yaml correctly? (instructions from https://github.com/Koenkk/zigbee2mqtt/issues/17239#issuecomment-1496374000)

meganiekje, 2023-04-05:
@kevindegrootgrave : When I was struggeling with an external converter, I noticed that adding the ext_converter.js by hand to the configfiguration.yaml, did not work. When I used the zigbee2mqtt UI : settings, external converter, to add the ext_converter.js, it did work !

kevindegrootgrave, 2023-04-05:
Thanks for the suggestion @meganiekje. I tried removing ext_converter.js from configuration.yaml and adding it through the UI, but unfortunately still no luck.

@Koenkk I think I did everything the right way. Here are some screenshots:

[image]

[image]

meganiekje, 2023-04-05:
@kevindegrootgrave : does the ext_converter.js show up in the zigbee2mqtt UI like this:

[image]

kevindegrootgrave, 2023-04-05:
Yes it does

[image]

meganiekje, 2023-04-05:
Is the ext_converter.js in the same folder as the configuration.yaml ?
Then restart zigbee2mqtt,  and re-pair the device...

kevindegrootgrave, 2023-04-05:
@meganiekje Yes, it's in the same folder (visible in screenshots). I already tried restarting Z2M and re-pairing the device numerous times, but I keep getting the same 'Device NOT supported' error.

meganiekje, 2023-04-05:
Maybe you need to insert this into the ext_converter.js:

ManufacturerName: '_TZ3000_qyo2yzwk',

If that does not work, I can not help you.

kevindegrootgrave, 2023-04-05:
Added ManufacturerName: '_TZ3000_qyo2yzwk' to ext_converter.js file, but still doesn't work unfortunately. Thanks for your help @meganiekje

Koenkk, 2023-04-06:
I made a mistake, updated https://github.com/Koenkk/zigbee2mqtt/issues/17239#issuecomment-1496374000

kevindegrootgrave, 2023-04-06:
It worked!! Thank you so much for your help, coffee has been bought!

Koenkk, 2023-04-08:
Great, added the device. You can remove the external converter after updating to the next release.

closed 2023-04-08 by Koenkk

kevindegrootgrave, 2023-04-08:
Thanks Koen. FYI: the min_brightness attribute does not work (Status 'UNSUPPORTED_ATTRIBUTE')

Koenkk, 2023-04-09:
@kevindegrootgrave thanks, removed it.

fredqm, 2023-04-25:
In the Tuya app I'm able to set min and max brightness.
These are the IDs i pulled from the Tuya Developer Site, if it's any help.
Function		DP ID
Switch Led 1		1
Bright Value 1		2
Brightnes Min 1		3
Led Type 1		4
Brightnes Max 1		5
Countdown 1		6
Relay Status		14
Switch Type Setting	47

kevindegrootgrave, 2023-04-26:
Thanks @fredqm 

If it would be possible to implement the min_brightness attribute in a different way, that would be great. The light connected to my dimmer module now stays off at any set value below 40%. Not a huge deal, but would be the icing on the cake if that could be adjusted.

artem-zinnatullin, 2023-05-02:
Hey folks,

1) I was just mounting `TS0052` today and faced unsupported device message, was like eh gotta have to send patch and then I saw a Zigbee2Mqtt 1.30.4 released few hours ago adding support for `TS0052`, lucky timing, thanks for your work!

[image]

[image]

😅 

2) There is an issue with transition time between switch/dimmer states

When I try to use "Light Turn Off" in HomeAssistant with `TS0052` and any transition time, dimmer seems to react instantly without respect for the transition time and what in fact happens is that it sets brigtness to 1 and stays ON!

[image]

I've noticed in the logs and attributes that dimmer has `on_off_transition_time` in payload which in my case is set to `50` and is not related to HomeAssistant transition time nor if I change "Settings (specific)" → "transition" to any value like 10:

- Dimmer does not respect that value
- "State" still shows `"on_off_transition_time": 50`

EricPalmquist, 2023-05-03:
> 

> Thanks @fredqm
> 
> If it would be possible to implement the min_brightness attribute in a different way, that would be great. The light connected to my dimmer module now stays off at any set value below 40%. Not a huge deal, but would be the icing on the cake if that could be adjusted.

I see the same thing, but at a numeric value of 40 which is 18%.

Laboe3000, 2023-06-01:
Hi there. I have the same device and I am wondering whether there is any option to change the switch type. No need to do this via mqtt as possible with the TS110E. If there is a trick like pushing the physical button three times (does not work), I'd be just a happy. Would love to use it in momentary mode. Thank you for any help

boojew, 2023-12-18:
I just got this dimmer and I have the same issue. Did anyone ever find a solution?

Dunkelschunkel, 2024-01-03:
I got some of those Relais a week ago. It is recognized and shows some controls. DND is throwing errors when toggling. Switching it of does sometimes only toggle between high and low brightness. Even if there is no minimum and maximum settings, just a normal brightness slider.

eduardorgos, 2024-02-02:
Hi all,

I've been using the external converter shared in this discussion, it was working fine (both channels were exposed and dimmer was working too), but after updating to Zigbee2MQTT 1.35.2 the external converter script does not work anymore.
Seems like the following method is not defined `tuya.extend.light_onoff_brightness()` and checking the documentation points that TuYa devices has to be defined quite different compared to the old way.

Alsop tried disabling the external converter and pairing the dimmer again. Whilst device are reconized it keeps failing to recognize the 2 channels.

boojew, 2024-02-02:
I had a similar issue. I removed the external converters and the old issues came back- where the light never reports off

eduardorgos, 2024-02-02:
I've posted a possible solution in the https://github.com/Koenkk/zigbee2mqtt/issues/18535 discussion if someone want's to give it a try.

boojew, 2024-02-04:
This works to discover the 2nd channel, but they report 1 or 2% when turned off.