Koenkk/zigbee2mqtt #17240: Tuya ZS06 in Z2M as Tuya TS1201

Jos-77, 2023-04-04:
### What happened?

When adding a new tuya ZS06 is appears in Z2M as Tuya TS1201.

On the device it says it is a ZS06. Also the chip is the same as the ZS06

I have tried to reconneced but no diverence.

### Zigbee2MQTT version

1.18.1

### Adapter firmware version

 20210708

### Adapter

Tuya ZS06

### Debug log

Successfully interviewed '0x70ac08fffe34d4e1', device has successfully been paired
warn  2023-04-04 16:11:27: Device '0x70ac08fffe34d4e1' with Zigbee model 'TS1201' and manufacturer name '_TZ3290_7v1k4vufotpowp9z' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Koenkk, 2023-04-04:
You are using an outdated z2m version, update to 1.30.3.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

Jos-77, 2023-04-04:
I'm us8ng the home assistant intergration. That should auto update,

Jos-77, 2023-04-05:
fixed, in home assistant there was an old repro.