Koenkk/zigbee2mqtt #17249: Sonoff zigbee 3.0 usb dongle plus (P) Error: network commissioning timed out.

booua, 2023-04-04:
### What happened?

Hi :) 
Starting off - i think this project is amazing. I've been playing around with it using 2 basic Aqara motion sensors and a DIN rail mounted energy/power meter. It worked really well using Sonoff bridge with Tasmota and MQTT in HASS hosted on virtualbox.
Sadly I've experienced some inconsistencies with the bridge, so I decided to swap it for better and supposedly easier to use Sonoff zigbee 3.0 usb dongle plus (the P version). I've set it up so HASS.io sees it no problem, it came preflashed, but I reflashed it according to the manual with the latest firmware and for a whole day I've been stuck on this `network commissioning timed out.` error. 
Here's a log:

```
[21:39:42] INFO: Preparing to start...
[21:39:44] INFO: Socat not enabled
[21:39:47] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2023-04-04 21:40:06: Logging to console and directory: '/config/zigbee2mqtt/log/2023-04-04.21-40-06' filename: log.txt
Zigbee2MQTT:info  2023-04-04 21:40:06: Starting Zigbee2MQTT version 1.30.3 (commit #unknown)
Zigbee2MQTT:info  2023-04-04 21:40:06: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:error 2023-04-04 21:41:15: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-04-04 21:41:15: Failed to start zigbee
Zigbee2MQTT:error 2023-04-04 21:41:15: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-04-04 21:41:15: Exiting...
Zigbee2MQTT:error 2023-04-04 21:41:16: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

Here's my config:

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  user: mqtt-user
  password: ****
  server: mqtt://ip-address:port
serial:
  port: /dev/ttyUSB0
advanced:
  pan_id: GENERATE
  ext_pan_id:
    - 11
    - 21
    - 31
    - 41
    - 51
    - 61
    - 71
    - 81
  network_key: GENERATE
```

And pretty much I'm lost there.

I had the ZHA running, but I disabled and uninstalled it. I had the autodiscovery turned off in the config. It's still crashing on me.

### What did you expect to happen?

I've expected z2m to start up properly.

### Zigbee2MQTT version

1.30.3-1

### Adapter firmware version

20221226

### Adapter

sonoff zigbee 3.0 usb dongle plus P

### Debug log

[21:39:42] INFO: Preparing to start...
[21:39:44] INFO: Socat not enabled
[21:39:47] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2023-04-04 21:40:06: Logging to console and directory: '/config/zigbee2mqtt/log/2023-04-04.21-40-06' filename: log.txt
Zigbee2MQTT:info  2023-04-04 21:40:06: Starting Zigbee2MQTT version 1.30.3 (commit #unknown)
Zigbee2MQTT:info  2023-04-04 21:40:06: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:error 2023-04-04 21:41:15: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-04-04 21:41:15: Failed to start zigbee
Zigbee2MQTT:error 2023-04-04 21:41:15: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-04-04 21:41:15: Exiting...
Zigbee2MQTT:error 2023-04-04 21:41:16: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

Koenkk, 2023-04-05:
Probably a Zigbee network is already running on the same channel/panid. Try using a different channel, e.g.:

```yaml
advanced:
  channel: 12
```

booua, 2023-04-06:
> Probably a Zigbee network is already running on the same channel/panid. Try using a different channel, e.g.:
> 
> ```yaml
> advanced:
>   channel: 12
> ```

Sadly still the same thing. I tried multiple channels - 11, 12, 25... 
still nothing :/

Koenkk, 2023-04-06:
Then there is probably same heavy interference (check antenna and make sure you are using an usb extension cable) or you are using a pi with bad power supply.

booua, 2023-04-06:
I will definitely try that, thank you :) I'll report my findings

booua, 2023-04-07:
Okay. It seems that nothing helps. Tried moving it away from potential sources of interference, extnsion cables, turning off the devices around. Nothing. I will blame that on potentially defective stick :/

Rick100694, 2023-04-07:
I have the same problem and have been stuck on it for several days. I also used a USB extension cable and set the panid to generate and changed the channel. nevertheless I still have no success. I have another idea for you, namely to use the complete device path id for the port (/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0) You can also find this under System/Hardware. does anyone have a solution? it works with the Conbee 2...but I would like to use the Sonoff stick.

triggerx, 2023-04-10:
Having the same issue on a new z2m setup with a sonoff dongle P.  Interesting to see people with the same exact error all within the past few days... which tells me I'm not alone.  Perhaps a wider problem?  Hoping to find a solution as well.  Tested ZHA and it works fine with this stick.

booua, 2023-04-10:
@triggerx @Rick100694 @Koenkk I've managed to "fix" the issue. Or at least start Z2M properly. 
Generally I've been following this guide: https://www.diyenjoying.com/2022/11/05/home-assistant-sonoff-zigbee/
Here's my part of the config I've changed:
```
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
  rtscts: false
  disable_led: true
advanced:
  pan_id: GENERATE
  network_key: GENERATE
  channel: 25
  transmit_power: 15
```
BUT the most influencial tip I've encountered was:
Prior to setting up the Z2M, and inserting it to the HASS server usb port, remove antenna from the dongle and wrap it in aluminium foil. Then insert it to the usb port, start Z2M and in my case it magically started.
Then I removed the foil, added the antenna and tested if restarting the Z2M or even a whole server would break things. It still works, but it's been 1-2hours maybe, so I'll let it run for a week and give my update.

Hope this helps anyone.

bikeymouse, 2023-05-12:
I have the same error message during start up `(" Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby" )`, when trying to use a new Sonoff Zigbe 2.0 dongle plus P to work with my system.
I prepared the new one by flashing it with the new firmware  and writing the same secondary IEEE-address to it as the the the old stick has ([as described in the FAQ](https://www.zigbee2mqtt.io/guide/adapters/flashing/copy_ieeaddr.html)).
 
I don't know how to verify that this was successful, because as Zigbee2Mqqt doesn't start-up I can't get to the GUI.

Would love to hear of others having done this successfully.

closed 2023-06-20 by github-actions[bot]

ivorsmorenburg, 2023-10-27:
Same issue here any solutions?