Koenkk/zigbee2mqtt #17250: TuYa TS130F Calibration not working

teunaldo, 2023-04-05:
### What happened?

The gate only opens only to roundabout 85%. In Z2M it says the gate is fully opened.

The closure of the gate doesn't stop. When I stop it at the place I want, Z2M says it is opened for 14%.

When I try this again after some times
The gate only opens only to roundabout 75%. In Z2M it says the gate is fully opened.

The closure of the gate doesn't stop. When I stop it at the place I want, Z2M says it is opened for 25%.

### What did you expect to happen?

When I press OPEN in state, the gate should completely open.

When I press CLOSE in state, the gate should close to the point I have set up.

### How to reproduce it (minimal and precise)

The device I used is a Loratap Curtain Switch 2nd Generation module
https://nl.aliexpress.com/item/1005003123388237.html?spm=a2g0o.order_list.order_list_main.27.28a579d2KcWLFG&gatewayAdapt=glo2nld

As the documentation described I did the following:
**Press the open button on the switch, wait until the curtains are completely open.**
The gate was opened completely. 
**Press the pause button on the switch.**
I pressed the STOP button in Z2M state
**Put the device into calibration mode, see [below](https://www.zigbee2mqtt.io/devices/TS130F.html#calibration-binary).**
I set the calibration to "ON".
I have done this 2 ways:
- MQTT publish
- Enable option in Z2M
**Press the close button on the switch, wait until curtains are fully closed.**
I closed the gate to the lowest position I want.
**Press the pause button on the switch.**
I pressed the STOP button in Z2M state
Disable the calibration mode, see [below](https://www.zigbee2mqtt.io/devices/TS130F.html#calibration-binary)
I set the calibration to "OFF".
I have done this 2 ways:
- MQTT publish
- Enable option in Z2M

### Zigbee2MQTT version

1.30.3-1

### Adapter firmware version

20221226

### Adapter

Slaesh cc2652-stick 

### Debug log

debug 2023-04-05 10:41:07Received MQTT message on 'zigbee2mqtt/0xa4c138b658803b99/set' with data '{"state":"OPEN"}'
debug 2023-04-05 10:41:07Publishing 'set' 'state' to '0xa4c138b658803b99'
debug 2023-04-05 10:41:07Received Zigbee message from '0xa4c138b658803b99', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":14,"tuyaMovingState":0}' from endpoint 1 with groupID 0
info 2023-04-05 10:41:07MQTT publish: topic 'zigbee2mqtt/0xa4c138b658803b99', payload '{"backlight_mode":"high","calibration":"OFF","calibration_time":16.2,"linkquality":65,"motor_reversal":"OFF","moving":"UP","position":14,"state":"OPEN"}'
info 2023-04-05 10:41:07MQTT publish: topic 'zigbee2mqtt/0xa4c138b658803b99/mode', payload '{"backlight":"high"}'
debug 2023-04-05 10:41:22Received Zigbee message from '0xa4c138b658803b99', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100,"tuyaMovingState":1}' from endpoint 1 with groupID 0
info 2023-04-05 10:41:22MQTT publish: topic 'zigbee2mqtt/0xa4c138b658803b99', payload '{"backlight_mode":"high","calibration":"OFF","calibration_time":16.2,"linkquality":65,"motor_reversal":"OFF","moving":"STOP","position":100,"state":"OPEN"}'
info 2023-04-05 10:41:22MQTT publish: topic 'zigbee2mqtt/0xa4c138b658803b99/mode', payload '{"backlight":"high"}'
debug 2023-04-05 10:41:33Received MQTT message on 'zigbee2mqtt/0xa4c138b658803b99/set' with data '{"state":"CLOSE"}'
debug 2023-04-05 10:41:33Publishing 'set' 'state' to '0xa4c138b658803b99'
debug 2023-04-05 10:41:33Received Zigbee message from '0xa4c138b658803b99', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100,"tuyaMovingState":2}' from endpoint 1 with groupID 0
info 2023-04-05 10:41:33MQTT publish: topic 'zigbee2mqtt/0xa4c138b658803b99', payload '{"backlight_mode":"high","calibration":"OFF","calibration_time":16.2,"linkquality":65,"motor_reversal":"OFF","moving":"DOWN","position":100,"state":"OPEN"}'
info 2023-04-05 10:41:33MQTT publish: topic 'zigbee2mqtt/0xa4c138b658803b99/mode', payload '{"backlight":"high"}'
debug 2023-04-05 10:41:45Received MQTT message on 'zigbee2mqtt/0xa4c138b658803b99/set' with data '{"state":"STOP"}'
debug 2023-04-05 10:41:45Publishing 'set' 'state' to '0xa4c138b658803b99'
debug 2023-04-05 10:41:45Received Zigbee message from '0xa4c138b658803b99', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":25,"tuyaMovingState":1}' from endpoint 1 with groupID 0
info 2023-04-05 10:41:45MQTT publish: topic 'zigbee2mqtt/0xa4c138b658803b99', payload '{"backlight_mode":"high","calibration":"OFF","calibration_time":16.2,"linkquality":65,"motor_reversal":"OFF","moving":"STOP","position":25,"state":"OPEN"}'
info 2023-04-05 10:41:45MQTT publish: topic 'zigbee2mqtt/0xa4c138b658803b99/mode', payload '{"backlight":"high"}'

closed 2023-05-13 by github-actions[bot]

stryna90, 2023-06-21:
@teunaldo Did you maybe found solution for your problem? I have exactly the same issue

RECOIL101, 2023-06-22:
I have the same problem. The blind does not open to the final position, but is approx. 5-10% closed. State value says open (100%). Has anyone found a solution to this problem?

teunaldo, 2023-06-22:
Nope, haven’t found a solution.

platte-73, 2023-06-24:
I have also this problem. Only opens for 85%. When I push after  opening on the switch itself it will open 100%.

Hope someone has a fix for this

platte-73, 2023-06-24:
Found a fix. my calibration time was 26.4 s and that was not enough to close to 100%

so this is wat I did

From the Dev console, if I select endpoint=1, cluster=Closures and moesCalibrationTime attribute and change their manual value to adjust to my needs. Everything is OK.
Changed value to 29s and now it is closing completly

RECOIL101, 2023-06-25:
> Found a fix. my calibration time was 26.4 s and that was not enough to close to 100%
> 
> so this is wat I did
> 
> From the Dev console, Select endpoint=1, cluster=Closures and moesCalibrationTime attribute and change their manual value to adjust to my needs. Everything is OK. Changed value to 29s and now it is closing completly

Thx for the info. 

Saddly this is not working for me. I've changed calibration_time value but still blinds but the blinds still do not close completely. Maybe someone knows if the switch can be reset to factory default settings?

platte-73, 2023-06-25:
> Thx for the info.
>
> Saddly this is not working for me. I've changed calibration_time value but still blinds but the blinds still do not close completely. Maybe someone knows if the switch can be [...]

What kind of blinds are you using with the switch. Are you sure the motor is not stopping the blind instead of the switch.

finipini, 2023-10-05:
My problem with this device is that even after calibration, if the shutter is fully opened ( 100 % in z2m) and if i click the open button it will close the open relay again for the time setted in calibration time.

In other words : every time i click up or down ( in z2m or local buttons) it will always close the respective relay for the time set in calibration-time, no mather what position it is.  

Of course the shutter have it,s own switchs to stop in the 2 ends, but this should not work this way.

TobiasB91, 2024-02-19:
I have the same problem...:(

RECOIL101, 2024-03-19:
Maybe someone will find this solution for incomplete closure of the blinds usefull. This is how I've solved this problem. 

```
- id: b54984cd-9946-436b-86d2-cd91671100e6
  alias: If cover blinds right open
  trigger:
    - platform: state
      entity_id: sensor.blinds_right_moving
      from: 'UP'
      to: 'STOP'
  condition:
    condition: and
    conditions:  
    - condition: numeric_state
      entity_id: sensor.blinds_right_position
      above: 99
    - condition: state
      entity_id: input_boolean.right_blind_helper
      state: 'on'      
  action:
# bedno TUYA stikalo - zaradi razlike v času dviganja in spuščanja se rolete ne zaprejo 100%. Popravljam pozicijo "ročno".     
    - service: mqtt.publish
      data:
        topic: "zigbee2mqtt/blinds right/1/set"
        payload: '{"write":{"cluster":"closuresWindowCovering","options":{},"payload":{"currentPositionLiftPercentage": 80}}}'
    - delay: '00:00:01'
    - service: cover.open_cover
      entity_id: cover.blinds_right
    - service: input_boolean.turn_off
      entity_id: input_boolean.right_blind_helper      
  mode: single
```

diverudi, 2024-06-28:
> Thx for the info.
>
> Saddly this is not working for me. I've changed calibration_time value but still blinds but the blinds still do not close completely. Maybe someone knows if the switch can be [...]

Hello all,

if you are interested, I could solve the problem for me (this switch has two calibration values, one for open and one for close)

I did the calibration and then write the same value manually:
From the Dev console, Select endpoint=2, cluster=Closures and moesCalibrationTime attribute and change their manual value to adjust to my needs. Everything is OK. 

Please take care on the endpoint! " endpoint=2"

fets13, 2024-06-30:
@diverudi  what is your model?

diverudi, 2024-06-30:
In Z4M it shows "Tuya TS130F" 
on aliexpress it is LoraTap 2nd generation "Tuya Smart Leben ZigBee 3,0"

aldifesu, 2024-09-28:
> Hello all,
>
> if you are interested, I could solve the problem for me (this switch has two calibration values, one for open and one for close)
>
> I did the calibration and then write the same value [...]

Hello, I also have the LoraTap 2nd generation.
 What do you mean is that the rise/open time is set for endpoint 1 and the time it takes to go down/close is set for endpoint 2?
 Or do you set the same time for endpoint1 and endpoint2?

diverudi, 2024-09-28:
> @diverudi Hello, I also have the LoraTap 2nd generation. What do you mean is that the rise/open time is set for endpoint 1 and the time it takes to go down/close is set for endpoint 2? Or do you set the same time for endpoint1 and endpoint2?

@aldifesu I set the same value in endpoint 2 as in endpoint 1, so the same time in both entries works well for my devices.

joelsteinbiss, 2024-12-11:
> @aldifesu I set the same value in endpoint 2 as in endpoint 1, so the same time in both entries works well for my devices.

I also habe the LoraTap 2nd Gen, but I have a problem writing to endpoint 2. Maybe u can help me here?

This is the code, which gives me an error. Writing to endpoint 1 works fine.
`action: zha.set_zigbee_cluster_attribute
data:
  ieee: a4:c1:38:95:5b:9d:b6:56
  endpoint_id: 2
  cluster_id: 258
  attribute: 61443
  value: "212"
  cluster_type: in`
[image]

diverudi, 2024-12-14:
@joelsteinbiss I'm sorry, but with your error message it looks like you are using ZHA, but I use Zigbee2MQTT and here 
it is easy then you go to Developer Console and here select the right entry. For ZHA I don't know if the device will work well.

Suschibomber, 2025-08-15:
Hey, I have a problem: I don't see a calibration button. What do I have to do to get it displayed in Z2MQTT? 
[image]