Koenkk/zigbee2mqtt #17261: z2m addon won't start after Update HA 2023.4

mex1254, 2023-04-06:
### What happened?

After updateing to HA 2023.4.0 and installing Silicon Labs Multiprotocol 1.1.1 z2m add-on wont start anymore. I am running HA as an VirtualMachine on a Synology NAS. I can see the stick in the hardware section.

[12:35:16] INFO: Preparing to start...
[12:35:17] INFO: Socat not enabled
[12:35:19] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-04-06 12:35:27: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-04-06 12:35:27: Logging to console and directory: '/config/zigbee2mqtt/log/2023-04-06.12-35-27' filename: log.txt
Zigbee2MQTT:debug 2023-04-06 12:35:27: Removing old log directory '/config/zigbee2mqtt/log/2023-04-06.11-10-45'
Zigbee2MQTT:info  2023-04-06 12:35:27: Starting Zigbee2MQTT version 1.30.3 (commit #unknown)
Zigbee2MQTT:info  2023-04-06 12:35:27: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:debug 2023-04-06 12:35:27: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_78892a05e842ed11962643ab2a61ed69-if00-port0"}}'
Zigbee2MQTT:error 2023-04-06 12:35:48: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-04-06 12:35:48: Failed to start zigbee
Zigbee2MQTT:error 2023-04-06 12:35:48: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-04-06 12:35:48: Exiting...
Zigbee2MQTT:error 2023-04-06 12:35:49: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:540:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:338:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

### What did you expect to happen?

it is working after update - never had such issues before

### How to reproduce it (minimal and precise)

Even when restoring a previus backup, z2m isnt't starting anymore

Things i already tried:

- unplug/plug stick
- reboot
- reconnect stick

### Zigbee2MQTT version

1.30.3-1

### Adapter firmware version

1.0.0 i think cannot figure out right now

### Adapter

SkyConnect Stick

### Debug log

[13:07:55] INFO: Preparing to start...
[13:07:56] INFO: Socat not enabled
[13:07:56] INFO: Zigbee Herdsman debug logging enabled
[13:07:58] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-04-06 13:08:06: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-04-06 13:08:06: Logging to console and directory: '/config/zigbee2mqtt/log/2023-04-06.13-08-06' filename: log.txt
Zigbee2MQTT:debug 2023-04-06 13:08:06: Removing old log directory '/config/zigbee2mqtt/log/2023-04-06.11-12-06'
Zigbee2MQTT:info  2023-04-06 13:08:06: Starting Zigbee2MQTT version 1.30.3 (commit #unknown)
Zigbee2MQTT:info  2023-04-06 13:08:06: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:debug 2023-04-06 13:08:06: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_78892a05e842ed11962643ab2a61ed69-if00-port0"}}'
2023-04-06T11:08:07.392Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-04-06T11:08:07.396Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_78892a05e842ed11962643ab2a61ed69-if00-port0","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-04-06T11:08:07.402Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_78892a05e842ed11962643ab2a61ed69-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-04-06T11:08:07.515Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 1 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
[... 10 lines cut ...]
2023-04-06T11:08:17.549Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 3 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:76:22)
    at /app/node_modules/@serialport/stream/dist/index.js:112:18
2023-04-06T11:08:22.551Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 4
2023-04-06T11:08:22.552Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_78892a05e842ed11962643ab2a61ed69-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-04-06T11:08:22.562Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 4 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:76:22)
    at /app/node_modules/@serialport/stream/dist/index.js:112:18
2023-04-06T11:08:27.566Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 5
2023-04-06T11:08:27.568Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4}
Zigbee2MQTT:error 2023-04-06 13:08:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-04-06 13:08:27: Failed to start zigbee
Zigbee2MQTT:error 2023-04-06 13:08:27: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-04-06 13:08:27: Exiting...
Zigbee2MQTT:error 2023-04-06 13:08:27: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:540:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:338:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

olgru, 2023-04-06:
I'm not an expert in all this, but I think the error line is quite clear:

2023-04-06T11:08:17.538Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_78892a05e842ed11962643ab2a61ed69-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-04-06T11:08:17.549Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 3 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'

Cannot lock port.
Thinking about synology, it's running a linux based system. It means that anything is a filesystem entry. Not being able to lock the port would me trigger to check:
- There is another application using an blocking it
- You don't have sufficent access rights to access the port and create a lock file on your own (managing port access is on privilege rights) 
- As you running in a VM, maybe the hostOS or another VM is blocking as well
- There is an old lock file which has not been removed

But I don't think that this is on Z2M... Not without being sure that none of the above is the reason....
There is also a FAQ about:
https://www.zigbee2mqtt.io/information/FAQ.html#error-resource-temporarily-unavailable-cannot-lock-port

mex1254, 2023-04-07:
That pushed me in an other direction - thank you.

I did some further investigation and found this thread

https://github.com/home-assistant/addons/issues/2901

maybe a solution can be found there

m0wlheld, 2023-04-07:
Better file this issue at https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#installation, as Z2M 13.0.3 standalone with Home Assistant 2023.4(.1) works like a charm.

olgru, 2023-04-09:
Good hint-  but first time I see that there is something special for Home-Assistant related to Z2M. Usually I click on the link inside z2m and this leads to this page...

mex1254, 2023-04-09:
i found a workaround for me. i flashed back the EmberZ Firmware manually with the the homepage 

https://skyconnect.home-assistant.io/firmware-update/

. Started z2m without Silicon Labs Multiprotocol addon - and all went back to normal. 
I tried to reinstall the addon and flashed back the FW that comes with the plugin, but same behavior.

For now i dont need any Multiprotocol stuff so I will stay with this solution for now - will give it another try when i have the first devices that use matter...

thank you all for your advices

Mrsash2020, 2023-04-09:
Every time there is an update it stops working. I physically remove it from the PC and plug it back in. Then I restart HA. Give it a couple of minutes and settles eventually.

closed 2023-05-17 by github-actions[bot]