Koenkk/zigbee2mqtt #17271: 2CH curtain module not showing second channel

draggero, 2023-04-07:
### What happened?

Z2M not recognized second channel in module. In ZHA showing it properly.
Z2M recognizing module as TS130F
[image]

### What did you expect to happen?

I Want to control each channel separately.

### How to reproduce it (minimal and precise)

Just go to configuration :)
[image]

### Zigbee2MQTT version

1.30.3-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBdongle-E

closed 2023-04-28 by draggero

sodiniss, 2023-05-03:
Hello, I have the same problem, how did you fix it? Thanks

draggero, 2023-05-03:
Hi, i made external converter from here https://github.com/Koenkk/zigbee2mqtt/issues/8705 but not tested yet.