Koenkk/zigbee2mqtt #17273: Xiaomi RTCGQ11LM Hacked - Ghost motion

perseus177, 2023-04-07:
### What happened?

From last 2 months, Xiaomi RTCGQ11LM hacked motions sensors log ghost motions every 2 hours. 
The sensors are in the dark corridor of the apartment without windows or mirrors. 
I am updating zigbee2mqttt always when new version is available. 
It worked 2 years without problem.

Configuration
```
"0x00158d0002e29c47": 
{
                "debounce": 1,
                "friendly_name": "0x00158d0002e29c47",
                "occupancy_timeout": 5,
                "temperature_precision": 1
}
```

Hack according to - https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806

### What did you expect to happen?

Do not log ghost motion, only real motion.

### Zigbee2MQTT version

1.30.3 

### Adapter firmware version

20220219

### Adapter

CC2652P

Koenkk, 2023-04-08:
I've also experienced this before, in my case putting in a. new battery solved the issue. (your sensor is reporting 2.9v which is quite low)

perseus177, 2023-04-08:
According to Zigbee2MQTT GUI - battery has 57%

perseus177, 2023-04-10:
Thank you @Koenkk , it fixed my problem

closed 2023-04-10 by perseus177