Koenkk/zigbee2mqtt #17275: Danalock - Not able to finish interview

FredyReady, 2023-04-07:
### What happened?

Hello,
So I have been using Z2M for past 2 months, it is great, most of the things work right out of the box.

About a month ago I bought the Danalock V3-BTZB/V3-BTZBE, at that time I did not have network extended to my main door, so I tried to pair the lock just near the main coordinator - SONOFF ZB-Dongle-E running 6.10.3.0 build 297 and all went well, the lock was controllable from Z2M. So then I focused on extending my network to main door.

After month (Just yesterday) I managed to finally get stable Zigbee network signal to my main door (Tuya plug right next to the door has LQI of around 180). So I was happy I can finally use the Danalock, I switched the mechanical lock, isntalled Danalock and here the issue began.

I tried pairing the danalock via the pinhole (1 push). The Danalock starts to blink green, Z2M notification "Device '0x70ac08fffe845662' joined" popped up, immediately following with "Starting interview of '0x70ac08fffe845662'" the Danalock stops blinking green and after about 10 seconds another notification pops up "Device '0x70ac08fffe845662' left the network".

I have been fiddling and trying various scenarios:

- Pairing Danalock near the coordinator - Failed
- Resetting whole network and pair Danalock only - Failed
- Re-flashing firmware on ZB-Dongle-E - 6.10.3.0 build 297 - and then pair again - Failed
- Flashing to EZSP v9 - 7.1.1.0 build 273 as [suggested](https://github.com/Koenkk/zigbee2mqtt/issues/16093#issuecomment-1427747732) in #16093 by @kizio1000 - Failed (I flashed back to 6.10.3.0 build 297)
- Pairing with Danalock app, updated firmware to latest version - 0.21.7 and the pair - Failed
- I have tried to pair for at least 30 times by now - Failed every single time
- Restart Z2M, restart HA, updated to latest version and HA Core and then pair - Failed
- Remove second ZB-Dongle-E with Router firmware and pair again - Failed

I am getting despaired... I have spent 300 Euro on the lock with the cylinder and now I cannot use it...

The Danalock works well with the app via Bluetooth. And as mentioned, it worked month ago (lock bought on 1st of March, I do not remember which version of Z2M I was running, but I usually try to run latest version).

I can see some resemblance with #14568 and #16093 but no solution available now. I suspect some bug in Z2M maybe?

Any help or solution would be much appreciated, I am prepared to even pay for fix at this stage, as Danalock is the only available Zigbee HA compatible lock available to me.

No other devices on my Zigbee have any issue (Using Sonoff, IKEA, Lidl, EGLO, Tuya)

Also if needed, I will provide any logs, insight, or whatever, I really want to solve this.

I am providing the debug log with try on pairing the Danalock, the address is '0x70ac08fffe845662'.

### What did you expect to happen?

Danalock to pair with Z2M as it did about month ago

### How to reproduce it (minimal and precise)

Pair Danalock V3 FW 0.21.7 with Z2M [1.30.3] using SONOFF ZB-Dongle-E

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF ZB-Dongle-E

### Debug log

Zigbee2MQTT:info  2023-04-07 17:52:02: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":357,"formaldehyd":4,"humidity":49.4,"linkquality":248,"temperature":16.4,"voc":1}'
Zigbee2MQTT:info  2023-04-07 17:52:02: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":357,"formaldehyd":4,"humidity":49.5,"linkquality":248,"temperature":16.4,"voc":1}'
Zigbee2MQTT:info  2023-04-07 17:52:02: Device '0x70ac08fffe845662' joined
Zigbee2MQTT:info  2023-04-07 17:52:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe845662","ieee_address":"0x70ac08fffe845662"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-04-07 17:52:02: Starting interview of '0x70ac08fffe845662'
Zigbee2MQTT:info  2023-04-07 17:52:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe845662","ieee_address":"0x70ac08fffe845662","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-04-07 17:52:03: MQTT publish: topic 'zigbee2mqtt/Bathroom WashingMachine Plug', payload '{"child_lock":"UNLOCK","current":9.69,"energy":4.38,"indicator_mode":"off/on","linkquality":192,"power":2111,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":222}'
Zigbee2MQTT:info  2023-04-07 17:52:03: MQTT publish: topic 'zigbee2mqtt/Bathroom WashingMachine Plug', payload '{"child_lock":"UNLOCK","current":8.62,"energy":4.38,"indicator_mode":"off/on","linkquality":192,"power":2111,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":222}'
Zigbee2MQTT:info  2023-04-07 17:52:04: MQTT publish: topic 'zigbee2mqtt/Bathroom WashingMachine Plug', payload '{"child_lock":"UNLOCK","current":8.62,"energy":4.38,"indicator_mode":"off/on","linkquality":192,"power":1963,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":222}'
Zigbee2MQTT:info  2023-04-07 17:52:04: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":357,"formaldehyd":3,"humidity":49.5,"linkquality":248,"temperature":16.4,"voc":1}'
Zigbee2MQTT:info  2023-04-07 17:52:04: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":357,"formaldehyd":3,"humidity":49.5,"linkquality":248,"temperature":16.4,"voc":4}'
Zigbee2MQTT:info  2023-04-07 17:52:04: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":357,"formaldehyd":3,"humidity":49.5,"linkquality":248,"temperature":16.4,"voc":4}'
[... 19 lines cut ...]
Zigbee2MQTT:info  2023-04-07 17:52:14: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":357,"formaldehyd":4,"humidity":48.9,"linkquality":248,"temperature":16.7,"voc":1}'
Zigbee2MQTT:info  2023-04-07 17:52:14: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":357,"formaldehyd":2,"humidity":48.9,"linkquality":244,"temperature":16.7,"voc":1}'
Zigbee2MQTT:info  2023-04-07 17:52:14: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":357,"formaldehyd":2,"humidity":48.9,"linkquality":248,"temperature":16.7,"voc":3}'
Zigbee2MQTT:info  2023-04-07 17:52:14: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":358,"formaldehyd":2,"humidity":48.9,"linkquality":248,"temperature":16.7,"voc":3}'
Zigbee2MQTT:info  2023-04-07 17:52:15: MQTT publish: topic 'zigbee2mqtt/Bathroom WashingMachine Plug', payload '{"child_lock":"UNLOCK","current":9.65,"energy":4.38,"indicator_mode":"off/on","linkquality":192,"power":2138,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:info  2023-04-07 17:52:15: MQTT publish: topic 'zigbee2mqtt/Bathroom WashingMachine Plug', payload '{"child_lock":"UNLOCK","current":9.65,"energy":4.39,"indicator_mode":"off/on","linkquality":192,"power":2138,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:info  2023-04-07 17:52:16: MQTT publish: topic 'zigbee2mqtt/Bathroom WashingMachine Plug', payload '{"child_lock":"UNLOCK","current":9.65,"energy":4.39,"indicator_mode":"off/on","linkquality":192,"power":2127,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":228}'
Zigbee2MQTT:info  2023-04-07 17:52:17: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":358,"formaldehyd":2,"humidity":48.9,"linkquality":244,"temperature":16.6,"voc":3}'
Zigbee2MQTT:info  2023-04-07 17:52:17: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":358,"formaldehyd":2,"humidity":49,"linkquality":248,"temperature":16.6,"voc":3}'
Zigbee2MQTT:info  2023-04-07 17:52:18: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":358,"formaldehyd":4,"humidity":49,"linkquality":244,"temperature":16.6,"voc":3}'
Zigbee2MQTT:info  2023-04-07 17:52:18: MQTT publish: topic 'zigbee2mqtt/Workshop TH Sensor', payload '{"co2":358,"formaldehyd":4,"humidity":49,"linkquality":244,"temperature":16.6,"voc":1}'
[... cut ...]
Zigbee2MQTT:info  2023-04-07 17:52:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe845662","ieee_address":"0x70ac08fffe845662","status":"failed"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-04-07 17:52:24: MQTT publish: topic 'zigbee2mqtt/Bathroom WashingMachine Plug', payload '{"child_lock":"UNLOCK","current":9.71,"energy":4.39,"indicator_mode":"off/on","linkquality":192,"power":2076,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":228}'

kirovilya, 2023-04-07:
@FredyReady can you record herdsman log of pairing proccess? https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging

FredyReady, 2023-04-08:
@kirovilya Thanks for taking look at this, much apprecaited!

I tried enabling via Z2M config, adding "zigbee_herdsman_debug: true" at the end as suggested in provided link, and somehow it breaks my coordinator, debug is enabled but coordinator won't start, so whole network goes down. If I disable herdsman log, restart Z2M, all is fine. Am I doing something wrong

[image]

I am attaching log with herdsman logging enabled:

`2023-04-08T09:03:39.516Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 1 | 2
2023-04-08T09:03:39.516Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 18900145000004010000010100010000f290c0c6f1ffff0f08f50a010020c0e2ff2036e4ff2000
2023-04-08T09:03:39.517Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":242},"lastHopLqi":144,"lastHopRssi":-64,"sender":61894,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[8,245,10,1,0,32,192,226,255,32,54,228,255,32,0]}}
2023-04-08T09:03:39.628Z zigbee-herdsman:adapter:ezsp:uart <-- [415ab1a97d312a15b65894a524ab5593499c876f05b21d9874f0cfe98bfc7d5d29a5ebc9de6f8ffef0657e]
2023-04-08T09:03:39.629Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,1,0): 415ab1a9112a15b65894a524ab5593499c876f05b21d9874f0cfe98bfc7d29a5ebc9de6f8ffef0657e
2023-04-08T09:03:39.629Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2023-04-08T09:03:39.630Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2023-04-08T09:03:39.630Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 1 | 2
2023-04-08T09:03:39.630Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 189001450000040100ef010100010000c948ae5fd3ffff0d098a0200031602000400000001
2023-04-08T09:03:39.632Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":201},"lastHopLqi":72,"lastHopRssi":-82,"sender":54111,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[9,138,2,0,3,22,2,0,4,0,0,0,1]}}
2023-04-08T09:03:39.715Z zigbee-herdsman:adapter:ezsp:uart <-- [515ab1a97d312a15b658944a24ab5593499cbd5b102b3f9874f2ce9683fd7d5e1f670932fe596b00e7dbe4867e]
2023-04-08T09:03:39.716Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,1,0): 515ab1a9112a15b658944a24ab5593499cbd5b102b3f9874f2ce9683fd7e1f670932fe596b00e7dbe4867e
2023-04-08T09:03:39.716Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (6)
2023-04-08T09:03:39.716Z zigbee-herdsman:adapter:ezsp:uart --> [8610be7e]
2023-04-08T09:03:39.717Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 1 | 2
2023-04-08T09:03:39.717Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 18900145000004010000010100010000f37cbbc6f1ffff0f08f50a010020c0e2ff2036e4ff2000
2023-04-08T09:03:39.718Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":243},"lastHopLqi":124,"lastHopRssi":-69,"sender":61894,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[8,245,10,1,0,32,192,226,255,32,54,228,255,32,0]}}
2023-04-08T09:03:39.829Z zigbee-herdsman:adapter:ezsp:uart <-- [615ab1a97d312a15b65894a524ab5593499c846f05b21d9874f0cfe98bfc7d5d29a5ebc9de6f8ffede837e]
2023-04-08T09:03:39.830Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,1,0): 615ab1a9112a15b65894a524ab5593499c846f05b21d9874f0cfe98bfc7d29a5ebc9de6f8ffede837e
2023-04-08T09:03:39.830Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2023-04-08T09:03:39.830Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2023-04-08T09:03:39.831Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 1 | 2
2023-04-08T09:03:39.831Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 189001450000040100ef010100010000ca48ae5fd3ffff0d098a0200031602000400000001
2023-04-08T09:03:39.832Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":202},"lastHopLqi":72,"lastHopRssi":-82,"sender":54111,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[9,138,2,0,3,22,2,0,4,0,0,0,1]}}
2023-04-08T09:03:39.916Z zigbee-herdsman:adapter:ezsp:uart <-- [715ab1a97d312a15b658944a24ab5593499cbaab142b3f9874f2ce9683fd7d5e1f670932fe596b00e7db07157e]
2023-04-08T09:03:39.916Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,1,0): 715ab1a9112a15b658944a24ab5593499cbaab142b3f9874f2ce9683fd7e1f670932fe596b00e7db07157e
2023-04-08T09:03:39.916Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2023-04-08T09:03:39.917Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2023-04-08T09:03:39.917Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 1 | 2
2023-04-08T09:03:39.917Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 18900145000004010000010100010000f48cbfc6f1ffff0f08f50a010020c0e2ff2036e4ff2000
2023-04-08T09:03:39.918Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":244},"lastHopLqi":140,"la [... cut ...]
2023-04-08T09:03:41.585Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2023-04-08T09:03:41.586Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2023-04-08T09:03:41.586Z zigbee-herdsman:adapter:ezsp:uart -?- waiting reset
2023-04-08T09:03:41.587Z zigbee-herdsman:adapter:ezsp:uart <-- [c9de6f8f6560ef7e]
2023-04-08T09:03:41.588Z zigbee-herdsman:adapter:ezsp:uart <-- CRC error: c9de6f8f6560ef7e|60ef|006e
2023-04-08T09:03:41.588Z zigbee-herdsman:adapter:ezsp:uart --> [a0547d3a7e]
2023-04-08T09:03:41.702Z zigbee-herdsman:adapter:ezsp:uart <-- [11]`

kirovilya, 2023-04-08:
@FredyReady please post the log as a file (or to pastebin) - it's very inconvenient to watch it. post the device paring log, not just the errors.
do you have a usb extension stick? not used with USB3 port?

FredyReady, 2023-04-08:
@kirovilya 

Sorry, not used to doing such things.

[Here is the log](https://gist.github.com/FredyReady/912ffb85ce361662344e3d626ad5a0d6) - I do not know what is happening lately, not even pastebin is loading for me now, so I created github gist.

I am unable to post any pairing log with herdsman debug enabled, as it breaks the coordinator.

I have extension cable, and using USB 3.0 as I have only USB 3.0 ports available, HA is running on Intel NUC which has only 3.0 ports. I have USB 2.0 extension cable available, if that would help?

FredyReady, 2023-04-08:
So I tried switching to USB 2.0 extension cable. With herdsman debugging disabled, the adapter works, network is online. With herdsman debugging enabled, the adapter breaks, same errors, network offline.

Just to rule out additional factors. I tried pairing Danalock with USB 2.0 extension, and issue is still occuring.

kirovilya, 2023-04-08:
@FredyReady I see some connection problems:
[image]

try to changing the USB cable... maybe some other software is preventing you from working with your serial-port

I still want to see the process of pairing your device in order to understand what prevents you from passing the interview

FredyReady, 2023-04-08:
@kirovilya 

Yes, that is what I wrote, once I enable herdsman debug, the coordinator will not connect.

So I am unable to provide herdsman log while pairing the Danalock. I can only provide the Z2M log while pairing. I understand that seeing the pairing process debug is crucial, but I am in dead end now.

I tried 3 different cables already, one USB 2.0, two USB 3.0. The coordinator cannot connect while the debug is turned on, once it is turned off, the coordinator communication is estabilished.

[Here](https://gist.github.com/FredyReady/c5db77158d96bffaf50bd4f9b1559046) is log right after Z2M reboot with herdsman debug logging **enabled**.
[Here ](https://gist.github.com/FredyReady/933eec1d5a236726e22e91687a448a28)is log right after Z2M reboot with herdsman debug logging **disabled**.

It appears that in parallel I have another issue merging, enabling herdsman debug logging breaks the coordinator communication.

Any idea, why enabling debug causes the inability of coordinator connection? If you have any suggestions on how to fix it, let me know, I am ready to configure what's needed.

kirovilya, 2023-04-09:
@FredyReady I see communication problems in the log - data transfer errors. this once again confirms the assumption that there are problems with the serial port. perhaps with disabled hurdsman-log, communications are faster and more stable.
I don't have any idea why this might be.

FredyReady, 2023-04-09:
@kirovilya 

Yeah I do not understand why this happens... But I will try to make some changes with the hope that I would be able to get the herdsman logging working with functional coordinator. If I manage that, then I will post the pairing log.

Thanks for the effort so far anyway :)

FredyReady, 2023-05-07:
So there was apparently some issue with my adapter after all... I jsut received Sonoff ZB-Dongle-P, switched to it, re-pair the whole network and Danalock has paired too!

So I will stay on P version for now, maybe I will try E version in the future to check.

Thanks for the help, closing the thread.

closed 2023-05-07 by FredyReady

FredyReady, 2023-09-12:
> Any update on this? It's now no longer possible to get the P-version, so I'm stuck with Dongle-E... And have the exact same issue with my Danalock. In fact, this adapter seems to have numerous issues, but hopefully it will be updated at some point. Would be great with some input on the Danalock, though.

Unfortunately no update since the issue was with the E version of dongle. No issue with the lock since May. I can actually say it is really reliable.

Anyways, you can get P version on Aliexpress ;) Plenty of pcs on stock. I do recommend switching to P. While E has some benefits like higher transmitting power, this can be compensated with more routers and stable network overall.

Tul47, 2023-09-12:
Thanks, I might try to switch... But I have 50+ devices paired, and many are hard to get to 😭