Koenkk/zigbee2mqtt #17280: Power Plug (PLUG-001SPB2). Device didn't respond to OTA request

garry0garry, 2023-04-08:
### What happened?

Model ID: TS011F
Manufacturer name: _TZ3000_gjnozsaz
Zigbee2MQTT 1.30.3

There is an update, but it cannot be installed. Why?

```
{
    "child_lock": "UNLOCK",
    "indicator_mode": "off/on",
    "last_seen": "2023-04-08T17:21:59+03:00",
    "linkquality": 255,
    "power_outage_memory": "restore",
    "state": "ON",
    "update": {
        "installed_version": 77,
        "latest_version": 192,
        "state": "available"
    },
    "voltage": 213,
    "current": 0,
    "energy": 0,
    "power": 0,
    "update_available": null
}

```

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

7.2.0.0 build 108

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

debug 2023-04-08 17:13:55 Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"0xa4c138c20463410f","transaction":"ttr7l-28"}'

info 2023-04-08 17:13:55 Updating '0xa4c138c20463410f' to latest firmware

debug 2023-04-08 17:13:55 Updating to latest '0xa4c138c20463410f' (TS011F)

debug 2023-04-08 17:13:55 Using endpoint '1'

debug 2023-04-08 17:13:55 Update of '0xa4c138c20463410f' failed (Error: Device didn't respond to OTA request)

info 2023-04-08 17:13:55 MQTT publish: topic 'zigbee2mqtt/0xa4c138c20463410f', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-04-08T17:06:12+03:00","linkquality":255,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":77,"latest_version":192,"state":"available"},"update_available":null,"voltage":214}'
info 2023-04-08 17:13:55MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0xa4c138c20463410f"},"error":"Update of '0xa4c138c20463410f' failed (Device didn't respond to OTA request)","status":"error","transaction":"ttr7l-28"}'

error 2023-04-08 17:13:55 Update of '0xa4c138c20463410f' failed (Device didn't respond to OTA request)

debug 2023-04-08 17:13:55 Error: Device didn't respond to OTA request at requestOTA (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:191:15) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:279:21) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:267:41)

garry0garry, 2023-04-09:
Solution: downgrade adapter firmware to 6.10.3.

malgoe, 2023-04-10:
Same problem here except I'm using a Conbee 1. Coordinator revision: 0x26400500. My plug has installed version 74 instead of 77 but otherwise same behaviour.

closed 2023-05-19 by github-actions[bot]