Koenkk/zigbee2mqtt #17283: Adding 230V AC dimmer HOMA1004 to shenzhen_homa.js

Stefan-3, 2023-04-08:
### Link

https://global.discourse-cdn.com/business4/uploads/athom/original/3X/0/8/085f27006f1209d678c41532484bb2f53dbc9a27.jpeg

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x00124b0014ca06bf","nwkAddr":11640,"manufId":0,"manufName":"ShenZhen_Homa","powerSource":"Mains (single phase)","modelId":"HOMA1004","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"HOMA1004","manufacturerName":"ShenZhen_Homa","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":94}}},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2","zclVersion":1,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1680970188315,"defaultSendRequestWhen":"immediate"}

### Comments

As a workaround I've added the missing device 'HOMA1004' to IoBroker file "shenzhen_homa.js":
/opt/iobroker/node_modules/zigbee-herdsman-converters/devices/shenzhen_homa.js

    {
        zigbeeModel: ['HOMA1002', 'HOMA1004', 'HOMA0019', 'HOMA0006', 'HOMA000F', '019'],
        model: 'HLC610-Z',
        vendor: 'Shenzhen Homa',
        description: 'Wireless dimmable controller',
        extend: extend.light_onoff_brightness(),
    },

and afterwards I added 'HLC610-Z' to the IoBroker file devices.js:
/opt/iobroker/node_modules/iobroker.zigbee/lib/devices.js

    // Shenzhen Homa
    {
        models: ['HLD812-Z-SC', 'HLC821-Z-SC', 'HLC610-Z'],
        icon: 'img/smart_led_driver.png',
        states: lightStates,
        linkedStates: [comb.brightnessAndState],
    },

It works well. Maybe several other people are using the same HOMA1004 dimmer and would be happy if it's support without this handmade modification :-)

Koenkk, 2023-04-09:
Added!

closed 2023-04-09 by Koenkk

Stefan-3, 2023-04-10:
Wow, great. Many thanks for the extremely quick response :-)