Koenkk/zigbee2mqtt #17293: Invalid option for select._backlight_mode: 'low' (valid options: ['LOW', 'MEDIUM', 'HIGH'])

Bastiencc, 2023-04-09:
### What happened?

I get error in HA logs for my devices TuYa - TS130F

### What did you expect to happen?

Uppercase values

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

20221226

### Adapter

zzh! zStack3x0

### Debug log

Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:152
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 19:28:42 (39 occurrences)
Last logged: 20:18:40

    Invalid option for select.chambre_2_volet_2_backlight_mode: 'low' (valid options: ['LOW', 'MEDIUM', 'HIGH'])
    Invalid option for select.cuisine_volet_backlight_mode: 'low' (valid options: ['LOW', 'MEDIUM', 'HIGH'])
    Invalid option for select.couloir_volet_backlight_mode: 'low' (valid options: ['LOW', 'MEDIUM', 'HIGH'])

Koenkk, 2023-04-10:
This has been fixed yesterday in https://github.com/Koenkk/zigbee-herdsman-converters/pull/5641 

Fix will be included in the next release.

closed 2023-04-10 by Koenkk

Bastiencc, 2023-08-06:
Hi 
I still got this message with the latest version (1.32.2), is it normal?