Koenkk/zigbee2mqtt #17294: [New device support]: ZemiSmart Smart Wave Switch SM0001 _TZ3000_jcqs2mrv, white label ZM-H7

kiselio, 2023-04-10:
### Link

https://www.aliexpress.com/item/1005005175284937.html

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c1381a4f7346e5","nwkAddr":63649,"manufId":4107,"manufName":"_TZ3000_jcqs2mrv","powerSource":"Mains (single phase)","modelId":"SM0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"SM0001","manufacturerName":"_TZ3000_jcqs2mrv","powerSource":1,"zclVersion":3,"appVersion":5,"stackVersion":2,"hwVersion":0,"dateCode":"20221027","swBuildId":"2.0.1"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe7d7925","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":5,"stackVersion":2,"hwVersion":0,"dateCode":"20221027","swBuildId":"2.0.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1681108222073,"defaultSendRequestWhen":"immediate"}

### Comments

I need help supporting this ZemiSmart Smart Switch, based on a mmWave sensor. I created a partially working converter, cannot get the power on/off state to work. Or maybe exposes some other clusters, if available (proximity etc.)?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'SM0001',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_jcqs2mrv',
        },
    ],
    zigbeeModel: ['SM0001'],
    model: 'SM0001',
    vendor: 'ZemiSmart',
    description: 'ZemiSmart Wave Switch',
    extend: tuya.extend.switch(),
    whiteLabel: [{vendor: 'ZemiSmart', model: 'SM0001', description: 'Wave Switch'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);    
    },
};

module.exports = definition;
```

Koenkk, 2023-04-25:
Added, thanks @aginux ! Device will be supported in the next release (1 May)

Koenkk, 2023-04-27:
Yes please provide the log

closed 2023-06-04 by github-actions[bot]