Koenkk/zigbee2mqtt #17304: "...." left the network spam

DanielGroot, 2023-04-11:
### What happened?

I'm constantly trying to add new lights but whenever i want to add them I just get a spam of ""id" left the network"

[image]

And i don't know what to do.

Its happening with these lights: https://www.zigbee2mqtt.io/devices/HG06462A.html#lidl-hg06462a

### Zigbee2MQTT version

1.30.3-1

### Adapter firmware version

6.10.3.0 or something, idk i couldnt get it to update firmware

### Adapter

Sonoff Zigbee Dongle 3.0 Plus

DanielGroot, 2023-04-11:
Found a weird workaround.

1. Put the light in pairing mode without having Permit Join on.
2. Restart zigbee2mqtt.
3. Once restarted turn on Permit Join and it should work

szymucha94, 2023-04-12:
check adapter fw version: z2m > settings > about > coordinator revision
Last time I had such issues it was original firmware my adapter (sonoff cc2652p) came with. Currently I'm on zStack3x0 20220219 and it's no longer an issue.

MattWestb, 2023-04-12:
@szymucha94 EZSP is on stable version (6.10.3.0  or the 6.8.X) is normally not having this problems and is different with TI coordinators that you is having.

closed 2023-05-21 by github-actions[bot]