Koenkk/zigbee2mqtt #17306: Pressing physical button on Xiaomi WS-USC04 does not update state

ic-dev21, 2023-04-11:
### What happened?

I need to manually press the update button for it to show.

I would like to use the physical pressing for automations but this does not work.

### What did you expect to happen?

When I press the button I expect the state to update.

### How to reproduce it (minimal and precise)

Press the button, does not update.

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

20210708

### Adapter

Zdongle-P

ic-dev21, 2023-04-13:
Updated firmware for sonoff stick. Issue remains.

ic-dev21, 2023-04-20:
Setting up reporting manually fixed this.

calkeo, 2023-04-20:
@ic-dev21 can you share the reporting that you manually set up for this? Seems to be related to #17305

ic-dev21, 2023-04-20:
> @ic-dev21 can you share the reporting that you manually set up for this? Seems to be related to #17305

Sure thing, I added the onoff state of both switches as endpoints with a 10s internval. Here you go:

[image]

closed 2023-05-29 by github-actions[bot]

Simon-CR, 2024-05-02:
I am still having this issue. (need to remove stale)