Koenkk/zigbee2mqtt #17309: Don't have access to frontpage

AndreScalaPT, 2023-04-12:
### What happened?

Trying to setup zigbee2mqtt in my docker compose, running docker on ubuntu 22.04 (VM).
I added web access to the portainer using cloudflare tunnels.
Finally I don't get more erros in the logs but for some reason I don't have access to the fronthead page. 

I've changed the port because I'm using that port (8080).

### What did you expect to happen?

Have access to the gui from the web and inside home assistant (iframe).

### How to reproduce it (minimal and precise)

/opt/docker-compose.yaml

...
```
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: unless-stopped
    volumes:
      - /opt/zigbee2mqtt/data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      # Frontend port
      - 8487:8080
    environment:
      - TZ=Europe/Amsterdam
    devices:
      # Make sure this matched your adapter location tee
      #- /dev/ttyACM0:/dev/ttyACM0
      - /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221130082011-if00:/dev/ttyACM0
```

opt/zigbee2mqtt/data/configuration.yaml

```
# Home Assistant integration (MQTT discovery)
homeassistant: true

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://192.168.30.11'
  # MQTT server authentication, uncomment if required:
  user: MYUSR
  password: 'MYPW'

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0
```

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

don't know

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Debug log

Using '/app/data' as data directory

Zigbee2MQTT:info  2023-04-12 11:01:39: Logging to console and directory: '/app/data/log/2023-04-12.11-01-39' filename: log.txt

Zigbee2MQTT:info  2023-04-12 11:01:40: Starting Zigbee2MQTT version 1.30.3 (commit #24c6b2e)

Zigbee2MQTT:info  2023-04-12 11:01:40: Starting zigbee-herdsman (0.14.103)

Zigbee2MQTT:info  2023-04-12 11:01:42: zigbee-herdsman started (resumed)

Zigbee2MQTT:info  2023-04-12 11:01:42: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'

Zigbee2MQTT:info  2023-04-12 11:01:42: Currently 0 devices are joined:

Zigbee2MQTT:warn  2023-04-12 11:01:42: `permit_join` set to  `true` in configuration.yaml.

Zigbee2MQTT:warn  2023-04-12 11:01:42: Allowing new devices to join.

Zigbee2MQTT:warn  2023-04-12 11:01:42: Set `permit_join` to `false` once you joined all devices.

Zigbee2MQTT:info  2023-04-12 11:01:42: Zigbee: allowing new devices to join.

Zigbee2MQTT:info  2023-04-12 11:01:42: Connecting to MQTT server at mqtt://192.168.30.11

Zigbee2MQTT:info  2023-04-12 11:01:42: Connected to MQTT server

Zigbee2MQTT:info  2023-04-12 11:01:42: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

Zigbee2MQTT:info  2023-04-12 11:01:42: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"24c6b2e","coordinator":{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"},"log_level":"info","network":{"channel":11,"extendedPanID":221,"panID":6754},"permit_join":true,"version":"1.30.3"}'

Zigbee2MQTT:info  2023-04-12 11:01:42: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

Zigbee2MQTT:info  2023-04-12 11:01:42: Zigbee2MQTT started!

Zigbee2MQTT:info  2023-04-12 11:02:20: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'

Zigbee2MQTT:info  2023-04-12 11:02:20: Disconnecting from MQTT server

Zigbee2MQTT:info  2023-04-12 11:02:20: Stopping zigbee-herdsman...

Zigbee2MQTT:info  2023-04-12 11:02:20: Stopped zigbee-herdsman

Zigbee2MQTT:info  2023-04-12 11:02:20: Stopped Zigbee2MQTT

Using '/app/data' as data directory

Zigbee2MQTT:info  2023-04-12 11:02:21: Logging to console and directory: '/app/data/log/2023-04-12.11-02-21' filename: log.txt

Zigbee2MQTT:info  2023-04-12 11:02:21: Starting Zigbee2MQTT version 1.30.3 (commit #24c6b2e)

Zigbee2MQTT:info  2023-04-12 11:02:21: Starting zigbee-herdsman (0.14.103)

Zigbee2MQTT:info  2023-04-12 11:02:24: zigbee-herdsman started (resumed)

Zigbee2MQTT:info  2023-04-12 11:02:24: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'

Zigbee2MQTT:info  2023-04-12 11:02:24: Currently 0 devices are joined:

Zigbee2MQTT:warn  2023-04-12 11:02:24: `permit_join` set to  `true` in configuration.yaml.

Zigbee2MQTT:warn  2023-04-12 11:02:24: Allowing new devices to join.

Zigbee2MQTT:warn  2023-04-12 11:02:24: Set `permit_join` to `false` once you joined all devices.

Zigbee2MQTT:info  2023-04-12 11:02:24: Zigbee: allowing new devices to join.

Zigbee2MQTT:info  2023-04-12 11:02:24: Connecting to MQTT server at mqtt://192.168.30.11

Zigbee2MQTT:info  2023-04-12 11:02:24: Connected to MQTT server

Zigbee2MQTT:info  2023-04-12 11:02:24: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

Zigbee2MQTT:info  2023-04-12 11:02:24: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"24c6b2e","coordinator":{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"},"log_level":"info","network":{"channel":11,"extendedPanID":221,"panID":6754},"permit_join":true,"version":"1.30.3"}'

Zigbee2MQTT:info  2023-04-12 11:02:24: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

Zigbee2MQTT:info  2023-04-12 11:02:24: Zigbee2MQTT started!

MrRagga-, 2023-04-12:
https://www.zigbee2mqtt.io/guide/configuration/frontend.html

`frontend: true` I think you need this in your configuration.yaml

AndreScalaPT, 2023-04-12:
> https://www.zigbee2mqtt.io/guide/configuration/frontend.html
> 
> `frontend: true` I think you need this in your configuration.yaml

Already tried that :(
And there's an sub option that you can specified also the domain but still nothing !
thanks

AndreScalaPT, 2023-04-12:
> https://www.zigbee2mqtt.io/guide/configuration/frontend.html
> 
> `frontend: true` I think you need this in your configuration.yaml

i was trying troubleshooting the mqtt and just notice I have that option in the wrong place!!! thanks is working now!
I had it in the MQTT settings lol he need to be outside MQTT settings :P

```
#  MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
```

closed 2023-04-12 by AndreScalaPT