Koenkk/zigbee2mqtt #17313: BRT-100-TRV wrong firmware update

mnowok, 2023-04-12:
### What happened?

I bought 2 Moes BRT-100-TRV thermostatic radiator valves (Official Moes Aliexpress shop). They work fine with Z2H and firmware update is being detected for them. I updated one of them and unfortunately it turned out that the device stopped working properly. Zigbee mopdule stopped working - no connection with Z2H, pressing pair button (long minus) doesn't start pairing mode anymore. Moreover the display doesn't turn off anymore - it only dims a little, but it will drain batteries very fast.

I read about issue of wrong firmware update (https://github.com/Koenkk/zigbee2mqtt/issues/10140) and disassembled the device, but the zigbee module is newer than in described thread. It is ZT5 and has completely different pin layout. I read about this module and it seems there are no JLINK pins, so I couldn't try to flash old firmware.
[image]
[image]
[image]

I have another device which hasn't been (fortunately) updated, so I can deliver some extra info, if needed.

### What did you expect to happen?

Z2H shouldn't offer update for this version of TS0601 (_TZE200_b6wax7g0).

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

 0x26780700

### Adapter

ConBee2/RaspBee2

vmishurov, 2023-05-10:
A couple of days ago I had exactly the same thing as you.
After OTA update, the zigbee module does not work (and the screen does not turn off).
Probably the device can be thrown into the trash ...

chilled35, 2023-05-10:
You can 'jlink' update this newer ZT5 module, the pins labelled 0 and 00 form part of the interface.  That said, I even went down the road of making an ESP32-Jlink gizmo called a BlackMagic - https://forum.blackmagicdesign.com/viewtopic.php?f=12&t=103495 and basically gave up after a day of scratching my head trying to make a valid connection.

In the end I simply logged a customer support ticket with Moes via AE and complained the TRV was dead on arrival and wouldn't go into zigbee pairing mode.  I had to supply a VIDEO demonstrating the valve not going into pairing mode.  They /eventually/ offered me a replacement if I paid an extra £1 processing fee.  Replacement arrived about 10 days later and still unto today, Z2M is offering an update I'm steering well clear of.

vmishurov, 2023-05-11:
[Here](https://github.com/Koenkk/zigbee-herdsman-converters/issues/2932) we discussed another strange behavior.
The same version of z2m, the same device.
But in one case there is an update for TRV, and in the other there is not.

mnowok, 2023-05-11:
> You can 'jlink' update this newer ZT5 module, the pins labelled 0 and 00 form part of the interface. 

Why do you claim pins 0 and 00 are jlink pins? Zt% documentaions says:

0 | SWS | I/O | Burning pin, which corresponds to SWS (Pin 5) on the internal IC
00 | B5 | I/O | Support hardware PWM and correspond to B5 (Pin 15) on the internal IC

chilled35, 2023-05-11:
Because it's listed on the Tuya IoT dev pages, here: https://developer.tuya.com/en/docs/iot/burn-and-authorize-zstyzs-series-modules?id=Ka8x29inhcx8d

vmishurov, 2023-06-11:
Error not fixed.
I got a new device (_TZE200_b6wax7g0), but the problem is old- wrong firmware version for update:
    "update": {
        "installed_version": 67,
        "latest_version": 192,
        "state": "available"
    },

closed 2023-07-20 by github-actions[bot]

MattWestb, 2023-07-21:
Beware that the original TRV was shipped with one module that was having one Silabs chip and the above problems ones is having one of tuya own cooked one do the EZSP OTA file shall not working or braking the module.

22MarioZ, 2023-07-22:
Does anybody have original FW dumb?

MattWestb, 2023-07-22:
The question is is it for the new Zigbee module `ZT5` or the original Silabs `TYZS3-IPEX`.
They is 256% not compatible and is tuya have make it wrong its possible flashing the wrong version and braking the Zigbee module with OTA.

Also from dumped tuya MCU devices Zigbee modules we is knowing that the Zigbee module is hawing one "factory name" that is not being used if the MCU is working OK then its read the ID from the MCU and is using it as its manufacture like `_TZE200_b6wax7g0` is the Zigbee manufactur but is not in the flash of the Zigbee module its coming from MCU it have reading it.

22MarioZ, 2023-07-23:
I'm looking for the new Zigbee module ZT5. For the original Silabs the instructions and FW are described in this [post](https://github.com/Koenkk/zigbee2mqtt/issues/10140#issuecomment-1006876103) by @YoannLe.

MattWestb, 2023-07-23:
Great so you is getting the right firmware and not braking the Zigbee module.

I have only experience of flashing Silabs modules like LIDL one and they its always possible reading the flash back as long its one first gen (MG1X) but if its one MG21 its normally not protected but can like IKEA have start debug locking there modules so cant reading the flash but they was making one mistake the have shipping may unprotected MG21 devices like STYRBAR we can dumping and then rolling on the last OTA file for one °new device° and getting it as one original but without the protection.

How to read and writ the ZT5 i dont knowing but if i remember right have you instruction in the tuya dev docks.

jonathan-gatard, 2023-12-09:
Hi everybody,
Solution can be found here : [https://github.com/Koenkk/zigbee2mqtt/issues/19173](https://github.com/Koenkk/zigbee2mqtt/issues/19173)