Koenkk/zigbee2mqtt #17315: Bosch RFPR-ZB-SH-EU - Problem with pairing

smekac95, 2023-04-12:
### What happened?

I bought this model motion sensor - Bosch RFPR-ZB-SH-EU, I try paring with coordinator, but is not succesfull. Device was reset and is  ready to pairing. First after insert battery flash green, yellow LED stil is flashing and the ended red light, next is slow yellow flashing.

`2023-04-12 21:03:39 Device '0x000d6f0011f8692d' joined
2023-04-12 21:03:40 Starting interview of '0x000d6f0011f8692d'
2023-04-12 21:05:58Failed to interview '0x000d6f0011f8692d', device has not successfully been paired`

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

20201010

### Adapter

CC2652P

itkama, 2023-04-26:
Did you use the install code? To quote from the Bosch Radiator Thermostat:

> Pairing
To pair this device you have to install the device via its installation code. The installation code can be obtained by scanning the QR-code on the inside of the battery cover with your smartphone. Then get the device into pairing mode. In zigbee2mqtt navigate to "Settings" --> "Tools" and click on "Add install code". Paste the code you got from the QR-code and confirm by clicking "OK" which will get zigbee2mqtt into pairing mode automatically. Wait for your device to be joined.

smekac95, 2023-04-28:
Hmmm, this is looks like solving. But, now is too late. I musted send product back to seller. I didn´t know how to use install code. So this issue is maybe solved. :-) Thanks, for your reply, was usefull.

closed 2023-04-28 by smekac95

mgrzelczyk, 2023-06-19:
I have the similar problem. Doing with instruction https://www.zigbee2mqtt.io/devices/RFPR-ZB-SH-EU.html I've put install code from QR picture. It is not the same as short wrote on back of detector. Nothing change (still red blink), but after few times of past code I've got zigbee2mqtt error as:
[image]

itkama, 2023-06-19:
@mgrzelczyk What coordinator are you running?

mgrzelczyk, 2023-06-19:
SONOFF ZBDongle-P 3.0.

mgrzelczyk, 2023-11-09:
I just want to say how do I fixed problem with pairing; I bought from second hand device so need to reset after or before put QR numbers in zigbee2mqtt; Reseting is simple: 1) remove battery, 2) hold button on back and insert battery, 3) should blink green LED and until blinking yellow drop button on back 4) If we set up join device new Bosch should be on list;