Koenkk/zigbee2mqtt #17316: Shyne LED Filament (recognized as model EcoDim )flashes flashes sporadically

richimaint, 2023-04-12:
### What happened?

II have 2 Shyne A-371884 LED filament lamps that are recognized as EcoDim ED-10041 in Z2M. When these are turned off they flash sporadically, likewise when they are turned on these go off briefly.

### What did you expect to happen?

do not flash constantly 

### How to reproduce it (minimal and precise)

do not know how to reproduce this 

### Zigbee2MQTT version

have these only since version 1.29

### Adapter firmware version

20220219

### Adapter

Sonoff ZigBee Dongle 3.0 plus

richimaint, 2023-05-15:
nobody the same problem?

richimaint, 2023-06-16:
nobody

closed 2023-07-25 by github-actions[bot]

ckulas, 2024-01-10:
I have it too while using 1.35.1.

richimaint, 2024-01-10:
the same with 1.35.1

ckulas, 2024-01-11:
Can we identify the exact version where it started?

richimaint, 2024-01-11:
I suspect that it is not due to Z2M, since I have had these lights, they have been flashing sporadically.

ckulas, 2024-01-11:
I had them coupled with my Hue bridge - no flashing. I was running (I think) 1.15 and didn't see the flashing. What do you suspect it to be if not Z2M?

richimaint, 2024-01-11:
I think I've only had Z2M since 1.28.0 and that's where it happened.

ckulas, 2024-01-24:
Solved my issue by resetting and repairing the lights. This is a pain in the ass with these lights, but I have a smart power outlet which I was scripting for this purpose.

richimaint, 2024-01-24:
How did you reset them?

FrazerDungeon, 2024-07-06:
Hello,
I have the same problem with the same product. I use the lamps since almost three years and have had an old version of z2m. About 6 months ago I updated z2m to the latest version and since then this kind of lamps a blinking sporadic. I update since the z2m regularly and also reseted the lamp as described above. But no success. I dont know which version number I had before.
Any suggestions?