Koenkk/zigbee2mqtt #17318: CRASH on startup (timeout, error in unhelpful)

corporategoth, 2023-04-13:
### What happened?

Every time I attempt to start zigbee2mqtt, it crashes.

This is happening repeatedly.  With the same error.  Every time I try to start up.

I've been updating some of my switches firmwares.  I got through about 15 or so switches when I had a crash (no big, it's happened before).
But now I cant start zigbee2mqtt at all. It straight up crashes every single time.

### What did you expect to happen?

zigbee2mqtt to start up.  And keep going.

Heck, I also expect a single timer timing out not to crash the whole application either.
This HAS been working previously.  With the same set of devices.

### How to reproduce it (minimal and precise)

Start it up.  It happens every single time in my environment.  Always with error:

**Error: {"address":26664,"clusterId":32801,"sequence":9} after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)**

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle-E

### Debug log

```
Using '/data' as data directory
Zigbee2MQTT:debug 2023-04-13 02:29:02: Loaded state from file /data/state.json
Zigbee2MQTT:info  2023-04-13 02:29:02: Logging to console and directory: '/data/log/2023-04-13.02-29-02' filename: log.txt
Zigbee2MQTT:debug 2023-04-13 02:29:02: Removing old log directory '/data/log/2023-04-13.02-13-51'
Zigbee2MQTT:info  2023-04-13 02:29:02: Starting Zigbee2MQTT version 1.30.3 (commit #24c6b2e)
Zigbee2MQTT:info  2023-04-13 02:29:02: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:debug 2023-04-13 02:29:02: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/data/coordinator_backup.json","databaseBackupPath":"/data/database.db.backup","databasePath":"/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2023-04-13 02:29:05: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-04-13 02:29:05: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:debug 2023-04-13 02:29:05: Zigbee network parameters: {"channel":11,"extendedPanID":221,"panID":6754}
Zigbee2MQTT:info  2023-04-13 02:29:05: Currently 149 devices are joined:
Zigbee2MQTT:info  2023-04-13 02:29:05: Basement Hallway Light (0x6c5cb1fffe56cbd3): VZM31-SN - Inovelli Inovelli 2-in-1 switch + dimmer (Router)
Zigbee2MQTT:info  2023-04-13 02:29:05: Gym Lights (0xe0798dfffed92d59): VZM31-SN - Inovelli Inovelli 2-in-1 switch + dimmer (Router)
Zigbee2MQTT:info  2023-04-13 02:29:05: Laundry Room Lights Slave (0xe0798dfffed28c95): VZM31-SN - Inovelli Inovelli 2-in-1 switch + dimmer (Router)

--- SNIP ---

Zigbee2MQTT:info  2023-04-13 02:29:05: Post 6 Light - Tall (0xb0ce1814001d5517): E1F-N5E - Sengled Element color plus E12 (EndDevice)
Zigbee2MQTT:info  2023-04-13 02:29:05: Post 6 Light - Medium (0xb0ce18140016beda): E1F-N5E - Sengled Element color plus E12 (EndDevice)
Zigbee2MQTT:info  2023-04-13 02:29:05: Post 6 Light - Small (0xb0ce1814001d46e2): E1F-N5E - Sengled Element color plus E12 (EndDevice)
Zigbee2MQTT:info  2023-04-13 02:29:05: Zigbee: disabling joining new devices.
Zigbee2MQTT:debug 2023-04-13 02:29:05: Received Zigbee message from 'Master Bathroom Bath Light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16908808,"imageType":257,"manufacturerCode":4655}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-04-13 02:29:05: Connecting to MQTT server at mqtt://mqtt-mosquitto.ix-mqtt.svc.cluster.local
Zigbee2MQTT:debug 2023-04-13 02:29:05: Using MQTT login with username: zigbee2mqtt
Zigbee2MQTT:info  2023-04-13 02:29:05: Connected to MQTT server
Zigbee2MQTT:info  2023-04-13 02:29:05: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:debug 2023-04-13 02:29:06: Received MQTT message on 'zigbee2mqtt/bridge/groups' with data '[{"friendly_name":"master_bedroom_blinds","id":1,"members":[{"endpoint":1,"ieee_address":"0xb4e3f9fffed5e5f3"},{"endpoint":1,"ieee_address":"0x0c4314fffe49d8cf"},{"endpoint":1,"ieee_address":"0xb4e3f9fffe9f7b9c"}],"scenes":[]},{"friendly_name":"matildas_bedroom_blinds","id":2,"members":[{"endpoint":1,"ieee_address":"0xb4e3f9fffe8fb903"},{"endpoint":1,"ieee_address":"0x0c4314fffe49d8cf"}],"scenes":[]},{"friendly_name":"all_blinds","id":3,"members":[{"endpoint":1,"ieee_address":"0xb4e3f9fffed5e5f3"},{"endpoint":1,"ieee_address":"0xb4e3f9fffe8fb903"},{"endpoint":1,"ieee_address":"0x0c4314fffe49d4bf"},{"endpoint":1,"ieee_address":"0x0c4314fffe49dda7"},{"endpoint":1,"ieee_address":"0xb4e3f9fffe9f7b9c"},{"endpoint":1,"ieee_address":"0x0c4314fffe49d8cf"}],"scenes":[]},{"friendly_name":"rear_door_left_lights","id":4,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016c339"},{"endpoint":1,"ieee_address":"0xb0ce18140016b637"}],"scenes":[]},{"friendly_name":"rear_door_right_lights","id":5,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016c64a"},{"endpoint":1,"ieee_address":"0xb0ce18140016c48b"}],"scenes":[]},{"friendly_name":"rear_door_lights","id":6,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016c64a"},{"endpoint":1,"ieee_address":"0xb0ce18140016c48b"},{"endpoint":2,"ieee_address":"0x6c5cb1fffe5e8b21"},{"endpoint":1,"ieee_address":"0xb0ce18140016c339"},{"endpoint":1,"ieee_address":"0xb0ce18140016b637"}],"scenes":[]},{"friendly_name":"deck_post_left_lights","id":7,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016bdf9"},{"endpoint":1,"ieee_address":"0xb0ce18140016b4e3"},{"endpoint":1,"ieee_address":"0xb0ce18140016c0e9"}],"scenes":[]},{"friendly_name":"deck_post_right_lights","id":8,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016c990"},{"endpoint":1,"ieee_address":"0xb0ce18140016bdce"}],"scenes":[]},{"friendly_name":"deck_post_lights","id":9,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016bdf9"},{"endpoint":1,"ieee_address":"0xb0ce18140016b4e3"},{"endpoint":1,"ieee_address":"0xb0ce18140016c990"},{"endpoint":1,"ieee_address":"0xb0ce18140016bdce"},{"endpoint":1,"ieee_address":"0xb0ce18140016c0e9"},{"endpoint":2,"ieee_address":"0x943469fffe089ca7"}],"scenes":[]},{"friendly_name":"porch_left_lights","id":10,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016c8a6"},{"endpoint":1,"ieee_address":"0xb0ce18140016ca14"},{"endpoint":1,"ieee_address":"0xb0ce18140016bf2a"}],"scenes":[]},{"friendly_name":"porch_right_lights","id":11,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016b8a4"},{"endpoint":1,"ieee_address":"0xb0ce18140016b50d"},{"endpoint":1,"ieee_address":"0xb0ce18140016c33b"}],"scenes":[]},{"friendly_name":"porch_lights","id":12,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016c8a6"},{"endpoint":1,"ieee_address":"0xb0ce18140016b8a4"},{"endpoint":1,"ieee_address":"0xb0ce18140016b50d"},{"endpoint":1,"ieee_address":"0xb0ce18140016c33b"},{"endpoint":1,"ieee_address":"0xb0ce18140016ca14"},{"endpoint":2,"ieee_address":"0x385b44fffeee0c34"},{"endpoint":1,"ieee_address":"0xb0ce18140016bf2a"}],"scenes":[]},{"friendly_name":"post_1_lights","id":13,"members":[{"endpoint":1,"ieee_address":"0xb0ce1814001d4f3c"},{"endpoint":1,"ieee_address":"0xb0ce1814001d4ed3"},{"endpoint":1,"ieee_address":"0xb0ce1814001d7a6a"}],"scenes":[]},{"friendly_name":"post_2_lights","id":14,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016bb98"},{"endpoint":1,"ieee_address":"0xb0ce1814001d776d"},{"endpoint":1,"ieee_address":"0xb0ce18140016c3b0"}],"scenes":[]},{"friendly_name":"post_3_lights","id":15,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016c8c5"},{"endpoint":1,"ieee_address":"0xb0ce18140016c952"},{"endpoint":1,"ieee_address":"0xb0ce18140016c52d"}],"scenes":[]},{"friendly_name":"post_4_lights","id":16,"members":[{"endpoint":1,"ieee_address":"0xb0ce18140016bf4d"},{"endpoint":1,"ieee_address":"0xb0ce18140016bf23"},{"endpoint":1,"ieee_address":"0xb0ce18140016b58d"}],"scenes":[]},{"friendly_name":"post_5_lights","id":17,"members":[{"endpoint":1,"ieee_address":"0xb0ce181400170ec1"},{"endpoint":1,"ieee_address":"0xb0ce18140016b80a"},{"endpoint":1,"ieee_address":"0xb0ce1814001d383c"}],"scenes":[]},{"friendly_name":"post_6_lights","id":18,"members":[{"endpoint":1,"ieee_address":"0xb0ce1814001d46e [... cut ...] rsion":"2.14"},"max":255,"min":0,"name":"Gym Lights defaultLevelRemote","state_topic":"zigbee2mqtt/Gym Lights","unique_id":"0xe0798dfffed92d59_defaultLevelRemote_zigbee2mqtt","value_template":"{{ value_json.defaultLevelRemote }}"}'
```
**Error: {"address":26664,"clusterId":32801,"sequence":9} after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)**

corporategoth, 2023-04-13:
So my zigbee2mqtt managed to start up sometime this morning.

It's possible that this could be due to high I/O load on the system? though that doesn't make sense.

Either way.  1) is there a way to change that timeout (if it IS due to high load, I want to give it more time)
and 2) The process should NOT terminate because a timeout expired.  In fact, ideally the process should never terminate unless something critical has happened that forces it not to continue (which is not indicated with a timeout).

corporategoth, 2023-04-16:
FYI I could get rid of this by manually editing the ezsp.js and driver.js files (probably only needed the latter, but I did both) with `sed 's/timeout = 10000/timeout = 20000/g'`

Meaning that this timeout is both too small (at least for a larger network), AND this should be something configurable. Oh, and yeah, it should not cause the app to die outright.

tomamplius, 2023-04-17:
I have the same issue. I will test increase timeout.
~~~
journalctl -f -n 1000 | grep waitress
avril 16 19:46:44 zigbee2mqtt npm[1653]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
avril 16 20:52:58 zigbee2mqtt npm[3278]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
avril 16 21:51:14 zigbee2mqtt npm[3325]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
avril 17 00:59:29 zigbee2mqtt npm[3370]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
avril 17 04:19:45 zigbee2mqtt npm[3502]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)

tomamplius, 2023-04-17:
After update to 20 000, same issue :
~~~
journalctl -f -n 1000 | grep waitress
avril 17 08:23:22 zigbee2mqtt npm[3741]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
avril 17 09:59:22 zigbee2mqtt npm[3798]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
avril 17 11:49:36 zigbee2mqtt npm[3854]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)

corporategoth, 2023-04-17:
@tomamplius There are different reasons for timeouts - you should include the log line PRIOR to the _onTimeout to see the actual error. It could still be this, but it could be something else entirely.

corporategoth, 2023-04-17:
In my case, I'm pretty sure that zigbee2mqtt is not allowing enough time for startup to finish, and it's timing out - so I doubled the amount of time to wait for startup.  But if yours is happening 3-4 hours in, it's likely a different cause.  Though likely also should not cause a crash.

corporategoth, 2023-04-17:
Looking at that, you're not using an EZSP driver.  Therefore the changes to the files I mentioned above would not change the timeout for you.  As evidenced by:

```
avril 17 18:49:52 zigbee2mqtt npm[3911]: Error: Timeout - 29590 - 1 - 2 - 1794 - 1 after 10000ms
```

It still shows 10 seconds, not 20.  If you want to test this, you need to update the timers for your driver (`zigate` it appears).

tomamplius, 2023-04-18:
You are right, thanks.

My new update is : 

~~~diff zigate/driver/zigate.js zigate/driver/zigate.js.origine
30c30
<     default: 20000,
---
>     default: 10000,
root@zigbee2mqtt:/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter# diff zigate/adapter/zigateAdapter.js zigate/adapter/zigateAdapter.js.origine
159c159
<         return this.driver.sendCommand(constants_1.ZiGateCommandCode.GetNetworkState, {}, 20000)
---
>         return this.driver.sendCommand(constants_1.ZiGateCommandCode.GetNetworkState, {}, 10000)

tomamplius, 2023-04-22:
Error timeout is not link to the timeout of adapter driver .

~~~
avril 22 22:38:15 zigbee2mqtt npm[7842]: zlinky config: historique, single_phase, false, BASE
avril 22 22:38:22 zigbee2mqtt npm[7842]: Zigbee2MQTT:debug 2023-04-22 22:38:22: Received Zigbee message from 'linky', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,14591638]}' from endpoint 1 with groupID null
avril 22 22:38:22 zigbee2mqtt npm[7842]: Zigbee2MQTT:info  2023-04-22 22:38:22: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_power":-32768,"active_power_max":-32768,"active_power_ph_b":-32768,"active_register_tier_delivered":"TH..","apparent_power":590,"available_power":
30,"current_date":"","current_price":"","current_summ_delivered":14591.638,"current_tarif":"BASE","current_tier1_summ_delivered":14544.965,"current_tier2_summ_delivered":0,"last_seen":1682195902661,"linkquality":219,"message1":"","message2":"","meter_serial_number":"021761642440","rms
_current":2,"rms_current_max":90,"rms_voltage":65535,"site_id":"","status_register":"","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"warn_d_p_s":null}'
avril 22 22:38:23 zigbee2mqtt npm[7842]: Zigbee2MQTT:debug 2023-04-22 22:38:23: Received Zigbee message from 'Prise Serveur', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":578}' from endpoint 1 with groupID null
avril 22 22:38:23 zigbee2mqtt npm[7842]: Zigbee2MQTT:info  2023-04-22 22:38:23: MQTT publish: topic 'zigbee2mqtt/Prise Serveur', payload '{"child_lock":"UNLOCK","current":0.58,"energy":794.9,"indicator_mode":"off/on","last_seen":1682195903143,"linkquality":219,"power":105,"power_outag
e_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":235}'
avril 22 22:38:25 zigbee2mqtt npm[7842]: Error: Timeout - 29590 - 1 - 24 - 65382 - 1 after 10000ms
avril 22 22:38:25 zigbee2mqtt npm[7842]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
avril 22 22:38:25 zigbee2mqtt npm[7842]:     at listOnTimeout (node:internal/timers:559:17)
avril 22 22:38:25 zigbee2mqtt npm[7842]:     at processTimers (node:internal/timers:502:7)
avril 22 22:38:25 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
avril 22 22:38:25 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
avril 22 22:38:25 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Consumed 44.296s CPU time.
avril 22 22:38:25 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 28.
avril 22 22:38:25 zigbee2mqtt systemd[1]: Stopped zigbee2mqtt.
avril 22 22:38:25 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Consumed 44.296s CPU time.
avril 22 22:38:25 zigbee2mqtt systemd[1]: Started zigbee2mqtt.
avril 22 22:38:26 zigbee2mqtt npm[7908]: > zigbee2mqtt@1.30.2 start
avril 22 22:38:26 zigbee2mqtt npm[7908]: > node index.js
~~~

But i can find 20 000 on this timeout now, when device is not available the driver timeout is use. 
But not on this issue?
~~~
avril 22 22:37:48 zigbee2mqtt npm[7842]: Zigbee2MQTT:error 2023-04-22 22:37:48: Failed to execute LQI for 'Prise srv backup'
avril 22 22:37:48 zigbee2mqtt npm[7842]: Zigbee2MQTT:debug 2023-04-22 22:37:48: Error: LQI REQUEST FAILED - addr: 0xa2c8 Error: [object Object] after 20000ms
avril 22 22:37:48 zigbee2mqtt npm[7842]:     at request (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/zigate/adapter/zigateAdapter.ts:274:43)
avril 22 22:37:48 zigbee2mqtt npm[7842]:     at Object.func (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/zigate/adapter/zigateAdapter.ts:278:28)
avril 22 22:37:48 zigbee2mqtt npm[7842]:     at Queue.executeNext (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
~~~

tomamplius, 2023-04-22:
i test npm migration from 16.19 to 16.20

tomamplius, 2023-04-22:
No issue now after upgrade nodejs on map generator

tomamplius, 2023-04-23:
Same error in  16.20
~~~
avril 23 09:29:47 zigbee2mqtt npm[9279]: Zigbee2MQTT:info  2023-04-23 09:29:47: MQTT publish: topic 'zigbee2mqtt/volet_salon', payload '{"last_seen":1682234987496,"linkquality":93,"position":95,"state":"OPEN"}'
avril 23 09:29:47 zigbee2mqtt npm[9279]: Zigbee2MQTT:debug 2023-04-23 09:29:47: Device 'volet_salon' requested OTA
avril 23 09:29:47 zigbee2mqtt npm[9279]: Zigbee2MQTT:debug 2023-04-23 09:29:47: Responded to OTA request of 'volet_salon' with 'NO_IMAGE_AVAILABLE'
avril 23 09:29:48 zigbee2mqtt npm[9279]: Error: Timeout - 29590 - 1 - 122 - 2820 - 1 after 10000ms
avril 23 09:29:48 zigbee2mqtt npm[9279]:     at Timeout._onTimeout (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
avril 23 09:29:48 zigbee2mqtt npm[9279]:     at listOnTimeout (node:internal/timers:559:17)
avril 23 09:29:48 zigbee2mqtt npm[9279]:     at processTimers (node:internal/timers:502:7)
avril 23 09:29:48 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
avril 23 09:29:48 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
avril 23 09:29:48 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Consumed 2min 12.231s CPU time.
avril 23 09:29:49 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
avril 23 09:29:49 zigbee2mqtt systemd[1]: Stopped zigbee2mqtt.
avril 23 09:29:49 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Consumed 2min 12.231s CPU time.
avril 23 09:29:49 zigbee2mqtt systemd[1]: Started zigbee2mqtt.
avril 23 09:29:49 zigbee2mqtt npm[9671]: > zigbee2mqtt@1.30.2 start
avril 23 09:29:49 zigbee2mqtt npm[9671]: > node index.js
avril 23 09:29:51 zigbee2mqtt npm[9684]: Zigbee2MQTT:debug 2023-04-23 09:29:51: Loaded state from file /var/lib/zigbee2mqtt/state.json
avril 23 09:29:51 zigbee2mqtt npm[9684]: Zigbee2MQTT:info  2023-04-23 09:29:51: Logging to console and directory: '/var/lib/zigbee2mqtt/log/2023-04-23.09-29-51' filename: log.txt
avril 23 09:29:51 zigbee2mqtt npm[9684]: Zigbee2MQTT:debug 2023-04-23 09:29:51: Removing old log directory '/var/lib/zigbee2mqtt/log/2023-04-22.15-23-14'
avril 23 09:29:51 zigbee2mqtt npm[9684]: Zigbee2MQTT:info  2023-04-23 09:29:51: Starting Zigbee2MQTT version 1.30.2 (commit #cdf62ea)
~~~

corporategoth, 2023-05-11:
@Koenkk Can you make the timeout values adjustable in the config?  Rather than hard-coding them to 10s?

Also - part of this may be I/O.  when I up my logging threshold to 'warning' (ie. it doesn't dump out all the messages it's sending to HA, which happens at INFO for some reason) the 10s timeout is not hit.  But even so, the fact increasing the timeout to 20 worked means this really needs to be a value that can be adjusted based on environment.

corporategoth, 2023-05-12:
After discussion on the discord.

The real reason this is a problem is I have a number of Inovelli switches (more than 60), which EACH generate 150 entities.  When Z2M starts up, it and HA do discovery - and I end up having more than 10,000 entities that HA has to discover.

With logging turned off, this happens fine.  With logging turned ON, the logging itself interferes with the communications with the coordinator, because logging is synchronous.

This means, if anything causes a rapid burst of logs (in this case more than 20 mb worth of logs) in 10 seconds, it will cause Z2M to time out it's communications with the coordinator.

1) Logs should be done async - ideally in a separate thread.  I'm going to see if I can make an MR for this, but I'm NOT a javascript coder (I primarily write in C++ and some python).
2) The timeout values should still be adjustable, and not hard-coded to 10 seconds.

NOTE - with logging set to WARN (ie. it's not trying to log all the discovery activity to home assistant), both Z2M and HA handle the 10k entities just fine.  So this is explicitly a problem with logging causing a timeout.  Meaning that Z2M is in a situation where logging can cause it to not do it's job, which should not be the case.

closed 2023-06-20 by github-actions[bot]

TonyBunce, 2023-11-27:
> After discussion on the discord.
>
> The real reason this is a problem is I have a number of Inovelli switches (more than 60), which EACH generate 150 entities. When Z2M starts up, it and HA do [...]

**THANK YOU**  I also have a bunch of Inovelli switches and have been banging my head against the keyboard trying to figure out why Z2M suddenly wouldn't start up.  Disabled debug logging and now I'm back in business