Koenkk/zigbee2mqtt #17328: [New device support]: Philips Hue Signe Table Lamp - 915005987401

kidinstinct, 2023-04-14:
### Link

https://www.amazon.com/gp/product/B09BDL522H/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0x001788010c538ee3","nwkAddr":315,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"915005987401","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"zclVersion":8,"modelId":"915005987401","manufacturerName":"Philips","powerSource":1,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20221006","swBuildId":"1.101.2"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20221006","swBuildId":"1.101.2","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1681433644707,"defaultSendRequestWhen":"immediate"}

### Comments

I'm using the docker installation and attempted to follow the instructions [here ](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)for adding new devices. See external converter below, I took the sample code from an existing closed issue, but received error and container wouldn't start.

```
/app/data/extension/externally-loaded.js:14
        toZigbee: [tzLocal.gradient_scene, ...hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).toZigbee],
                   ^

ReferenceError: tzLocal is not defined
    at /app/data/extension/externally-loaded.js:14:20
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:300:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
Using '/app/data' as data directory
/app/data/extension/externally-loaded.js:14
        toZigbee: [tzLocal.gradient_scene, ...hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).toZigbee],
                   ^

ReferenceError: tzLocal is not defined
    at /app/data/extension/externally-loaded.js:14:20
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:300:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
Using '/app/data' as data directory
```

I also switched to the latest-dev tag just in case.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['915005987401'],
        model: '915005987401',
        vendor: 'Philips',
        description: 'Hue Gradient Signe Floor Lamp',
        toZigbee: [tzLocal.gradient_scene, ...hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).toZigbee],
        exposes: [exposes.enum('gradient_scene', ea.SET, Object.keys(gradientScenes)),
            ...hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).exposes],
        extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    };

    module.exports = definition;
```

kidinstinct, 2023-04-14:
Ok I was able to get it to work with this code below. Took me a bit, forgive me I'm slow. Not sure if all functionality is there, but it seems to work.

I just copied from the converters repo and changed the model.

```js
const extend = require('zigbee-herdsman-converters/lib/extend');
const philips = require('zigbee-herdsman-converters/lib/philips');

const definition = {
  zigbeeModel: ['915005987401'],
        model: '915005987401',
        vendor: 'Philips',
        description: 'Hue White and color ambiance Gradient Signe table lamp (black)',
        extend: philips.extend.light_onoff_brightness_colortemp_color_gradient({colorTempRange: [153, 500]})
};

module.exports = definition;
```

Koenkk, 2023-04-14:
Added, thanks for your contribution!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-04-14 by Koenkk