Koenkk/zigbee2mqtt #17329: [New device support]: Aqara Smart Plug - US

kidinstinct, 2023-04-14:
### Link

https://www.aqara.com/us/support/smart-plug-us

### Database entry

{"id":7,"type":"Unknown","ieeeAddr":"0x00158d000837947b","nwkAddr":42701,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1681435958593,"defaultSendRequestWhen":"immediate"}

### Comments

Hopefully this is not too much trouble to add, I'm trying to get away from vendor cloud accounts and keep things local.

Log from device..

```
Info 2023-04-13 21:32:38Starting interview of '0x00158d000837947b'
Info 2023-04-13 21:32:38MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d000837947b","ieee_address":"0x00158d000837947b","status":"started"},"type":"device_interview"}'
```

[image]

### External converter

```shell
I wouldn't know how to even get started with this one on adding a fully unsupported device.
```

kidinstinct, 2023-04-14:
So I tried with this code below, but I'm assuming since it can't see the model number, it doesn't act on it.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  zigbeeModel: ['lumi.plug.aq1'],
  model: 'ZNCZ12LM',
  vendor: 'Xiaomi',
  description: 'Aqara Smart Plug - US',
  fromZigbee: [fz.on_off, fz.xiaomi_power, fz.ignore_occupancy_report, fz.xiaomi_basic],
  toZigbee: [
    tz.on_off,
    tz.xiaomi_power,
    tz.xiaomi_led_disabled_night,
    tz.xiaomi_switch_power_outage_memory,
    tz.xiaomi_auto_off,
  ],
  exposes: [
    e.switch(),
    e.power().withAccess(ea.STATE_GET),
    e.energy(),
    e.device_temperature(),
    e.voltage(),
    e.power_outage_memory(),
    e.led_disabled_night(),
    e.auto_off(30),
  ],
};

module.exports = definition;
```

kidinstinct, 2023-04-14:
Looks like force removing the device, then re-pairing did work with the above code.

[image]

Koenkk, 2023-04-14:
In the first screenshot you see that the interviewing failed, seems it worked the second attempt. Enjoy!

closed 2023-04-14 by Koenkk