Koenkk/zigbee2mqtt #17346: Niko double switch action not published for right switch

svenjochems, 2023-04-15:
### What happened?

Niko double rocker switch model 552-721X2
Both left and right are configured in decoupled mode.

When pushing the left switch, the following mqtt messages are published:
```
zigbee2mqtt/living_switch/action hold
zigbee2mqtt/living_switch/action release
```

When pushing the right switch, no change is published.

### What did you expect to happen?

The same 'single', 'hold' and 'release' actions are published for the right switch.

### How to reproduce it (minimal and precise)

Set both left and right switch to 'Decoupled' mode.
Push left switch => action is published
Push right switch => nothing is published

### Zigbee2MQTT version

1.30.3 commit: 24c6b2e

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8 (Sonoff Dongle-E)

### Debug log

 2023-04-15 13:42:22Received Zigbee message from 'living_switch', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16}' from endpoint 1 with groupID 0
info 2023-04-15 13:42:22MQTT publish: topic 'zigbee2mqtt/living_switch', payload '{"action":null,"action_l1":null,"action_l2":null,"led_enable":true,"led_enable_l1":false,"led_enable_l2":false,"led_state":"OFF","led_state_l1":"OFF","led_state_l2":"OFF","linkquality":156,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
debug 2023-04-15 13:42:22Received Zigbee message from 'living_switch', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":64}' from endpoint 1 with groupID 0
info 2023-04-15 13:42:22MQTT publish: topic 'zigbee2mqtt/living_switch', payload '{"action":"single","action_l1":null,"action_l2":null,"led_enable":true,"led_enable_l1":false,"led_enable_l2":false,"led_state":"OFF","led_state_l1":"OFF","led_state_l2":"OFF","linkquality":156,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
info 2023-04-15 13:42:22MQTT publish: topic 'zigbee2mqtt/living_switch', payload '{"action":"","action_l1":null,"action_l2":null,"led_enable":true,"led_enable_l1":false,"led_enable_l2":false,"led_state":"OFF","led_state_l1":"OFF","led_state_l2":"OFF","linkquality":156,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
info 2023-04-15 13:42:22MQTT publish: topic 'zigbee2mqtt/living_switch/action', payload 'single'
debug 2023-04-15 13:42:25Received Zigbee message from 'living_switch', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":4096}' from endpoint 1 with groupID 0
info 2023-04-15 13:42:25MQTT publish: topic 'zigbee2mqtt/living_switch', payload '{"action_l1":null,"action_l2":null,"led_enable":true,"led_enable_l1":false,"led_enable_l2":false,"led_state":"OFF","led_state_l1":"OFF","led_state_l2":"OFF","linkquality":160,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'
debug 2023-04-15 13:42:25Received Zigbee message from 'living_switch', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16384}' from endpoint 1 with groupID 0
info 2023-04-15 13:42:25MQTT publish: topic 'zigbee2mqtt/living_switch', payload '{"action_l1":null,"action_l2":null,"led_enable":true,"led_enable_l1":false,"led_enable_l2":false,"led_state":"OFF","led_state_l1":"OFF","led_state_l2":"OFF","linkquality":160,"operation_mode":"control_relay","operation_mode_l1":"decoupled","operation_mode_l2":"decoupled","state_l1":"OFF","state_l2":"OFF"}'

svenjochems, 2023-04-16:
Below the logs for `single` followed by `hold/release` actions on the right rocker of the double switch.

Looking at https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/niko.js#L33
The codes for the single switch or left rocker are configured as `{16: null, 64: 'single', 32: 'hold', 48: 'release'};`. It looks like this should be `{4096: null, 16384: 'single', 8192: 'hold', 12288: 'release'};` for the right rocker.

```
2023-04-16 15:57:43Received Zigbee message from 'living_switch', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":4096}' from endpoint 1 with groupID 0
2023-04-16 15:57:43Received Zigbee message from 'living_switch', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":16384}' from endpoint 1 with groupID 0
2023-04-16 15:57:53Received Zigbee message from 'living_switch', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":4096}' from endpoint 1 with groupID 0
2023-04-16 15:57:53Received Zigbee message from 'living_switch', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":8192}' from endpoint 1 with groupID 0
2023-04-16 15:57:54Received Zigbee message from 'living_switch', type 'attributeReport', cluster 'manuSpecificNiko2', data '{"switchAction":12288}' from endpoint 1 with groupID 0
```

sjorge, 2023-04-16:
https://github.com/Koenkk/zigbee2mqtt/issues/13737#issuecomment-1362030574

Nobody has been able to confirm that this fixes it though, so it never made it to a PR.

closed 2023-05-24 by github-actions[bot]